OpenZWave/open-zwave #1617: Z-wave problems in 0.79.3

follox, 2018-10-04:
Iam having problems with hass.io 1 after upgrading to 0.79.3.
When I am trying to toggle a light, the toggle goes back to off immediately after turning it on.
The lamp sometimes turns on, sometimes not. If not and I continue clicking on, nothing happening. The toogle just returns to off position.

This is on fibaro wall switch.

follox, 2018-10-04:
@Dinth Yeah, sorry about that. I have changed the title. Hopefully someone can look into this,

Fishwaldo, 2018-10-05:
OZW Logs please.

follox, 2018-10-05:
How should I upload this? I get a limit on pastebin. Any other ways? Do you want the whole ozw log?

follox, 2018-10-05:
@Fishwaldo 
[ozw.txt](https://github.com/OpenZWave/open-zwave/files/2449282/ozw.txt)

Fishwaldo, 2018-10-05:
`2018-10-04 19:23:39.900 Info, Node016, Value::Set - COMMAND_CLASS_SWITCH_BINARY - Switch - 0 - 1 - True
2018-10-04 19:23:39.900 Info, Node016, SwitchBinary::Set - Setting node 16 to On
...
2018-10-04 19:23:39.951 Detail, Expected callbackId was received
2018-10-04 19:23:39.951 Detail, Expected reply was received
2018-10-04 19:23:39.951 Detail, Message transaction complete
...
2018-10-04 19:23:39.952 Info, Node016, Sending (Send) message (Callback ID=0x68, Expected Reply=0x04) - SwitchBinaryCmd_Get (Node=16): 0x01, 0x09, 0x00, 0x13, 0x10, 0x02, 0x25, 0x02, 0x25, 0x68, 0x9d
...
2018-10-04 19:23:40.192 Info, Node016, Received SwitchBinary report from node 16: level=On
2018-10-04 19:23:40.193 Detail, Node016, Refreshed Value: old value=false, new value=true, type=bool
2018-10-04 19:23:40.193 Detail, Node016, Changes to this value are not verified
2018-10-04 19:23:40.193 Detail, Node016, Notification: ValueChanged`

This is fine. You Set it to On, it returns "on"

`2018-10-04 19:24:08.722 Info, Node016, Value::Set - COMMAND_CLASS_SWITCH_BINARY - Switch - 0 - 1 - False
2018-10-04 19:24:08.722 Info, Node016, SwitchBinary::Set - Setting node 16 to Off
...
2018-10-04 19:24:09.095 Detail, Node016, Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x10, 0x03, 0x25, 0x03, 0x00, 0xc7
2018-10-04 19:24:09.095 Detail,
2018-10-04 19:24:09.095 Info, Node016, Received SwitchBinary report from node 16: level=Off
2018-10-04 19:24:09.095 Detail, Node016, Refreshed Value: old value=true, new value=false, type=bool
2018-10-04 19:24:09.095 Detail, Node016, Changes to this value are not verified
2018-10-04 19:24:09.096 Detail, Node016, Notification: ValueChanged
...
2018-10-04 19:24:10.247 Info, Node016, Sending (Send) message (Callback ID=0x6a, Expected Reply=0x04) - SwitchBinaryCmd_Get (Node=16): 0x01, 0x09, 0x00, 0x13, 0x10, 0x02, 0x25, 0x02, 0x25, 0x6a, 0x9f
...
2018-10-04 19:24:10.777 Detail, Node016, Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x10, 0x03, 0x25, 0x03, 0x00, 0xc7
2018-10-04 19:24:10.777 Detail,
2018-10-04 19:24:10.778 Info, Node016, Response RTT 529 Average Response RTT 513
2018-10-04 19:24:10.778 Info, Node016, Received SwitchBinary report from node 16: level=Off
2018-10-04 19:24:10.778 Detail, Node016, Refreshed Value: old value=false, new value=false, type=bool
2018-10-04 19:24:10.778 Detail, Node016, Changes to this value are not verified
2018-10-04 19:24:10.779 Detail, Node016, Expected reply and command class was received
2018-10-04 19:24:10.779 Detail, Node016, Message transaction complete`

Also fine.

A lot of these:
`2018-10-04 19:24:27.482 Error, Node016, ERROR: Dropping command, expected response not received after 1 attempt(s)
2018-10-04 19:24:27.482 Detail, Node016, Removing current message
2018-10-04 19:24:27.483 Detail, Node016, Notification: Notification - TimeOut`

which indicates an unstable network or busy device. Removing the association mappings might help, but then you would not get updates if the device is locally changed. 

All in all - Nothing wrong with OZW in the Logs you sent. So this is either a 1) hass.io issue, 2) network problem, or 3) crappy device (but no one else has reported to OZW about these devices, so I'd say its 1 or 2.

Fishwaldo, 2018-10-05:
> Unable to find service sensor/turn_off 08:26 core.py (WARNING) Unable to find service sensor/turn_on 08:26 core.py (WARNING)

Thats not OZW code. Can't help you there.

> {"association": "remove", "node_id": 3, "group": 2, "target_node_id": 255}

Also not a OZW code so no idea if thats valid or not, but you have no Node 3

closed 2018-10-05 by Fishwaldo

arcangel321, 2020-04-17:
I know this has been closed along time ago, but I experience the same with Aeotec ZW140

arcangel321, 2020-04-17:
This looks like the solutions, still testing:
Configuration
Z-Wave
Select the Node you have issues with
In Node config options select 80: Notification status
As value select one of Basic or Hail
Set config parameter
Be happy