OpenZWave/open-zwave #1620: Heatit Z-trm2 include with s2

follox, 2018-10-05:
I have a heatit Z-trm2 thermostat that wont work as it should in Home Assistant as it should be added with S2 security. Not sure if the template is correct either as I can't control it that well.

The thermostat is running firmware 2_11

Fishwaldo, 2018-10-06:
And?
Nothing for us to diagnose or ?

S2 security should be backwards compatible with S0 so what’s the problem?

follox, 2018-10-06:
The problem is that the thermostat is not working in hass?
[ozw.txt](https://github.com/OpenZWave/open-zwave/files/2452790/ozw.txt)

The thermostat is node21.
```
2018-10-06 08:45:51.270 Error, Node021, Received a MultiChannelEncap for endpoint 5 for Command Class 50, which we can't find
2018-10-06 08:45:51.341 Detail, Node021, Received: 0x01, 0x18, 0x00, 0x04, 0x10, 0x15, 0x12, 0x60, 0x0d, 0x05, 0x01
```

follox, 2018-10-07:
[image]

[image]

Things are not working as it should with the current template

Fishwaldo, 2018-10-07:
You have a lot of these messages:
2018-10-05 14:56:39.265 Error, Node012, ERROR: ZW_SEND_DATA could not be delivered to Z-Wave stack
2018-10-05 14:56:39.267 Detail, Node012, CAN received...triggering resend

Are you sure no other application is running on your device and it’s free from interference (microwaves, WiFi access points etc etc etc)

Fishwaldo, 2018-10-08:
> When added in secure mode the ozw is getting spammed with information from the termostat

Then you need to change the configuration on the device to not send so often.

> When added normally the temperature is not updating until you refresh the node manually

Check the configurtation of the device to see if it supports sending the temp, otherwise enable polling.

 > The temperature in climate card is wrong. It is saying 16.1 degrees when floor sensor it stating 22.9

`2018-10-07 20:11:20.232 Detail, Node028, Received: 0x01, 0x10, 0x00, 0x04, 0x00, 0x1c, 0x0a, 0x60, 0x0d, 0x02, 0x01, 0x31, 0x05, 0x01, 0x22, 0x00, 0xa1, 0x25
2018-10-07 20:11:20.232 Detail,
2018-10-07 20:11:20.232 Info, Node028, Response RTT 182 Average Response RTT 179
2018-10-07 20:11:20.232 Info, Node028, Received a MultiChannelEncap from node 28, endpoint 2 for Command Class COMMAND_CLASS_SENSOR_MULTILEVEL
2018-10-07 20:11:20.232 Info, Node028, Received SensorMultiLevel report from node 28, instance 1, Temperature: value=16.1C
2018-10-07 20:11:20.233 Detail, Node028, Refreshed Value: old value=17.2, new value=16.1, type=decimal
2018-10-07 20:11:20.233 Detail, Node028, Changes to this value are not verified
2018-10-07 20:11:20.233 Detail, Node028, Expected reply and command class was received
2018-10-07 20:11:20.233 Detail, Node028, Message transaction complete
2018-10-07 20:11:20.233 Detail,
2018-10-07 20:11:20.233 Detail, Node028, Removing current message
2018-10-07 20:11:20.233 Detail, Node028, Notification: ValueChanged`

The device reports 16.1 to OZW... What do you expect OZW todo?

> The lack of update from the floor sensor is because OZW does not support the command classes the thermostat uses to report "live"

Which CC is that? I don't see anything in the logs that would indicate that.

follox, 2018-10-08:
@Fishwaldo You can't change any of the parameters in hass. It wont change anything.
The device report internal sensor for the thermostat, not the floor sensor temperature. This can't be changed either