OpenZWave/open-zwave #1635: GE 14287 Fan Control Switch - can't set LED to always off

raccettura, 2018-10-21:
Noticed that for this particular switch it's not possible to turn the LED light always off. As soon as the value is changed (tested in Home Assistant using the python wrapper and openzwave control panel) it immediately resets back to the light on when off state.

I was able to capture this:
```
post: key=50-CONFIGURATION-config-list-1-3 data=LED always off size=14
2018-10-21 01:58:42.533 Info, Node050, Value::Set - COMMAND_CLASS_CONFIGURATION - LED Light - 3 - 1 - LED always off
2018-10-21 01:58:42.533 Info, Node050, Configuration::Set - Parameter=3, Value=2 Size=1
2018-10-21 01:58:42.533 Detail, Node050, Queuing (Send) ConfigurationCmd_Set (Node=50): 0x01, 0x0c, 0x00, 0x13, 0x32, 0x05, 0x70, 0x04, 0x03, 0x01, 0x02, 0x25, 0x92, 0x14
2018-10-21 01:58:42.533 Detail, Node050, Queuing (Send) ConfigurationCmd_Get (Node=50): 0x01, 0x0a, 0x00, 0x13, 0x32, 0x03, 0x70, 0x05, 0x03, 0x25, 0x93, 0x17
2018-10-21 01:58:42.533 Detail, 
2018-10-21 01:58:42.533 Info, Node050, Sending (Send) message (Callback ID=0x92, Expected Reply=0x13) - ConfigurationCmd_Set (Node=50): 0x01, 0x0c, 0x00, 0x13, 0x32, 0x05, 0x70, 0x04, 0x03, 0x01, 0x02, 0x25, 0x92, 0x14
post: key=50-CONFIGURATION-config-list-1-3 data= size=0
2018-10-21 01:58:42.541 Detail, Node050,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2018-10-21 01:58:42.541 Detail, Node050,   ZW_SEND_DATA delivered to Z-Wave stack
2018-10-21 01:58:42.557 Detail, Node050,   Received: 0x01, 0x07, 0x00, 0x13, 0x92, 0x00, 0x00, 0x02, 0x7b
2018-10-21 01:58:42.557 Detail, Node050,   ZW_SEND_DATA Request with callback ID 0x92 received (expected 0x92)
2018-10-21 01:58:42.557 Info, Node050, Request RTT 24 Average Request RTT 23
2018-10-21 01:58:42.557 Detail,   Expected callbackId was received
2018-10-21 01:58:42.557 Detail,   Expected reply was received
2018-10-21 01:58:42.557 Detail,   Message transaction complete
2018-10-21 01:58:42.557 Detail, 
2018-10-21 01:58:42.557 Detail, Node050, Removing current message
2018-10-21 01:58:42.557 Detail, 
2018-10-21 01:58:42.557 Info, Node050, Sending (Send) message (Callback ID=0x93, Expected Reply=0x04) - ConfigurationCmd_Get (Node=50): 0x01, 0x0a, 0x00, 0x13, 0x32, 0x03, 0x70, 0x05, 0x03, 0x25, 0x93, 0x17
2018-10-21 01:58:42.566 Detail, Node050,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2018-10-21 01:58:42.566 Detail, Node050,   ZW_SEND_DATA delivered to Z-Wave stack
2018-10-21 01:58:42.582 Detail, Node050,   Received: 0x01, 0x07, 0x00, 0x13, 0x93, 0x00, 0x00, 0x02, 0x7a
2018-10-21 01:58:42.582 Detail, Node050,   ZW_SEND_DATA Request with callback ID 0x93 received (expected 0x93)
2018-10-21 01:58:42.582 Info, Node050, Request RTT 24 Average Request RTT 23
2018-10-21 01:58:42.582 Detail,   Expected callbackId was received
2018-10-21 01:58:42.593 Detail, Node050,   Received: 0x01, 0x0b, 0x00, 0x04, 0x00, 0x32, 0x05, 0x70, 0x06, 0x03, 0x01, 0x00, 0xb3
2018-10-21 01:58:42.593 Detail, 
2018-10-21 01:58:42.593 Info, Node050, Response RTT 36 Average Response RTT 35
2018-10-21 01:58:42.593 Detail, Node050, Refreshed Value: old value=0, new value=0, type=list
2018-10-21 01:58:42.593 Detail, Node050, Changes to this value are not verified
2018-10-21 01:58:42.593 Info, Node050, Received Configuration report: Parameter=3, Value=0
2018-10-21 01:58:42.593 Detail, Node050,   Expected reply and command class was received
2018-10-21 01:58:42.593 Detail, Node050,   Message transaction complete
2018-10-21 01:58:42.593 Detail, 
2018-10-21 01:58:42.593 Detail, Node050, Removing current message
2018-10-21 01:58:42.593 Detail, Node050, Notification: ValueChanged
2018-10-21 01:58:42.593 Info, Notification: Value Changed Home 0xce5f036c Node 50 Genre config Class CONFIGURATION Instance 1 Index 3 Type list
```

This seems to be the only field that is having this issue on this device.  It's also the only GE device I have with this issue. It updates immediately for any other dimmer or switch I have.

kpine, 2018-10-22:
This particular controller does not support disabling the LED via software. Your only choice is to use a hardware override. Tap the top button 3 times then the bottom button once.

The XML should probably should just remove this option from the config parameters.

Fishwaldo, 2018-10-22:
closing based on @kpine comment.

raccettura, 2018-10-22:
Got it. Thanks.  The config is a little misleading then.