OpenZWave/open-zwave #1639: Qubino 3-phase smart meter ZMNHXD1: Cannot find endpoint map to instance for Command Class COMMAND_CLASS_METER endpoint 1

hoschid, 2018-10-26:
Hi everybody!

I have a Qubino 3-phase smart meter **ZMNHXD1** together with the Qubino IR relay BICOM432-40-WM1. I try to use this device with **openzwave v1.5** and the npm module openzwave-shared v1.4.6 for node.js.

My openzwave device database has the files for the ZMNHXD1 from this commit: 9441bdc.

I did the following steps so far:
- add the device (with security enabled)
- set config parameter 100 to 1 to enable the IR relay
- remove the device
- add the device again (with security enabled)

Now I have the following problems: 

- For endpoint 1 (combined energy meter for all 3 phases) values were added for energy, power etc. but for the enpoints 2, 3, 4 (represent one energy meter for every of the 3 phases) and for endpoint 5 (represents the external relay) only the following values were added:
  - Command class 94, index 0: ZWave+ Version
  - Command class 94, index 1: InstallerIcon
  - Command class 94, index 2: UserIcon
  - Command class 152, index 0: Secured
but I'm missing some energy/power related values as in endpoint 1. And for endpoint 5 there should be a value to get the current state of the IR relay and toggle it.
- when I turn on a device connected to the meter to consume some energy it looks like the meter sends me some updates about the changed values but openzwave gives me the following errors:
**Error, Node003, Cannot find endpoint map to instance for Command Class COMMAND_CLASS_METER endpoint 1** (the same error also occures for endpoint 2)

I would really appreciate any help with this issue!

Here are the logs from adding the device:

```
2018-10-25T14:02:40.759Z zwave:routes:nodes:debug set controller to add node mode (security=true)
2018-10-25 16:02:40.759 Detail, Queuing (Controller) Add Device
2018-10-25 16:02:40.759 Info, Add Device
2018-10-25 16:02:40.759 Detail, contrlr, Queuing (Command) ControllerCommand_AddDevice: 0x01, 0x05, 0x00, 0x4a, 0xc1, 0x0a, 0x7b
2018-10-25 16:02:40.759 Detail, Notification: ControllerCommand - Starting
2018-10-25 16:02:40.760 Detail, 
2018-10-25 16:02:40.760 Info, contrlr, Sending (Command) message (Callback ID=0x0a, Expected Reply=0x4a) - ControllerCommand_AddDevice: 0x01, 0x05, 0x00, 0x4a, 0xc1, 0x0a, 0x7b
2018-10-25 16:02:40.761 Detail, contrlr,   Received: 0x01, 0x07, 0x00, 0x4a, 0x0a, 0x01, 0x00, 0x00, 0xb9
2018-10-25 16:02:40.762 Detail, 
2018-10-25 16:02:40.762 Info, contrlr, FUNC_ID_ZW_ADD_NODE_TO_NETWORK:
2018-10-25 16:02:40.762 Info, contrlr, ADD_NODE_STATUS_LEARN_READY
2018-10-25 16:02:40.762 Detail, Node001,   Expected callbackId was received
2018-10-25 16:02:40.762 Detail, Node001,   Expected reply was received
2018-10-25 16:02:40.762 Detail, Node001,   Message transaction complete
2018-10-25 16:02:40.762 Detail, 
2018-10-25 16:02:40.763 Detail, contrlr, Removing current message
2018-10-25 16:02:40.763 Detail, Notification: ControllerCommand - Waiting
2018-10-25 16:02:40.763 Info, WriteNextMsg Controller nothing to do
2018-10-25T14:02:40.765Z zwave:app:debug controller commmand feedback: msg=ControllerCommand - Starting, node=0, returnValue=1, state=0
2018-10-25T14:02:40.766Z zwave:app:debug controller commmand feedback: msg=ControllerCommand - Waiting, node=0, returnValue=4, state=0
2018-10-25 16:03:00.801 Detail,   Received: 0x01, 0x07, 0x00, 0x4a, 0x0a, 0x02, 0x00, 0x00, 0xba
2018-10-25 16:03:00.801 Detail, 
2018-10-25 16:03:00.801 Info, FUNC_ID_ZW_ADD_NODE_TO_NETWORK:
2018-10-25 16:03:00.801 Info, ADD_NODE_STATUS_NODE_FOUND
2018-10-25 16:03:00.802 Detail, Notification: ControllerCommand - InProgress
2018-10-25T14:03:00.802Z zwave:app:debug controller commmand feedback: msg=ControllerCommand - InProgress, node=0, returnValue=6, state=0
2018-10-25 16:03:01.154 Detail,   Received: 0x01, 0x1c, 0x00, 0x4a, 0x0a, 0x03, 0x03, 0x15, 0x04, 0x31, 0x03, 0x5e, 0x55, 0x5a, 0x73, 0x56, 0x98, 0x9f, 0x86, 0x72, 0x25, 0x70, 0x60, 0x85, 0x8e, 0x59, 0x32, 0x6c, 0x7a, 0x44
2018-10-25 16:03:01.155 Detail, 
2018-10-25 16:03:01.155 Info, FUNC_ID_ZW_ADD_NODE_TO_NETWORK:
2018-10-25 16:03:01.156 Info, ADD_NODE_STATUS_ADDING_SLAVE
2018-10-25 16:03:01.156 Info, Adding node ID 3 - Secure
2018-10-25 16:03:01.157 Detail, Notification: ControllerCommand - InProgress
2018-10-25T14:03:01.157Z zwave:app:debug controller commmand feedback: msg=ControllerCommand - InProgress, node=0, returnValue=6, state=0
2018-10-25 16:03:01.440 Detail,   Received: 0x01, 0x07, 0x00, 0x4a, 0x0a, 0x05, 0x03, 0x00, 0xbe
2018-10-25 16:03:01.441 Detail, 
2018-10-25 16:03:01.441 Info, FUNC_ID_ZW_ADD_NODE_TO_NETWORK:
2018-10-25 16:03:01.442 Info, ADD_NODE_STATUS_PROTOCOL_DONE
2018-10-25 16:03:01.442 Detail, contrlr, Queuing (Command) Add Node Stop: 0x01, 0x04, 0x00, 0x4a, 0x05, 0xb4
2018-10-25 16:03:01.443 Detail, Notification: ControllerCommand - InProgress
2018-10-25 16:03:01.443 Detail, 
2018-10-25 16:03:01.443 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x4a) - Add Node Stop: 0x01, 0x04, 0x00, 0x4a, 0x05, 0xb4
2018-10-25 16:03:01.495 Detail, contrlr,   Received: 0x01, 0x07, 0x00, 0x4a, 0x0a, 0x06, 0x03, 0x00, 0xbd
[... 1700 lines cut ...]
2018-10-26 10:36:13.407 Info, Raw: 0x98, 0x81, 0x5b, 0xc1, 0x12, 0x18, 0x12, 0xc5, 0xf8, 0x6a, 0x1c, 0x28, 0x62, 0x58, 0xf5, 0xbc, 0xf0, 0x9c, 0xab, 0xcb, 0xd3, 0x06, 0x83, 0x80, 0x54, 0xb0, 0x17, 0x23, 0x27, 0x51, 0xfa, 0x06, 0xe1, 0x5d, 0x45
2018-10-26 10:36:13.407 Detail, Node003, Decrypted Packet: 0x00, 0x60, 0x0d, 0x01, 0x01, 0x32, 0x02, 0x21, 0x34, 0x00, 0x00, 0x01, 0x9b, 0x00, 0x00
2018-10-26 10:36:13.407 Detail, 
2018-10-26 10:36:13.408 Error, Node003, Cannot find endpoint map to instance for Command Class COMMAND_CLASS_METER endpoint 1
2018-10-26 10:36:18.959 Detail, Node003,   Received: 0x01, 0x08, 0x00, 0x04, 0x00, 0x03, 0x02, 0x98, 0x40, 0x2a
2018-10-26 10:36:18.959 Info, Node003, Received SecurityCmd_NonceGet from node 3
2018-10-26 10:36:18.959 Info, NONCES: 0x2d, 0x3d, 0xcc, 0xbb, 0xa8, 0xf7, 0xa3, 0xc2
2018-10-26 10:36:18.959 Info, NONCES: 0x18, 0x22, 0x85, 0x14, 0x11, 0x34, 0x76, 0xd1
2018-10-26 10:36:18.959 Info, NONCES: 0x93, 0xc1, 0x0d, 0x28, 0xff, 0x34, 0xe3, 0x20
2018-10-26 10:36:18.960 Info, NONCES: 0xff, 0x0d, 0xde, 0x12, 0x01, 0xec, 0x98, 0x2e
2018-10-26 10:36:18.960 Info, NONCES: 0x2a, 0x64, 0xe9, 0xd1, 0x5b, 0x8d, 0x94, 0x73
2018-10-26 10:36:18.960 Info, NONCES: 0xb0, 0x19, 0x87, 0xc1, 0x4d, 0xfe, 0x93, 0xe0
2018-10-26 10:36:18.960 Info, NONCES: 0xbf, 0xa1, 0x09, 0xbf, 0xd5, 0xec, 0xdf, 0xd4
2018-10-26 10:36:18.960 Info, NONCES: 0xc0, 0x5e, 0x4b, 0x3b, 0x95, 0x3e, 0x27, 0xbb
2018-10-26 10:36:18.961 Info, Node003, Sending (Send) message (Callback ID=0x01, Expected Reply=0x00) - Nonce_Report - 0x01, 0x11, 0x00, 0x13, 0x03, 0x0a, 0x98, 0x80, 0xbf, 0xa1, 0x09, 0xbf, 0xd5, 0xec, 0xdf, 0xd4, 0x05, 0x01, 0x72:
2018-10-26 10:36:18.966 Detail,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2018-10-26 10:36:18.967 Detail,   ZW_SEND_DATA delivered to Z-Wave stack
2018-10-26 10:36:18.982 Detail,   Received: 0x01, 0x07, 0x00, 0x13, 0x01, 0x00, 0x00, 0x02, 0xe8
2018-10-26 10:36:18.982 Detail,   ZW_SEND_DATA Request with callback ID 0x01 received (expected 0x01)
2018-10-26 10:36:19.006 Detail, Node003,   Received: 0x01, 0x28, 0x00, 0x04, 0x00, 0x03, 0x22, 0x98, 0x81, 0x4f, 0x81, 0xf8, 0x70, 0xe4, 0xdd, 0xa7, 0x39, 0x3c, 0xe6, 0x9d, 0xf6, 0xfb, 0xf3, 0x70, 0x3a, 0x46, 0xc3, 0x27, 0xed, 0x2f, 0x58, 0x12, 0xbf, 0x53, 0x0a, 0xa3, 0xe0, 0x73, 0x62, 0x5e, 0xae, 0x97
2018-10-26 10:36:19.007 Info, Raw: 0x98, 0x81, 0x4f, 0x81, 0xf8, 0x70, 0xe4, 0xdd, 0xa7, 0x39, 0x3c, 0xe6, 0x9d, 0xf6, 0xfb, 0xf3, 0x70, 0x3a, 0x46, 0xc3, 0x27, 0xed, 0x2f, 0x58, 0x12, 0xbf, 0x53, 0x0a, 0xa3, 0xe0, 0x73, 0x62, 0x5e, 0xae, 0x97
2018-10-26 10:36:19.007 Detail, Node003, Decrypted Packet: 0x00, 0x60, 0x0d, 0x02, 0x01, 0x32, 0x02, 0x21, 0x34, 0x00, 0x00, 0x02, 0x79, 0x00, 0x00
2018-10-26 10:36:19.008 Detail, 
2018-10-26 10:36:19.008 Error, Node003, Cannot find endpoint map to instance for Command Class COMMAND_CLASS_METER endpoint 2
```

Fishwaldo, 2018-10-26:
> 2018-10-25 16:03:12.600 Info, Node003, Received MultiChannelCapabilityReport from node 3 for endpoint 1
2018-10-25 16:03:12.600 Info, Node003,     Endpoint is not dynamic, and is a Whole Home Meter Simple
2018-10-25 16:03:12.600 Info, Node003,     Command classes supported by the endpoint are:
2018-10-25 16:03:12.600 Info, Node003,         COMMAND_CLASS_ZWAVE_PLUS_INFO
2018-10-25 16:03:12.601 Info, Node003,         COMMAND_CLASS_SECURITY
2018-10-25 16:03:12.601 Info, Node003, AddCommandClass - Unsupported Command Class 0x9f
2018-10-25 16:03:12.601 Info, Node003, AddCommandClass - Unsupported Command Class 0x6c

The device only advertises ZWave+, Security S0 (Supported), Security S2 (Not Supported) and Supervision CC on Endpoint 1 but:

> 2018-10-26 10:36:08.608 Detail, Node003,   Received: 0x01, 0x28, 0x00, 0x04, 0x00, 0x03, 0x22, 0x98, 0x81, 0xab, 0x08, 0x51, 0x83, 0x8e, 0x8b, 0x0e, 0x70, 0x86, 0x5a, 0x80, 0x56, 0xa9, 0xd8, 0xf9, [...]
2018-10-26 10:36:08.608 Info, Raw: 0x98, 0x81, 0xab, 0x08, 0x51, 0x83, 0x8e, 0x8b, 0x0e, 0x70, 0x86, 0x5a, 0x80, 0x56, 0xa9, 0xd8, 0xf9, 0x95, 0x35, 0x75, 0x3a, 0xcf, 0x63, 0x67, 0xc4, 0xff, 0x05, 0xe9, 0x53, 0x7d, 0x62, 0xe6, 0xf7, 0x13, 0xde
2018-10-26 10:36:08.609 Detail, Node003, Decrypted Packet: 0x00, 0x60, 0x0d, 0x01, 0x01, 0x32, 0x02, 0x21, 0x34, 0x00, 0x00, 0x01, 0x00, 0x00, 0x00
2018-10-26 10:36:08.609 Detail, 
2018-10-26 10:36:08.610 Error, Node003, Cannot find endpoint map to instance for Command Class COMMAND_CLASS_METER endpoint 1

After reading the specs - It seems we should also do a SECURITY_CAPABILITIES_GET on each endpoint if S0/S2 is there. OZW doesn't do that currently. (and I have to read the specs a bit more to understand this).

I would suggest trying to include this device non-secure and see if it works.

Fishwaldo, 2018-11-13:
read 3.29.3.1 - Multichannel handling. 

it would be SECURITY_CAPABILITIES_GET -> MultiChannel Encap Message -> S0/S2 encrypted. Not the other way around.

Fishwaldo, 2018-11-13:
> I always get exactly the same messages with the same received bytes, no matter if the relay is on or off. I also asked the qubino support about this. First they told me that this is probably caused by my software respectively openzwave but now they will check if it's a bug of the meter. For me dealing with the support it would be helpful to know what you think about this.

Regardless of the Endpoint - If its always sending a "off" command even if you are turning it on, thats a bug. Unless that endpoint is representing some other state of the device (but it doesn't seem like it).

Fishwaldo, 2018-11-13:
> 

> read 3.29.3.1 - Multichannel handling.
> 
> it would be SECURITY_CAPABILITIES_GET -> MultiChannel Encap Message -> S0/S2 encrypted. Not the other way around.

Sorry, I mean SECURITY_COMMANDS_SUPPORTED_GET

aromanro, 2018-11-13:
I tried sending that to an individual channel (for a multi-channel that supports security) and it did not reply. My guess is that you should ask for multi-channel capabilities in an encrypted packet and you should get all the supported classes there, in the report. I don't think that sending SECURITY_COMMANDS_SUPPORTED_GET to individual channels is allowed.

Fishwaldo, 2018-11-13:
> I tried sending that to an individual channel (for a multi-channel that supports security) and it did not reply. 

When reading the specs it says this is the way to do that, so either the device is buggy, or you got it wrong:
The specs say:
`An End Point with support for Security MUST report the Security Command Class as supported for that End Point. The command classes supported for each endpoint securely is determined by using the Security Commands Supported Get command sent to each individual endpoint Security Encapsulated`

>My guess is that you should ask for multi-channel capabilities in an encrypted packet and you should get all the supported classes there, in the report. I don't think that sending SECURITY_COMMANDS_SUPPORTED_GET to individual channels is allowed.

We already do that:

> 2018-10-25 16:03:12.468 Info, Node003, MultiChannelCmd_CapabilityGet for endpoint 1
2018-10-25 16:03:12.469 Detail, Node003, Setting Encryption Flag on Message For Command Class COMMAND_CLASS_MULTI_INSTANCE/CHANNEL
2018-10-25 16:03:12.469 Detail, Node003, Queuing (Send) MultiChannelCmd_CapabilityGet (Node=3): 0x01, 0x0a, 0x00, 0x13, 0x03, 0x03, 0x60, 0x09, 0x01, 0x25, 0x1b, 0xb0
 
and it replies:
>2018-10-25 16:03:12.600 Info, Node003, Received MultiChannelCapabilityReport from node 3 for endpoint 1
2018-10-25 16:03:12.600 Info, Node003,     Endpoint is not dynamic, and is a Whole Home Meter Simple
2018-10-25 16:03:12.600 Info, Node003,     Command classes supported by the endpoint are:
2018-10-25 16:03:12.600 Info, Node003,         COMMAND_CLASS_ZWAVE_PLUS_INFO
2018-10-25 16:03:12.601 Info, Node003,         COMMAND_CLASS_SECURITY
2018-10-25 16:03:12.601 Info, Node003, AddCommandClass - Unsupported Command Class 0x9f
2018-10-25 16:03:12.601 Info, Node003, AddCommandClass - Unsupported Command Class 0x6c

so its only exposing the Security CC, none of the other CC's that is needed. 

But this may not be the only issue. I have to check the code path, to make sure when we try to comunicate with a Security CC on a endpoint other than the root, we properly encapsulate it in the MultiChannel and then encrypt. So determining what CC's we need to encrypt on a endpoint might only be half the problem.

aromanro, 2018-11-13:
Maybe I've got it wrong, I'll try again tomorrow and I'll let you know if it works. I'm not using open zwave, but my own code.

aromanro, 2018-11-14:
I've got it wrong. The multi-channel device I have that is supporting S0 security, does not have the security class as supported for channels. So I must not send SECURITY_COMMANDS_SUPPORTED_GET to them. I would need a device that has 0x98 as a supported class for channels in order to test it...

hoschid, 2018-11-26:
@Fishwaldo could you already organize this smart meter for testing? I'm still in contact with the qubino support about the wrong relay state sent by the meter. If you wan't I could ask them if they could send you a device for testing.

Fishwaldo, 2018-12-04:
I just got another device (Aeotec Doorbell 6) that supports the Security CC on endpoints. So I'll do some work on this soon.

ruudbeukema, 2019-01-21:
@Fishwaldo, I'm encountering the same behavior as described by @hoschid during my current multi-channel (3-endpoints) Z-Wave device.

I am capable of reading out the command classes supported by each of the three endpoints in S0 and S2 security mode by using Silicon Labs' PC Controller software.

If you like me to test things, let me know. It would be great to have this functionality implemented in OZW!

Fishwaldo, 2019-01-23:
@ruudbeukema @hoschid 
Could you checkout the Dev Branch, include your Device Securely, restart and send me logs?

(if needed, you can use your app to include it securely, then copy your existing options.xml file to the config directory for the Dev Branch you checked out, and run MinOZW for convenience... )

I suspect I have a buggy device... (it has a lot of other issues) and not responding:

> 2019-01-23 15:03:24.880 Info, Plain Text Packet:: 0x00, 0x60, 0x0d, 0x01, 0x08, 0x98, 0x02
2019-01-23 15:03:24.880 Info, Encrypted Packet: 0x95, 0x2e, 0x41, 0x56, 0x74, 0x9f, 0x63
2019-01-23 15:03:24.880 Info, Raw Auth (minus IV): 0x81, 0x01, 0x1a, 0x07, 0x95, 0x2e, 0x41, 0x56, 0x74, 0x9f, 0x63
2019-01-23 15:03:24.880 Info, Computed Auth: 0x0d, 0xdc, 0x28, 0xe7, 0x23, 0x3f, 0x9f, 0xbf
2019-01-23 15:03:24.880 Info, Encrypted Packet: 0x01, 0x21, 0x00, 0x13, 0x1a, 0x1a, 0x98, 0x81, 0x20, 0x32, 0x51, 0xa0, 0x20, 0xa6, 0x9e, 0xcc, 0x95, 0x2e, 0x41, 0x56, 0x74, 0x9f, 0x63, 0xbb, 0x0d, 0xdc, 0x28, 0xe7, 0x23, 0x3f, 0x9f, 0xbf, 0x25, 0x2b, 0x50
2019-01-23 15:03:24.880 Info, Length: 35
2019-01-23 15:03:24.880 Info, Node026, Sending (Command) message (Callback ID=0x2b, Expected Reply=0x04) - MultiChannel Encapsulated (instance=9): SecurityCmd_SupportedGet (Node=26): 0x01, 0x0d, 0x00, 0x13, 0x1a, 0x06, 0x60, 0x0d, 0x01, 0x08, 0x98, 0x02, 0x25, 0x2b, 0x0d
2019-01-23 15:03:24.887 Detail, Node026,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2019-01-23 15:03:24.888 Detail, Node026,   ZW_SEND_DATA delivered to Z-Wave stack
2019-01-23 15:03:24.904 Detail, Node026,   Received: 0x01, 0x07, 0x00, 0x13, 0x2b, 0x00, 0x00, 0x02, 0xc2
2019-01-23 15:03:24.905 Detail, Node026,   ZW_SEND_DATA Request with callback ID 0x2b received (expected 0x2b)
2019-01-23 15:03:24.905 Info, Node026, Request RTT 60 Average Request RTT 46
2019-01-23 15:03:24.905 Detail,   Expected callbackId was received
Failed to send data in request for `/poll.xml'.
2019-01-23 15:03:34.845 Error, Node026, ERROR: Dropping command, expected response not received after 1 attempt(s)

ruudbeukema, 2019-01-23:
Hi @Fishwaldo,

Please find the MinOZW-log at the bottom of this comment. I've issued several unsolicited commands on my Z-Wave slave device from the different endpoints. These seem to get processed correctly now.

I do notice that my Z-Wave device doesn't seem to send a _SecurityCommandsSupportedReport_ when a _SecurityCommandsSupportedGet_ is issued on an endpoint. I will look into that!

`
Starting MinOZW with OpenZWave Version 1.5.3533
2019-01-23 14:47:26.313 Error, Cannot find a path to the configuration files at ../../../config/, Using config/ instead...
2019-01-23 14:47:26.313 Info, Reading config/options.xml for Options
2019-01-23 14:47:26.313 Warning, Failed to Parse options.xml: Failed to open file
2019-01-23 14:47:26.313 Info, Options:
2019-01-23 14:47:26.313 Info, 	appendlogfile: false
2019-01-23 14:47:26.313 Info, 	associate: true
2019-01-23 14:47:26.313 Info, 	assumeawake: true
2019-01-23 14:47:26.313 Info, 	autoupdateconfigfile: true
2019-01-23 14:47:26.313 Info, 	configpath: config/
2019-01-23 14:47:26.313 Info, 	consoleoutput: true
2019-01-23 14:47:26.313 Info, 	customsecuredcc: 0x62,0x4c,0x63
2019-01-23 14:47:26.313 Info, 	drivermaxattempts: 0
2019-01-23 14:47:26.313 Info, 	dumptrigger: 4
2019-01-23 14:47:26.313 Info, 	dumptriggerlevel: 1
2019-01-23 14:47:26.313 Info, 	enablesis: true
2019-01-23 14:47:26.313 Info, 	enforcesecurereception: true
2019-01-23 14:47:26.313 Info, 	exclude: 
2019-01-23 14:47:26.313 Info, 	include: 
2019-01-23 14:47:26.313 Info, 	interface: 
2019-01-23 14:47:26.313 Info, 	intervalbetweenpolls: true
2019-01-23 14:47:26.313 Info, 	language: 
2019-01-23 14:47:26.313 Info, 	logfilename: OZW_Log.txt
2019-01-23 14:47:26.313 Info, 	logging: true
2019-01-23 14:47:26.313 Info, 	networkkey: 0x01, 0x02, 0x03, 0x04, 0x05, 0x06, 0x07, 0x08, 0x09, 0x0A, 0x0B, 0x0C, 0x0D, 0x0E, 0x0F, 0x10
2019-01-23 14:47:26.314 Info, 	notifyondriverunload: false
2019-01-23 14:47:26.314 Info, 	notifytransactions: false
2019-01-23 14:47:26.314 Info, 	performreturnroutes: true
2019-01-23 14:47:26.314 Info, 	pollinterval: 500
2019-01-23 14:47:26.314 Info, 	queueloglevel: 9
2019-01-23 14:47:26.314 Info, 	refreshallusercodes: false
2019-01-23 14:47:26.314 Info, 	reloadafterupdate: AWAKE
2019-01-23 14:47:26.314 Info, 	retrytimeout: 10000
2019-01-23 14:47:26.314 Info, 	saveconfiguration: true
2019-01-23 14:47:26.314 Info, 	saveloglevel: 9
2019-01-23 14:47:26.314 Info, 	securitystrategy: SUPPORTED
2019-01-23 14:47:26.314 Info, 	suppressvaluerefresh: false
2019-01-23 14:47:26.314 Info, 	userpath: ./open-zwave
2019-01-23 14:47:26.314 Info, 	validatevaluechanges: true
2019-01-23 14:47:26.314 Always, OpenZwave Version 1.5.3533 Starting Up
2019-01-23 14:47:26.323 Info, Loading Localization File config/Localization.xml
2019-01-23 14:47:26.324 Info, Loaded config/Localization.xml With Revision 4
2019-01-23 14:47:26.324 Always, Using Language Localization 
2019-01-23 14:47:26.326 Info, Loading NotificationCCTypes File config/NotificationCCTypes.xml
2019-01-23 14:47:26.326 Info, Loaded config/NotificationCCTypes.xml With Revision 6
2019-01-23 14:47:26.326 Info, Setting Up Provided Network Key for Secure Communications
2019-01-23 14:47:26.334 Info, Manufacturer_Specific.xml file Revision is 15
2019-01-23 14:47:26.578 Info, Product name collision: VRF01-1LZ Multilevel Scene Switch - 1.5A Fan type 1001 id 209 manufacturerid 1d, collides with VRF01-1LZ Quiet Fan Speed Control, type 1001 id 209 manufacturerid 1d
2019-01-23 14:47:26.578 Info, Product name collision: VRCS2-MRZ 2-Button Scene Controller with Switches type 1102 id 243 manufacturerid 1d, collides with VRCS2-MRZ 2-Button Scene Controller with Switches, type 1102 id 243 manufacturerid 1d
2019-01-23 14:47:26.578 Info, Product name collision: DZPD3-1LW Plug-In Dimming Lamp Module type 1902 id 334 manufacturerid 1d, collides with DZPD3-1LW Plug-In Lamp Module, type 1902 id 334 manufacturerid 1d
2019-01-23 14:47:26.726 Info, Queuing Lookup on mfs.db.openzwave.com for Node 0
2019-01-23 14:47:26.727 Info,   Opening controller /dev/ttyACM0
2019-01-23 14:47:26.727 Info, Trying to open serial port /dev/ttyACM0 (attempt 1)
2019-01-23 14:47:26.727 Info, mgr,     Added driver for controller /dev/ttyACM0
2019-01-23 14:47:26.727 Info, Starting DNSThread
2019-01-23 14:47:26.727 Info, LookupTxT Checking mfs.db.openzwave.com
2019-01-23 14:47:26.728 Info, Timer: thread starting
2019-01-23 14:47:26.728 Detail, Timer: waiting with timeout -1 ms
2019-01-23 14:47:26.728 Info, Lookup for mfs.db.openzwave.com returned 15
2019-01-23 14:47:26.729 Info, Serial port /dev/ttyACM0 opened (attempt 1)
2019-01-23 14:47:26.729 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9
2019-01-23 14:47:26.730 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_MEMORY_GET_ID: 0x01, 0x03, 0x00, 0x20, 0xdc
2019-01-23 14:47:26.730 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES: 0x01, 0x03, 0x00, 0x05, 0xf9
[... 1971 lines cut ...]
2019-01-23 14:51:10.970 Info, mgr,     Driver for controller /dev/ttyACM0 pending removal
2019-01-23 14:51:10.970 Detail, Notification: DriverRemoved
Notification: DriverRemoved
2019-01-23 14:51:10.970 Always, ***************************************************************************
2019-01-23 14:51:10.970 Always, *********************  Cumulative Network Statistics  *********************
2019-01-23 14:51:10.970 Always, *** General
[... cut ...]
2019-01-23 14:51:10.970 Always, Out of frame data flow errors:  . . . . . . . . . . . . . 0
2019-01-23 14:51:10.970 Always, Messages retransmitted: . . . . . . . . . . . . . . . . . 0
2019-01-23 14:51:10.970 Always, Messages dropped and not delivered: . . . . . . . . . . . 4
2019-01-23 14:51:10.970 Always, ***************************************************************************
2019-01-23 14:51:10.972 Info, Stopping DNSThread
2019-01-23 14:51:12.987 Info, mgr,     Driver for controller /dev/ttyACM0 removed
`

Fishwaldo, 2019-01-24:
> 2019-01-23 16:24:31.258 Detail, Node033, Received: 0x01, 0x27, 0x00, 0x04, 0x00, 0x21, 0x1f, 0x98, 0x81, 0xab, 0x12, 0x01, 0xff, 0x3d, 0xf5, 0xb3, 0x5b, 0xa4, 0x2c, 0x52, 0x9e, 0x2e, 0xd2, 0xc1, 0x0e, 0x62, 0xc2, 0xf3, 0x11, 0x26, 0x0d, 0xc8, 0x3e, 0x96, 0x23, 0x19, 0xb5, 0x6e, 0xd5, 0x00, 0xd6
2019-01-23 16:24:31.258 Info, Raw: 0x98, 0x81, 0xab, 0x12, 0x01, 0xff, 0x3d, 0xf5, 0xb3, 0x5b, 0xa4, 0x2c, 0x52, 0x9e, 0x2e, 0xd2, 0xc1, 0x0e, 0x62, 0xc2, 0xf3, 0x11, 0x26, 0x0d, 0xc8, 0x3e, 0x96, 0x23, 0x19, 0xb5, 0x6e, 0xd5
2019-01-23 16:24:31.258 Detail, Node033, Decrypted Packet: 0x00, 0x60, 0x0d, 0x01, 0x01, 0x98, 0x03, 0x00, 0x32, 0x85, 0x59, 0x8e
2019-01-23 16:24:31.259 Detail,
2019-01-23 16:24:31.259 Info, Node033, Response RTT 160 Average Response RTT 158
2019-01-23 16:24:31.259 Info, Node033, Received a MultiChannelEncap from node 33, endpoint 1 for Command Class COMMAND_CLASS_SECURITY
2019-01-23 16:24:31.259 Info, Node033, Received SecurityCmd_SupportedReport from node 33
2019-01-23 16:24:31.259 Detail, Node033, Initial read of value
2019-01-23 16:24:31.259 Info, Node033, Secured command classes for node 33:
2019-01-23 16:24:31.259 Info, Node033, COMMAND_CLASS_METER (Secured) - NotInNIF
2019-01-23 16:24:31.259 Info, Node033, COMMAND_CLASS_ASSOCIATION (Secured) - NotInNIF
2019-01-23 16:24:31.259 Info, Node033, Secure CommandClass 0x59 - NOT SUPPORTED
2019-01-23 16:24:31.259 Info, Node033, COMMAND_CLASS_MULTI_CHANNEL_ASSOCIATION (Secured) - NotInNIF
2019-01-23 16:24:31.259 Info, Node033, Secure CommandClass 0x2c - NOT SUPPORTED
2019-01-23 16:24:31.259 Info, Node033, UnSecured command classes for node 33:
2019-01-23 16:24:31.259 Info, Node033, COMMAND_CLASS_NO_OPERATION (Unsecured) - NotInNIF
2019-01-23 16:24:31.259 Info, Node033, COMMAND_CLASS_BASIC (Unsecured) - NotInNIF
2019-01-23 16:24:31.259 Info, Node033, COMMAND_CLASS_CRC_16_ENCAP (Unsecured) - NotInNIF
2019-01-23 16:24:31.259 Info, Node033, COMMAND_CLASS_ZWAVE_PLUS_INFO (Unsecured) - InNIF
2019-01-23 16:24:31.259 Info, Node033, COMMAND_CLASS_SECURITY (Unsecured) - InNIF
2019-01-23 16:24:31.259 Detail, Node033, Expected reply and command class was received
2019-01-23 16:24:31.259 Detail, Node033, Message transaction complete
2019-01-23 16:24:31.259 Detail,
2019-01-23 16:24:31.259 Detail, Node033, Removing current message

Perfect. So this confirms my device is bugged out. And confirms my initial code is working.

Question - Are the endpoints on this device now working correctly? (eg, try manually refreshing a Meter Reading on one of the endpoints). I can't test that CC's Secured on a endpoint are working with my device right now.

ruudbeukema, 2019-01-28:
> Question - Are the endpoints on this device now working correctly? (eg, try manually refreshing a Meter Reading on one of the endpoints). I can't test that CC's Secured on a endpoint are working with my device right now.

@Fishwaldo I will get back to you on this later, due to other priorities... Maybe @hoschid can verify that it works in the meantime...

vincemic, 2019-02-01:
@hoschid can you share your ozwcache.xml file? I'd like to see how your instances and endpoints mapped out for the ZMNHXD1.

hoschid, 2019-02-01:
I have no access to my device until middle of February. Then I will test it with the latest version as soon as possible.

hoschid, 2019-02-24:
@Fishwaldo @vincemic 

I installed openzwave from the latest commit e991af5 of the dev branch. Then I removed the smart meter, stopped ozw, deleted the ozwcache xml file, started ozw and re-included the device with security enabled.

It looks like the device gets added without errors but it lists only the values of the "main" endpoint. When power is consumed, the power value of the main endpoint updates and I don't get the errors from my original post anymore.

Here are the logs from adding the device:

```
2019-02-24 10:19:58.927 Detail, Queuing (Controller) Add Device
2019-02-24 10:19:58.927 Info, Add Device
2019-02-24 10:19:58.927 Detail, contrlr, Queuing (Command) ControllerCommand_AddDevice: 0x01, 0x05, 0x00, 0x4a, 0xc1, 0xa5, 0xd4
2019-02-24 10:19:58.927 Detail, Notification: AddDevice ControllerCommand - Starting
2019-02-24 10:19:58.928 Detail, 
2019-02-24 10:19:58.928 Info, contrlr, Sending (Command) message (Callback ID=0xa5, Expected Reply=0x4a) - ControllerCommand_AddDevice: 0x01, 0x05, 0x00, 0x4a, 0xc1, 0xa5, 0xd4
2019-02-24 10:19:58.928 Info, contrlr, Encrypted Flag is 0
2019-02-24 10:19:58.930 Detail, contrlr,   Received: 0x01, 0x07, 0x00, 0x4a, 0xa5, 0x01, 0x00, 0x00, 0x16
2019-02-24 10:19:58.930 Detail, 
2019-02-24 10:19:58.930 Info, contrlr, FUNC_ID_ZW_ADD_NODE_TO_NETWORK:
2019-02-24 10:19:58.930 Info, contrlr, ADD_NODE_STATUS_LEARN_READY
2019-02-24 10:19:58.931 Detail, Node001,   Expected callbackId was received
2019-02-24 10:19:58.931 Detail, Node001,   Expected reply was received
2019-02-24 10:19:58.931 Detail, Node001,   Message transaction complete
2019-02-24 10:19:58.931 Detail, 
2019-02-24 10:19:58.931 Detail, contrlr, Removing current message
2019-02-24 10:19:58.931 Detail, Notification: AddDevice ControllerCommand - Waiting
2019-02-24 10:19:58.931 Info, WriteNextMsg Controller nothing to do
2019-02-24T09:19:58.932Z zwave:app:debug controller commmand feedback: msg=AddDevice ControllerCommand - Starting, node=0, returnValue=1, state=0
2019-02-24T09:19:58.932Z zwave:app:debug controller commmand feedback: msg=AddDevice ControllerCommand - Waiting, node=0, returnValue=4, state=0
2019-02-24 10:20:13.441 Detail,   Received: 0x01, 0x07, 0x00, 0x4a, 0xa5, 0x02, 0x00, 0x00, 0x15
2019-02-24 10:20:13.441 Detail, 
2019-02-24 10:20:13.441 Info, FUNC_ID_ZW_ADD_NODE_TO_NETWORK:
2019-02-24 10:20:13.442 Info, ADD_NODE_STATUS_NODE_FOUND
2019-02-24 10:20:13.442 Detail, Notification: AddDevice ControllerCommand - InProgress
2019-02-24T09:20:13.442Z zwave:app:debug controller commmand feedback: msg=AddDevice ControllerCommand - InProgress, node=0, returnValue=6, state=0
2019-02-24 10:20:13.648 Detail,   Received: 0x01, 0x1c, 0x00, 0x4a, 0xa5, 0x03, 0x14, 0x15, 0x04, 0x31, 0x03, 0x5e, 0x55, 0x5a, 0x73, 0x56, 0x98, 0x9f, 0x86, 0x72, 0x25, 0x70, 0x60, 0x85, 0x8e, 0x59, 0x32, 0x6c, 0x7a, 0xfc
2019-02-24 10:20:13.648 Detail, 
2019-02-24 10:20:13.648 Info, FUNC_ID_ZW_ADD_NODE_TO_NETWORK:
2019-02-24 10:20:13.648 Info, ADD_NODE_STATUS_ADDING_SLAVE
2019-02-24 10:20:13.648 Info, Adding node ID 20 - Secure
2019-02-24 10:20:13.648 Detail, Notification: AddDevice ControllerCommand - InProgress
2019-02-24T09:20:13.648Z zwave:app:debug controller commmand feedback: msg=AddDevice ControllerCommand - InProgress, node=0, returnValue=6, state=0
2019-02-24 10:20:14.675 Detail,   Received: 0x01, 0x07, 0x00, 0x4a, 0xa5, 0x05, 0x14, 0x00, 0x06
2019-02-24 10:20:14.675 Detail, 
2019-02-24 10:20:14.675 Info, FUNC_ID_ZW_ADD_NODE_TO_NETWORK:
2019-02-24 10:20:14.675 Info, ADD_NODE_STATUS_PROTOCOL_DONE
2019-02-24 10:20:14.676 Detail, contrlr, Queuing (Command) Add Node Stop: 0x01, 0x04, 0x00, 0x4a, 0x05, 0xb4
2019-02-24 10:20:14.676 Detail, Notification: AddDevice ControllerCommand - InProgress
2019-02-24 10:20:14.676 Detail, 
2019-02-24T09:20:14.676Z zwave:app:debug controller commmand feedback: msg=AddDevice ControllerCommand - InProgress, node=0, returnValue=6, state=0
2019-02-24 10:20:14.676 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x4a) - Add Node Stop: 0x01, 0x04, 0x00, 0x4a, 0x05, 0xb4
2019-02-24 10:20:14.676 Info, contrlr, Encrypted Flag is 0
2019-02-24 10:20:14.728 Detail, contrlr,   Received: 0x01, 0x07, 0x00, 0x4a, 0xa5, 0x06, 0x14, 0x00, 0x05
2019-02-24 10:20:14.728 Detail, 
2019-02-24 10:20:14.728 Info, contrlr, FUNC_ID_ZW_ADD_NODE_TO_NETWORK:
2019-02-24 10:20:14.729 Info, contrlr, ADD_NODE_STATUS_DONE
2019-02-24 10:20:14.729 Info, Node020,   Basic device class    (0x04) - Routing Slave
2019-02-24 10:20:14.729 Info, Node020,   Generic device Class  (0x31) - Meter
2019-02-24 10:20:14.729 Info, Node020,   Specific device class (0x03) - Whole Home Meter Simple
2019-02-24 10:20:14.729 Info, Node020,     COMMAND_CLASS_BASIC is not mapped
2019-02-24 10:20:14.729 Info, Node020,   Mandatory Command Classes for Node 20:
2019-02-24 10:20:14.729 Info, Node020,     COMMAND_CLASS_MANUFACTURER_SPECIFIC
2019-02-24 10:20:14.729 Info, Node020,   Mandatory Command Classes controlled by Node 20:
2019-02-24 10:20:14.730 Info, Node020,     COMMAND_CLASS_BASIC
2019-02-24 10:20:14.730 Detail, Node020, Queuing (Command) SecurityCmd_SchemeGet (Node=20): 0x01, 0x0a, 0x00, 0x13, 0x14, 0x03, 0x98, 0x04, 0x00, 0x25, 0xa6, 0xee
2019-02-24 10:20:14.730 Info, Node020,   Optional command classes for node 20:
2019-02-24 10:20:14.730 Info, Node020,     COMMAND_CLASS_ZWAVE_PLUS_INFO
2019-02-24 10:20:14.730 Info, Node020,   CommandClass 0x55 - NOT REQUIRED
[... cut ...]
2019-02-24 10:20:30.719 Info, Node020, Processing (Send) Nonce Request message (Callback ID=0xea, Expected Reply=0x04)
2019-02-24 10:20:30.719 Info, Node020, Sending (Send) message (Callback ID=0x02, Expected Reply=0x04) - Nonce_Get(MeterCmd_Get) - 0x01, 0x09, 0x00, 0x13, 0x14, 0x02, 0x98, 0x40, 0x05, 0x02:
2019-02-24 10:20:30.726 Detail, Node020,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2019-02-24 10:20:30.726 Detail, Node020,   ZW_SEND_DATA delivered to Z-Wave stack
2019-02-24 10:20:30.742 Detail, Node020,   Received:

Fishwaldo, 2019-03-26:
I've just pushed a update to the Dev branch that should address this. Testing welcome!

(in particular, testing the different instances/endpoints are mapped correctly).

Obviously you need to include the device Securely for this code.

f-bor, 2019-04-25:
I just did the test with the current dev source and it's still failing to report all values. Only the first endpoint is reported.

Here is the log from ozwcp related to the node: https://pastebin.com/sfuhivHc

Also you were right: it's exactly the same symptoms as with fgbs-222 (if I include the node without secure mode, the values are there but it requires to enable polling to have updates).

Fishwaldo, 2019-04-26:
@f-bor could you try this please:
edit cpp/src/command_class/MultiInstance.cpp and edit line 546 so it looks like this:
`if (commandClassId == Security::StaticGetCommandClassId()) {`

restart after removing any ozwcache_*.xml and reinclude the device from scratch.

Thanks

f-bor, 2019-04-26:
I feel ashamed, I made a stupid mistake and tested with the master branch.

With the dev branch and without any modification, everything is working fine now. All the values are there and polling is not needed anymore.

Here is the log if you want to read it: https://pastebin.com/zURAxHe7
And the cache: https://pastebin.com/DntX21e6

The reactive power has no unit (endpoint 1, index 28) but I'm not sure var (volt-ampere-reactive) is quite common.

I'll test with an fgbs-222 later today.

thank you very much

f-bor, 2019-04-26:
It's also working very well with an fgbs-222 now.

I tried to backport your 2 commits into the master branch. It's working well for the values enumeration but I still have to poll to get value changes. Could you please tell me which commit(s) I should try to backport to have automatic value changes working as well ?

f-bor, 2019-04-27:
I did some more tests this morning and in the end, the polling is still required. Only one value on the first endpoint notifies its changes (the whole power consumption).

petergebruers, 2019-04-27:
Hi! I am very new to OZW but not new to Z-Wave. I am trying to help and understand at the same time. I think one possible cause of seeing only data reported on the first endpoint might be due to having "single channel" association instead of multichannel. If that does not ring a bell immediately, don't worry, I'll try to find out if what I say can be easily detected and changed... Will take some time...

f-bor, 2019-04-28:
few more things:
- I am not able to correctly include fgbs-222 anymore. Not sure why. The first time, it didn't worked until I power cycle the module. Now it's not working even if try to reset it to factory defaults. In my setup, it has 3 external temperature sensors that does not appear when included in secure mode.
- Regarding the ZMNHXD1 module, when the cache file is deleted, ozw is unable to discover it correctly again. It only reports 4 values.

Here are the files for fgbs-222:
- log: [OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/3124676/fgbs222.OZW_Log.txt)
- cache: [ozwcache_0xe94b847d.xml](https://github.com/OpenZWave/open-zwave/files/3124681/fgbs222.ozwcache_0xe94b847d.xml.txt)

And for ZMNHXD1:
- log (include): [OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/3124674/inclusion.OZW_Log.txt)
- cache (include): [ozwcache_0xe94b847d.xml](https://github.com/OpenZWave/open-zwave/files/3124679/ozwcache_0xe94b847d.xml)
- log (cache deleted): [OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/3124675/cache_deleted.OZW_Log.txt)
- cache (cache deleted): [ozwcache_0xe94b847d.xml](https://github.com/OpenZWave/open-zwave/files/3124680/cache_deleted.ozwcache_0xe94b847d.xml.txt)

f-bor, 2019-04-28:
I didnt saved the cache correctly for the fgbs222. Here is another run:
- [OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/3124692/OZW_Log.txt)
- [ozwcache_0xe94b847d.xml](https://github.com/OpenZWave/open-zwave/files/3124693/ozwcache_0xe94b847d.xml.txt)

petergebruers, 2019-04-28:
Regarding the FGBS-222 Smart Implant, this is a long shot, but Fibaro recently released firmware update 5.1. You'll need a HC2 or HCL, or contact your supplier or support to update, since they do not provide FW updates in binary form. Please do not shoot the messenger. I am not saying this will fix it, but here are the release notes:

Improved operation of external temperature sensors DHT22 and DS18B20
Improved B-button support in protection mod﻿e

The first one might be relevant, though they do not explicitly say it has to do with detection of the temp sensors... And the "protection class" is about limiting access via RF or button, so that one seems irrelevant.

I don't own a Smart Implant yet, so unfortunately I cannot test it.

That said... When I look at your Log, I do see instances with temperature reports... I am not familiar with this part in the OZW code but this sounds strange to me:

I search Node071.*SensorMultiLevel report and I get 3 relevant sections:

`2019-04-28 08:57:41.267 Info, Node071, Received a MultiChannelEncap from node 71, endpoint 8 for Command Class COMMAND_CLASS_SENSOR_MULTILEVEL
2019-04-28 08:57:41.267 Info, Node071, Received SensorMultiLevel report from node 71, instance 1, Temperature: value=18.8C
`

Based on the first line, that seems a MultiChannel report to me from instance 9 == endpoint 8 - this is probably one of the extra sensors, you might be able to tell if 18.8 makes sense... But when you look at the second line is says instance 1 == endpoint 0 == root of the device, this does not seem to match IMHO. But as I said, I am not familiar with the logic behind this and I don't want to jump to conclusions.

`2019-04-28 08:58:05.870 Info, Node071, Received SensorMultiLevel report from node 71, instance 1, Temperature: value=23.8C
019-04-28 08:58:05.870 Detail, Node071, Refreshed Value: old value=18.8, new value=23.8, type=decimal
`
So that seems a non-encapsulated report to me (instance 1 == endpoint 0 == root of the device) - this is probably the first sensor, you might be able to tell if 23.8 makes sense... I am guessing this is the built-in temperature sensor.

Notice it says temp changes from 18.8 (prev report) to 23.8 (this one) in about 20 seconds.

Next:

`2019-04-28 08:58:11.266 Info, Node071, Received a MultiChannelEncap from node 71, endpoint 8 for Command Class COMMAND_CLASS_SENSOR_MULTILEVEL
2019-04-28 08:58:11.266 Info, Node071, Received SensorMultiLevel report from node 71, instance 1, Temperature: value=18.2C
2019-04-28 08:58:11.266 Detail, Node071, Refreshed Value: old value=23.8, new value=18.2, type=decimal`

Again, the first line says "endpoint 8" but the second one says "instance 1".

Now the temperature decreases from 23.8 to 18.2 in about 6 seconds which seems implausible. To me it looks like reports from different sensors get mixed up.

If I look at the cache file:
`(...) COMMAND_CLASS_SENSOR_MULTILEVEL (...) <Instance index="1" />`

And no mention of other endpoints.

I would not want to jump to conclusions, it might be the device, it might be OZW, it might be the secure mode or even interpretation of the specifications... I'll try to discuss this with @Fishwaldo 

Because of (my) limited resources I am not dealing with the ZMNHXD1, I probably never wil own one of those, but I might buy a Smart Implant for testing and to use as a multi-temperature probe on my heating system. I'm sorry about that.

f-bor, 2019-04-28:
Thanks for your efforts and inputs, I appreciate. Unfortunatly, I don't have a home center box to update the firmware. Also, it's working quite well with non-secure mode. The only real issue is the polling necessity.

I also don't have real knowledge in zwave or openzwave. I gave a try to the command class spec but it's quite hard for a beginner. So I can't help on your analysis, even if the mixed up seems to make sense.

I did a run in non-secure mode in case it would provide some interresting information:
- [OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/3125040/OZW_Log.txt)
- [ozwcache_0xe94b847d.xml](https://github.com/OpenZWave/open-zwave/files/3125041/ozwcache_0xe94b847d.xml.txt)

Fishwaldo, 2019-04-28:
I haven’t got a chance to look at the logs, but you can try setting ForceInstances to true. Geez through the config files to find a example. (grep -R “ForceInstances” config/*)

f-bor, 2019-04-28:
Sure. With the ZMNHXD1 or the fgbs222 ? (or both)

f-bor, 2019-04-28:
It was already set at the end of [ZMNHXD1](https://github.com/OpenZWave/open-zwave/blob/94f0caf7963c8f8b4d2ff19331d8e5263a73fb8b/config/qubino/ZMNHXDx.xml) config file. I tried to force the value in the ozwcache file (it was set to false) and I tried to add it to fgbs222 and reinclude it => no change.

petergebruers, 2019-05-02:
Thank you for trying. I am going to toy a little with a Z-Uno, to better understand multichannel decoding in OZW and possible effect of encryption. With this device, I am able to simulate a device with arbitrary number of endpoints. I am sorry I am not able to help with your issue, yet, due to lack of understanding all the lovely details of multichannel encap...

Fishwaldo, 2019-05-03:
I'm closing this, as the current issue is different... can you reopen one if you still having issues?

closed 2019-05-03 by Fishwaldo