OpenZWave/open-zwave #1647: Zipato bulb 2 Model "rgbw2.eu" datecode "1807"

voggur, 2018-11-03:
Zipato bulb 2 Model "rgbw2.eu" datecode "1807" and I can not get it do anything other than on/off and dimm. I have tried editing RGBBulb2.xml with no luck. 
I am running Home Assistant 0.81.2 and python_openzwave 0.4.10
I can also see that there are lot more people over at the home assistant forums having the same problem.

voggur, 2018-11-04:
[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/2546413/OZW_Log.txt)

voggur, 2018-11-05:
this is part of the log whine i try to set color, please advice me if I need to do it in another way.
I can given you gui and ssh access if it will help in pm.

voggur, 2018-11-07:
[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/2555561/OZW_Log.txt)
I hope this is correct

Fishwaldo, 2018-11-07:
please try removing 
`  <CommandClass id="51" colorchannels="31" > 
  </CommandClass>`

From config/zipato/RGBBulb2.xml, shutdown, remove zwcfg*.xml and restart the application.
Send me a full log if there is no Color ValueID's again.

voggur, 2018-11-12:
[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/2570062/OZW_Log.txt)
Still not working

Fishwaldo, 2018-11-12:
I have a vague idea what's going on... But can you try this:

Please change the "Color Channels" Value to 0 in the app and post the logs. Again, shutdown, remove any zwcfg_*.xml, and restart before doing the above.

voggur, 2018-11-15:
ok I do not know if i am doing all correct or not. I am running "Home Assistant 0.81.6 " (hassbian) and running the following commands:
sudo service home-assistant@homeassistant stop
sudo rm  /home/homeassistant/.homeassistant/zwcfg_*.xml
sudo vi /srv/homeassistant/lib/python3.5/site-packages/python_openzwave/ozw_config/config/zipato/RGBBulb2.xml
sudo service home-assistant@homeassistant start
I think I have already tried all versions of RGBBulb2.xml on the github (master + v1.0~ v1.5) 
can you paste the RGBBulb2.xml verson here that I should be used.

Fishwaldo, 2018-11-26:
Ok. Closing.

closed 2018-11-26 by Fishwaldo

warcanoid, 2018-12-09:
I have the same problem: 
https://github.com/home-assistant/home-assistant/issues/19154
Copy rgbbulb2.xml to config/zipato/? Then stop HA, delete zwcfg_*.xml and restart ha?