OpenZWave/open-zwave #1650: enhancement request: support for NETWORK_MANAGEMENT_INSTALLATION_MAINTENANCE

timcoote, 2018-11-07:
### Issue Guidelines

Before opening a new issue, please review the following:
* The Github Issues is for OZW Bugs only. Please do not use this for general questions or howto type conversations. If you have a question, please post to our mailing list at https://groups.google.com/forum/#!forum/openzwave
* Please use the search feature to see if your issues have been raised or addressed first.
* Try the OpenZWave Log Analyzer at http://www.openzwave.com/log-analyzer if you are experiencing problems with a device on your network.
* Often we will require Log Files to help diagnose the issue. Please stop any application you are using, remove zwcfg_*.xml/ozwcache_*.xml files, and restart your application. After experiencing your issue, stop your application and attach the OZW_Log.txt file to this issue (please do not filter the log file, we often need the full log file to diagnose issues)
* If you want to add a new device to the OZW Database, please consult https://github.com/OpenZWave/open-zwave/wiki/Adding-Devices

Fishwaldo, 2018-11-07:
This is only valid with Z/IP Gateway, not the SerialAPI - So we can't implement it. 

You would have to use the Offical Developer Stack to do that.

closed 2018-11-07 by Fishwaldo

timcoote, 2018-11-12:
The SerialAPI specification (INS12350-Serial-API-Host-Appl.-Prg.-Guide.pdf), claims:

` 4.1.4 Version 6
In SDK 6.60.00 changed the Serial API communication interface to version 6 due to several extensions of the serial API [7], especially to improve support of installation and maintenance procedures (RSSI feedback, Routing algorithm feedback and Network statistics). The interface is backward compatible with version 5 as long as appended parameters in version 6 are ignored.
`
Have I got the wrong document, or am I misinterpreting that to mean the the CC in the title of this issue, and it actually means something else?

Fishwaldo, 2018-11-12:
From my understanding that’s different. The CC you are referring are built on top of the serial API (just like any other CC) but the Serial API reference below talks about extensions to the existing Serial API (specifically saying the extensions to the length field that can be ignored for older implementation). 

It’s like a transport level reporting (TCP/IP) versus a application level report (HTTP). 

The CC you referred to requires the controller node (Node 1 in typical OZW installations) to respond to CC Get’s, Set’s and Reports but this doesn’t work for as long as I’ve been doing OZW (I would love to know why tho). 

As far as Serial API version 6 - unless there was a a magic switch to turn on this extended info, none of the USB controllers (mainly Aeotec Z-Stick S2) I’ve worked with, (nor any of the log files that have been uploaded here for issue reports) contain this info. 

So I can conclude: Serial API version 6 has not been deployed to any known USB sticks (maybe the Z-Way razberry as the installer kit is built upon it?!?!  But again, I’ve never seen a log file that shows it) or it requires a magic command or firmware version to enable it (I would not be surprised - as previously Sigma sold the installer kit that offered this functionality - this could threaten the installer market the alliance goes after - although Silabs (new owner of ZWave) seem a bit more relaxed as far as openness of the protocol - under sigma the serial API was not a public document)

So the choices are: ship me a device that does Serial API 6, or write it yourself and send a PR! 

Now my own personal opinion: RSSI, network Routing etc on ZWave are next to useless. Apart from moving devices around, there is very little a end user can do to correct a network. You can’t get influence   routing etc. the power level CC should tell you both loss and latency between devices. Beyond that, the mesh network is self discovery, self healing and robust.

This is real world experience from numerous hotel installations that covered several floors, hundreds of nodes (and dozens of networks) that initially got me involved in OZW. Although I own a ZWave installer kit I’ve never actually used it in a real world deployment as common sense and the PowerLevel CC have provided enough info for me to diagnose RF range issues. 

In summary - not saying I am opposed to it in OZW if someone figures it out, but I’d prefer *my* time on practical enhancements like supporting new CC’s or updating existing ones to latest version. I’m already taking a lot of criticism for my limited time on OZW as it is today (as evident by the numerous when/now requests in the mailing list and here on the issue tracker).

timcoote, 2018-11-12:
Thanks. Very useful. I'll see if I can apply any influence to get further and better particulars about suitable controllers and see whether it's easier to create a PR or supply a controller.  If I can, I'll try for the former to reduce the load on yourself.

Interesting insights into the practicalities of ZWave installation. I've been involved in a couple of largish consumer facing IoT business, using various PANs (including ZWave). I've found that the costs come from support, rather than installation and not knowing the current state of the PAN can be a problem. Mesh is great for battefields, but less good (IMO) for long running networks. In many cases, post mortems of issues showed 'string of pearls' topologies emerging over months as stuff moved in the environment, causing unexpected outages.

satzg, 2018-11-13:
In case someone wants to purchase a diagnostic product. If you do please report back your experience.

https://www.zwaveproducts.com/shop/brands/zwaveproducts/zwp-tbx-pa-zwaveproducts-z-wave-toolbox-zwp-tbx-pa <https://www.zwaveproducts.com/shop/brands/zwaveproducts/zwp-tbx-pa-zwaveproducts-z-wave-toolbox-zwp-tbx-pa>

Thanks,
Greg