OpenZWave/open-zwave #1651: valueRefreshed Event not triggered with SensorBinary mapped from Basic Set

vibr77, 2018-11-07:
### Issue Guidelines

Before opening a new issue, please review the following:
* The Github Issues is for OZW Bugs only. Please do not use this for general questions or howto type conversations. If you have a question, please post to our mailing list at https://groups.google.com/forum/#!forum/openzwave
* Please use the search feature to see if your issues have been raised or addressed first.
* Try the OpenZWave Log Analyzer at http://www.openzwave.com/log-analyzer if you are experiencing problems with a device on your network.
* Often we will require Log Files to help diagnose the issue. Please stop any application you are using, remove zwcfg_*.xml/ozwcache_*.xml files, and restart your application. After experiencing your issue, stop your application and attach the OZW_Log.txt file to this issue (please do not filter the log file, we often need the full log file to diagnose issues)
* If you want to add a new device to the OZW Database, please consult https://github.com/OpenZWave/open-zwave/wiki/Adding-Devices

vibr77, 2018-11-07:
Hi all, 
I run the last version of openzwave with node-openzwave

I have a fibaro Motion Sensor with the following configuration 
`Node id="21" name="21 SORON C" location="COULOIR" basic="4" generic="32" specific="1" type="Routing Binary Sensor" listening="false" frequentListening="false" beaming="true" routing="true" max_baud_rate="40000" version="4" query_stage="CacheLoad">
		<Manufacturer id="10f" name="FIBARO System">
			<Product type="800" id="1001" name="FGMS001 Motion Sensor" />
		</Manufacturer>
		<CommandClasses>
			<CommandClass id="32" name="COMMAND_CLASS_BASIC" version="1" request_flags="4" after_mark="true" mapping="48" setasreport="true">
				<Instance index="1" />
			</CommandClass>
			<CommandClass id="48" name="COMMAND_CLASS_SENSOR_BINARY" version="1" request_flags="4" innif="true">
				<Instance index="1" />
				<Value type="bool" genre="user" instance="1" index="0" label="Sensor" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="False" />
				<SensorMap index="0" type="40" />
				<SensorMap index="0" type="56" />
				<SensorMap index="0" type="199" />
				<SensorMap index="0" type="215" />
			</CommandClass>`

As you can notice the Basic Set is mapped to the SensorBinary Class 

The issue is that when using openzwave : 

I received on the console log that shows value refreshed and value changed event but the Notification:Event is not triggered 

 `Node021, Received SensorBinary report: Sensor:40 State=On
2018-11-07 19:09:43.450 Detail, Node021, Refreshed Value: old value=false, new value=true, type=bool
2018-11-07 19:09:43.451 Detail, Node021, Changes to this value are not verified
2018-11-07 19:09:43.452 Detail, Node021, Notification: ValueChanged
2018-11-07 19:09:43.486 Detail, Node021,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x15, 0x03, 0x20, 0x01, 0xff, 0x3a`

I am listening to zwave.on("value changed") and value refreshed.
I can received these event where there is no mapping

In my config I do not have suppressrefreshedvalue

Can someone explain me what I m doing wrong ?

Fishwaldo, 2018-11-08:
Everything looks fine. I have no idea what your expecting. 

Please use the mailing list for help. This is a bug tracker.

closed 2018-11-08 by Fishwaldo