OpenZWave/open-zwave #1658: Qubino ZMNHADx create phantom/fake sensors humidity and CO2 sensors

bjacobse, 2018-11-29:
In my domoticz ver Version: 4.9700  3 pcs of  Qubino ZMNHADx Flush 1 Relay that all have DS18B20 temperature sensor suddenly seems to have have created extra sensors, so it contained a humidity and CO2 sensors, so then Domoticz created a combined sensor instead of only have the temperature sensor. I got some help from Domoticz forum and understand the issue is an OpenZwave issue.
The proposed manual edit is to remove those sensors created below COMMAND_CLASS_SENSOR_MULTILEVEL, but I don't know yet if this will "cure" the issue or not

I have the link to Domoticz forum where this have been discussed.
https://www.domoticz.com/forum/viewtopic.php?f=24&t=19280#p198890

Fishwaldo, 2018-11-29:
Please shutdown the app, remove zwcfg_*.xml and restart with logging enabled and post the OZW log file.

bjacobse, 2018-12-01:
It seems after removing zwcfg_*.xml file my Qubino devices is not being activated properly, I am not able to get temperature readings, and some configuration  seems to be missing for them too.
[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/2636084/OZW_Log.txt)

nechry, 2018-12-02:
I also have from time to time phantom/fake sensors reported to my devices.

nicweb84, 2019-01-23:
Same thing for me with a Qubino ZMNHIDX. It suddenly report a fake humidity sensor

bjacobse, 2019-01-23:
Hi Niceweb84
I can recommend to backup youtr cfg.xml file
Stop your homeautomation system
Then then edit cfg.xml file, find your device that suddenly have this fake humidity sensor and remote the whole line.
then restart home automation system and check if your fake sensor is gone

Fishwaldo, 2019-01-25:
Duplicate of #1687

closed 2019-01-25 by Fishwaldo