OpenZWave/open-zwave #1681: Oomi Multisensor 6 - could not save node config options (similar to Aeotec Multisensor 6)

benbeton, 2018-12-31:
I bought 2 new Oomi Multisonsors: http://www.oomi.com/discover/multisensor/

They are the same like “Aeotec Multisensor 6”

OpenZWave knows the specification: https://github.com/OpenZWave/open-zwave/blob/master/config/oomi/ft100.xml

After setting the parameter “Temperature Reporting Threshold” to “**20**”, the parameter reverts back to “**5121**”
here’s the log snip: https://pastebin.com/CVvPvTW7

sometimes I can see the following:
`"Unhandled Command Class 0x20"`
don’t know if that’s related to the issue

that happens with all config options, even with **wakeup timer**

Fishwaldo, 2019-01-06:
The log files?

benbeton, 2019-01-06:
above I posted this log:
https://pastebin.com/CVvPvTW7

can you work with it?

Fishwaldo, 2019-01-23:
Looking at the logs, I see nothing wrong. Please verify that a "20" is valid for this variable. (the original is 5121 so I suspect its not).

Z-Wave devices have no way to communicate errors around invalid configuration variables, so its most likely something wrong with what you are passing. 

Are you sure its identical to the Aeotec device?

Anyway, Nothing wrong with OZW, any more issues, please use mailing list.

closed 2019-01-23 by Fishwaldo

tobiasgraf, 2019-02-08:
I have the same device and tried exactly the same thing. (except the parameter size should be 2)
The value should be in a valid range (since its a temperature threshold with a precision of 1), the default value of 5121 is a bit high. 
I think this is a firmware issue of the device.

On a side note: the device (and other oomi devices) seems to have the configuration lock enabled by default, even though the manual says otherwise. Setting parameter 252 to 0 may help with other parameters of the device.