OpenZWave/open-zwave #1685: MCO Home MH7 thermostat setpoint failure

HWL1968, 2019-01-04:
Hi all,

I got a MCO Home MH7 for Christmas which is ZWave+ compatible. At least: it should be. 

After including into the network (which works fine), I get the following devices:
- temperature and humidity sensor: working fine
- thermostat 

mode: can be changed fine and also receives the mode set by the device / user on the device (works both ways)
- 3 thermostat setpoint devices: 'Auto changeover', 'Dry Air' and 'Cooling 1'. I can't set any of these three. 

Attached to this are:
- OZW_log.txt (is the OZW log file as plain text)
- zwcfg_xxx.txt (is the zwcfg file as plain tekst)
- the manual of the MH7H (the device I use is the MH7H-EH for electric heating - I am using an IR panel) as a pdf file

The node which isn't working is Node11. My Zwave network consists of:
- Aeotec Gen5 controller (node 1)
- 2 fibaro switches
- 2 Neo CoolCam switches

As far as I can see (but I am no zwave expert at all): the setpoint commands aren't handled properly by the MH7. But where the issue is, I am not sure).

Best regards and many thanks for your help.

HWL
[mco_home_mh7.pdf](https://github.com/OpenZWave/open-zwave/files/2726582/mco_home_mh7.pdf)
 
[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/2726581/OZW_Log.txt)

[zwcfg_0xdef01140.xml.txt](https://github.com/OpenZWave/open-zwave/files/2726579/zwcfg_0xdef01140.xml.txt)

HWL1968, 2019-01-09:
Hi Fishwaldo,

Great that you are willing to look into this issue. Is there anything I can do to help?

Best regards.

aseelert, 2019-01-10:
Its working for me with 0.84.6 (do not upgrade to 0.85.0 as zwave will stop working)

**change file:** /home/pi/homeassistant/lib/python3.5/site-packages/python_openzwave/manufacturer_specific.xml
**create new file:** /home/pi/homeassistant/lib/python3.5/site-packages/python_openzwave/ozw_config/mh7h.xml

`<?xml version="1.0" encoding="utf-8"?>
<Product xmlns="http://code.google.com/p/open-zwave/">
	
  <CommandClass id="64" name="COMMAND_CLASS_THERMOSTAT_MODE">
    <Value type="list" genre="user" instance="1" index="0" label="Mode" units="" read_only="false" write_only="false" min="0" max="0" value="0">
      <Item label="Off" value="0"/>
      <Item label="Heat" value="1"/>
      <Item label="Energy Heat" value="11"/>
      <Item label="Away" value="13"/>
    </Value>
    <SupportedModes>
      <Mode index="0" label="Off"/>
      <Mode index="1" label="Heat"/>
      <Mode index="11" label="Energy Heat"/>
      <Mode index="13" label="Away"/>
    </SupportedModes>
  </CommandClass>
  
   <CommandClass id="67" name="COMMAND_CLASS_THERMOSTAT_SETPOINT" version="1" request_flags="4" create_vars="true" base="0">
    <Instance index="1"/>
    <Value type="decimal" genre="user" instance="1" index="1" label="Heating" units="C" read_only="false" write_only="false" min="0" max="0" value="20"/>
    <Value type="decimal" genre="user" instance="1" index="11" label="Energy Heating" units="C" read_only="false" write_only="false" min="0" max="0" value="16"/>
  </CommandClass>

  
  <CommandClass id="112">
    <Value genre="config" instance="1" index="1" value="2" label="Upload temperature format automatically" units="" size="1" min="0" max="2" type="list">
      <Help>Upload temperature format automatically</Help>
      <Item value="0" label="Celsius"/>
      <Item value="1" label="Fahrenheit"/>
      <Item value="2" label="Follow the main display"/>
    </Value>
    <Value genre="config" instance="1" index="2" value="3" label="Upload temperature and humidity automatically" units="" size="1" min="0" max="3" type="list">
      <Help>Upload temperature and humidity automatically</Help>
      <Item value="0" label="OFF"/>
      <Item value="1" label="Upload the difference value only"/>
      <Item value="2" label="Timing upload mode only"/>
      <Item value="3" label="Upload the difference+timing upload mode"/>
    </Value>
    <Value genre="config" instance="1" index="3" value="5" label="Upload temperature difference" units="" size="2" min="3" max="1000" type="short">
      <Help>Base on 0.1°C unit, 5 by default, 5*0.1°C=0.5°C</Help>
    </Value>
    <Value genre="config" instance="1" index="4" value="30" label="Upload time interval regularly" units="" size="2" min="19" max="65535" type="short">
      <Help>Base on 1s unit, it suggest to be set above 30s</Help>
    </Value>
    <Value genre="config" instance="1" index="5" value="3" label="Upload humidity difference" units="" size="1" min="2" max="255" type="byte">
      <Help>Upload when percent difference exceeded</Help>
    </Value>
    <Value genre="config" instance="1" index="255" value="0" label="Factory setting" units="" size="1" min="0" max="255" type="byte">
      <Help>Restore the factory setting - write only, set to 85 to reset</Help>
    </Value>
  </CommandClass>

   
   <CommandClass id="133">
       <Associations num_groups="3">
           <Group index="1" max_associations="1" label="Lifeline"/>
           <Group index="2" max_associations="5" label="Basic_Set"/>
           <Group index="3" max_associations="5" label="Basic_Set"/>
       </Associations>
   </CommandClass>

</Product>
`
manufacture:

`	<Manufacturer id="015f" name="Mcohome">
		<Product type="0001" id="0001" name="Mini Energy Dimmer"/>
		<Product type="0002" id="0001" name="Mini Energy Switch"/>
		
		<Product type="210a" id="1302" name="MH-P210 Mini Energy Dimmer" config="mcohome/mhp210.xml"/>
		<Product type="3102" id="0201" name="MH-S311 One-load" config="mcohome/mhs311.xml"/>
		<Product type="3102" id="0202" name="MH-S312 Two-load" config="mcohome/mhs312.xml"/>
		<Product type="3102" id="0204" name="MH-S314 Four-load" config="mcohome/mhs314.xml"/>
		<Product type="5102" id="3141" name="MH-S314 Four-load" config="mcohome/mhs314.xml"/>
		<Product type="4102" id="0201" name="MH-S411 One-load" config="mcohome/mhs411.xml"/>
		<Product type="4102" id="0202" name="MH-S412 Two-load" config="mcohome/mhs412.xml"/>
		<Product type="4121" id="1302" name="MH-S412 Two-load" config="mcohome/mhs412.xml"/>
		<Product type="4121" id="5102" name="MH-S412 Two-load" config="mcohome/mhs412.xml"/>
		<Product type="0801" id="3102" name="MH8-FC-EU Thermostat" config="mcohome/mh8fceu.xml"/>
		<Product type="0802" id="3102" name="MH8-FC4-EU Thermostat" config="mcohome/mh8fceu.xml"/>
		<Product type="0905" id="0201" name="MH9-CO2-WD CO2 Monitor" config="mcohome/mh9co2.xml"/>
		<Product type="a800" id="1352" name="A8-9 Multi-sensor" config="mcohome/a8-9.xml"/>
        <Product type="0701" id="5102" name="MH7H Water/Electrical Heating Thermostat" config="mcohome/mh7h.xml"/>
        <Product type="0702" id="5102" name="MH7H Water/Electrical Heating Thermostat" config="mcohome/mh7h.xml"/>
        <Product type="0712" id="5102" name="MH7H Water/Electrical Heating Thermostat" config="mcohome/mh7h.xml"/>
        <Product type="0732" id="5102" name="MH7H Water/Electrical Heating Thermostat" config="mcohome/mh7h.xml"/>
		
		<Product type="5102" id="0103" name="Touch Panel Switch MH-S513" config="mcohome/mhs513.xml"/>
	</Manufacturer>`

you will see following devices:

climate.mcohome_mh7h_waterelectrical_heating_thermostat_energy_heating
climate.mcohome_mh7h_waterelectrical_heating_thermostat_heating
sensor.mcohome_mh7h_waterelectrical_heating_thermostat_relative_humidity
sensor.mcohome_mh7h_waterelectrical_heating_thermostat_temperature
zwave.mcohome_mh7h_waterelectrical_heating_thermostat

HWL1968, 2019-01-11:
Hi Aseelert,

I am using Domoticz and I assume this is Home Assistant. My SD card went corrupt last couple of days, but fortunately I made a backup last saturday. (didn't loose to much).

Now for the most important part: this has done the trick! Many thanks. I did a search for the xml definition file of the MH7 but couldn't find it. I am not sure whether I did a good search but am convinced I did. (since it is not included in the config directory of the sources neither).

Many thanks to you and a lot of kudo's. Is it possible to add it so everyone can use it whithout any hassle?

Best regards.

closed 2019-01-11 by HWL1968