OpenZWave/open-zwave #1687: Zwave nodes are getting values and command classes from other nodes.

koush, 2019-01-04:
I've looked over the issue guidelines, and unfortunately I do not think the deleting xml and getting a fresh log file method will work. This issue only occurs after the system has been running and restarted many times.

The issue:
After running for some time period, various zwave nodes start getting their command classes mixed in with other zwave nodes. For example, several of my dimmers now have command classes from my aoetec multisensor. This is visible in the xml files (attached) (Github does not support xml, zip is required). See node 50 for example. That's a dimmer that has a thermometer somehow.

[zwcfg.xml.zip](https://github.com/OpenZWave/open-zwave/files/2725633/zwcfg.xml.zip)

I'm well acquainted with open zwave, and have a working android port:
https://github.com/koush/open-zwave/commit/d2439fbfa7cf54556092e6589a20fe85c3718aae

Is this by chance fixed in the dev branch? I am running master, which seems more stable. Any idea where to look to find the cause? It seems to me the only way this could happen is some race condition with accessing nodes and values from multiple threads.

koush, 2019-01-06:
Here you go.
[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/2730945/OZW_Log.txt)

koush, 2019-01-07:
I noticed one of the erroneous command classes that showed up was COMMAND_CLASS_COLOR. I do not have any COMMAND_CLASS_COLOR devices on my zwave network. My guess is there's some sort of data corruption taking place. Some platforms don't zero memory out on new allocations (like Android).

RicardP, 2019-01-08:
I see this issue happen on regular basis (my case windows platform), and the problem does clearly increase with the affected Node bad signal/mesh quality.

It is said that there is a CRC to validate packets but it seems not work correctly, OR....

You wrote the corrupt data might occur by wrong code (race condition), however the issue is more frequent with a bad signal Node. Such a Node requires lots of z-wave packets re-sending if CRC works... Or "simply" because CRC handshake itself does not work as expected...

Anyway this issue causes zwcfg.xml growing cancer wise over time and at a point kill the OZW installation (if not manually deleting the erratic Nodes in zwcfg from time to time I need to do this), because parsing this ever growing file gives a Snow Ball Effect on false entries as it takes more and more CPU time........

If you look trough Issues tab there is also others that reported crazy values and mixed up switch states within one Node.
By manually performing the "Refresh Node Info" the Node "real" values are restored, but ghost entries can only be removed by deleting the Node in zwcfg.xml as said above.

I just wanted to tell my experiences ... not that it helps you much in bug tracking but anyway might give some clues.

Many thanks!

RicardP, 2019-01-08:
So then corrupt data and ghost CC's and swapped switch states (dual switches) do most likely happen because of bad RF signal. CRC check should avoid this or, but if the CRC level is not good enough maybe a corrupt packet now and then will show correct CRC "by chance"?

The work around to "freeze the model" (CC's) as you mention seems risking that not all CC's will be included, like power consumption cant take unknown time before it is reported?

Also, removing the Node in xml will in most cases (in my experience) need a "triple click" to force the Node to re-send its CC's. Performing only "Refresh Node Info" no not always work and, of course, not on battery powered Nodes.

koush, 2019-01-10:
I believe the Completed event is only shown after the node has been woken up and reported everything. I tested it with some door sensors, and those took upwards of 8 hours before becoming "completed".

RicardP, 2019-01-11:
Sure battery Nodes takes longer time to Complete, but still I experience that even mains powered such wont always Complete the CC's capabilities without performing a manual "triple click".

Any way, the real issue here is still that corrupt data/packets are accepted somehow and trashing the zwcfg.xml and measured values.

Fishwaldo, 2019-01-16:
I’ll do some testing to confirm the CRC validation and have a look over the log files. 

I have seen a few devices go haywire and send rubbish out in the past but never seen it end up as ghosted command classes. So I’m generally stumped.

Fishwaldo, 2019-01-23:
> I noticed one of the erroneous command classes that showed up was COMMAND_CLASS_COLOR. I do not have any COMMAND_CLASS_COLOR devices on my zwave network. My guess is there's some sort of data corruption taking place. Some platforms don't zero memory out on new allocations (like Android).

@koush There is no reference to COMMAND_CLASS_COLOR in the Log file supplied.

Fishwaldo, 2019-01-25:
@austinmroczek 
Hrm... Ok. That Luminescence Report is a "valid" packet as far as OZW is concerned. *EVERYTHING* checks out (it can't be a random corruption here. CRC is valid, and its structure right down to "encoding" a decimal value is correct!)

The *ONLY* thing I see is its a EXPLORER Frame, which basically means that the node didn't know how to reach the controller directly (I'm guessing the controller isn't a neighbor?) - Same with Node 43 as well.

Just for my Reference, here is the packet decode (of Node 43, but its basically identical to 42)
[image]

2019-01-01 09:32:31.233 Detail, Node042,   Received: 0x01, 0x0e, 0x00, 0x04, 0x10, 0x2a, 0x08, 0x31, 0x05, 0x03, 0x0c, 0x00, 0x00, 0x00, 0x24, 0xd8

(should note this packet is the "source" of the corruption here. We create a ValueID for the node, so its the first time we have seen it)

A few questions though:
- Around the time this came in Node 43 sent 4 Luminance Reports in 300milliseconds. Is Node 43 configured correctly? (I'm not sure your lighting changes so quickly!)
- Node 42 (goControl) is a battery powered device? I'm inferring its just waking up, and hadn't completed the interview process with OZW, so I'm guessing OZW was recently restarted?

So - In this case - Nothing wrong with OZW. Its a valid packet received from the device. 

I will commit a change soon that stops creating new ValueID's on the SensorMultiLevel CC in a Report Message (we should have all the valid ValueID's created during the interview process).

But that doesn't fix the root cause of this... and I have a feeling looking at this packet, its a bug in the Silabs/Sigma SDK. If we can get a few more log files, I can try working with them to figure this out.

Fishwaldo, 2019-01-25:
> 
> 
> I’ll do some testing to confirm the CRC validation and have a look over the log files.
> 
> I have seen a few devices go haywire and send rubbish out in the past but never seen it end up as ghosted command classes. So I’m generally stumped.

Just FYI - Our CRC validation checks out. Corrupt Packets are rejected. So unless we are very unlucky, its not RF corruption giving us random CommandClasses/Values

austinmroczek, 2019-01-25:
All three of these nodes do not show the controller as a neighbor.  Each has (similar if not exactly the same) multiple one-hop routes to the controller.

Node 42 can be battery powered, or continuously powered by the HVAC system.  I have it powered by HVAC.  I was having (non-zwave related) trouble with my HVAC system around this time period, which was causing the zwave device to lose power and re-boot multiple times per day, so that could be why it was doing an interview.  

Node 43 is battery powered.  It's configured to report at the "normal" rate (versus "frequent") which is not really defined in the documentation I have for the device.  Looking at my Home Assistant logs it appears to be updating only a few times per day.  However every time it reports, OZW_Log shows it sending 5 reports in about 2 seconds, each with exactly the same values.  That is usually followed by a warning saying there was no ACK received and the device may be asleep.  It has the smallest battery of any device I've seen.  Perhaps to save power it wakes up, sends multiple reports and the quickly goes back to sleep, hoping at least one report goes through?  

Thanks!

Fishwaldo, 2019-01-25:
> However every time it reports, OZW_Log shows it sending 5 reports in about 2 seconds, each with exactly the same values. 

Hrm - Check your association on the device. If it has multiple association groups and they are all are pointing at the controller node, maybe each group is configured to send the same info. (in which case you could remove 4 of the association's.... ) 

The manual should tell you what each association group is for.

koush, 2019-01-25:
> @koush There is no reference to COMMAND_CLASS_COLOR in the Log file supplied.

@Fishwaldo I'm aware. I'm not sure when/how it showed up. It was just a data point that I was providing-- a COMMAND_CLASS_COLOR showed up on one of my devices for no reason. It's visible only in the Xml, and was also reported up as a notification. It's another symptom of these phantom command classes.

[zwcfg_0xc73b479f.xml.zip](https://github.com/OpenZWave/open-zwave/files/2797515/zwcfg_0xc73b479f.xml.zip)

koush, 2019-01-25:
Another point of note regarding your theory that it may be in the Silabs sdk. This problem became noticeable when I started using a Aeotec Multisensor 6. However, it was also occurring (albeit less frequently) when I added an Ecolink door/widow sensor.

Fishwaldo, 2019-01-26:
> However every time it reports, OZW_Log shows it sending 5 reports in about 2 seconds, each with exactly the same values.
@austinmroczek 
You can confirm its 5 every time? In the logs you provided its 4 plus the packet for Node 42. 

What USB controller are you using?

austinmroczek, 2019-01-26:
It's not 5 every time.  I didn't check the entire file, but it seems to vary from 3 to 8.  

Also, the first report in each set is slightly different, showing the "old value" being different a different number (same value as reported by the previous set) and then a "new value" that is different.  Then for the remainder of the set, it the old value = new value.

I'm using a Razberry daughter card.  https://z-wave.me/products/razberry/

cybergrimes, 2019-02-21:
I'm seeing this issue with my Yale locks here in the last week or so, they've been installed for 6+ months but now starting to present entities that do not belong to them (cover and light) but then today I'm seeing a "level" entity that I don't otherwise have on my Z-Wave network or anywhere in my HA installation. Let me know if there's anything I can provide to help, thanks.

Fishwaldo, 2019-02-21:
I need logs from the time period when the new command classes were introduced.

cybergrimes, 2019-02-21:
I don't think I can get those, the oldest backup I have is 2/4 and I can see these off command classes listed in the zwcfg file like COMMAND_CLASS_THERMOSTAT_FAN_MODE and COMMAND_CLASS_MULTI_INSTANCE/CHANNEL. I know HA didn't create a level entity until sometime between yesterday and today since it wasn't in the history panel until today. I've attached logs from yesterday and today, hopefully something is there...

[OZW_Log-Feb21 8am.txt](https://github.com/OpenZWave/open-zwave/files/2890297/OZW_Log-Feb21.8am.txt)
[OZW_Log-Feb20-3pm.txt](https://github.com/OpenZWave/open-zwave/files/2890298/OZW_Log-Feb20-3pm.txt)

Fishwaldo, 2019-03-26:
Just FYI, I found a offbyone logic error in the Security CC code today, that could introduce ghost CC, but only during startup, not at random intervals. I'll commit a fix for that shortly.

Fishwaldo, 2019-03-26:
Fix added to both the Master and Dev branch. But as mentioned, I doubt this is the actual root cause you are all seeing as this fix only addresses a bug that happens when we are doing the initial query of a device and not at random times...

koush, 2019-03-28:
> Just FYI, I found a offbyone logic error in the Security CC code today, that could introduce ghost CC, but only during startup, not at random intervals. I'll commit a fix for that shortly.

I'm unsure if it is happening during startup or during long term execution. I'm restarting openzwave dozens of times per day, so that could certainly be it. But initial query is only when the node is associated, correct?

stiansoevik, 2019-04-10:
I have also experienced strange CCs on different devices, in addition to extreme (wrong) values. Can we be sure that it is not actual corrupted data that slips through? AFAIK, CRC polynoms are non-trivial to choose optimally, and depending on the choice, it can lead to errors not being detected. I bought a book on Z-Wave while trying to fix another issue, and what it says about checksums of 8 bits for "channel 1 and 2" (lower speeds, if my skim reading was right) does not seem reliable at all.

If this is the case, I think discarding unexpected CCs is a good solution.

Fishwaldo, 2019-04-12:
The CRC is mandated by the Spec, so we can't do anything there. 

As I keep mentioning (broken record?) I need log files from when the ghost CC is introduced to diagnose.

CommandClasses can only be added in 3 stages in OZW: 
1) During Inclusion when we get the NIF (though NIF's can come at any time). 
2) During Inclusion on Secure Devices when querying for supported CC 
3) Via the MultiInstance CommandClass for different instances.

Typically 1 and 2 only happen when you add a device, and for 95% of cases, 3 also happens during inclusion, but MultiInstance CommandClass does allow "Dynamic" endpoints, which allow a endpoint to change configuration at *any* time. 

Finally, what is a "unexpected" CommandClass? Vendors are free to add any additional CommandClasses over and above what the DeviceType is. For example, I have a doorbell that exposes a temperature sensor.

As the only log I've seen of a "Ghost" value is valid (https://github.com/OpenZWave/open-zwave/issues/1687#issuecomment-457454526)- The best assumption I can make is its a bug in the SiLabs SDK.... Until I see a OZW log file that shows otherwise, I can't do anything about this because I don't *KNOW* where the problem is.

stiansoevik, 2019-04-12:
Sorry, I don't have this in log files at it happened a long while ago, but I'll keep it in mind. Examples of unexpected CCs I see are tank capacity of a Fibaro multisensor, direction on an aeotec heavy duty relay, tide level on a fibaro dimmer...

ChPeSc, 2019-05-21:
I can confirm the issue in my network.
I have also a AEON multisensor 6 in the network
After some time some other devices (i.e. AEON multiswitch 6) receive extra values in the configuration command class from the conifguration class of the multi sensor.
I have a Z-Wave.Me UZB1 stick
I do not have logfiles yet but will try to get some. (Need some time though)

I never observed any similar issue with the older release of open-zwave (v1.4)

petergebruers, 2019-05-21:
Thank you @ChPeSc for joining this topic. It is both an intriguing and annoying problem... I've discussed this with a few users and the result is: "need more info" so if you could capture a complete OZW_Log that might help. It has been reported on 1.4 (the log at the start of this topic says "OpenZwave Version 1.4.3248 Starting Up") and you suggest that you are running 1.6. All possible explanations are open, including firmware controller bug (reported on different controllers but they are all based on an SDK provided by Silabs) or issue in OZW (because the "core" if I can call it like that of 1.4 and 1.6 are the same)...

ChPeSc, 2019-05-21:
I try to present some logfiles and additional info.

Word of caution: I am using python-openzwave which I have compiled myself against the latest version of openzwave v1.6 ... At least I think that I have done this... I might have screwed up...
I personally did not have problems with the python-openzwave "out of the box" package (which if I am not mistaken, has version 1.4 of ozw). But this might be some particularity of my setup, of course, since petergebruers confirmed that the problem had been reported with v1.4.  

What I did: 
Setup: 

- Controller is Z-Wave.Me UZB1 stick
- Active devices : AEON power switch Gen 5, 
- Everspring plug in lamp AN145  
- AEON Multisensor Gen 6 ON BATTERY, i.e. mostly sleeping

- I removed the ozwcache file to start from scratch. Discovery procedure was running until the 2 awake devices are successfully queried
- I copied the ozw cache file away to compare to it later (ozwcache_0xd11b6ae1.xml.2). It seems ok to me (did not study all details)
- Then at 18:45 my multisensor is waking up and is being queried. Afterwards the ozwcache file is "corrupted". The configuration class entry for the multiswitch contains entries from the multisensor. It seems that some configuration with a certain id/index combination are overwritten by those of the multisensor. (Helptext and Labelname changes).

I attach a zip file with the two ozwcache files and the OZW logfile with level Detail. I hope this helps.

Thanks a lot for looking into this!

[logfiles.zip](https://github.com/OpenZWave/open-zwave/files/3204077/logfiles.zip)

petergebruers, 2019-05-21:
Thanks for the detailed explanation. Indeed, something is not right with your build, the log starts with "Always, OpenZwave Version  Starting Up" but it should say something like "Always, OpenZwave Version 1.6-183-g600cee87-dirty". To be honest, I am not sure what to recommend, it took me a while to figure out what the different "flavors" of pyozw exactly do. At the moment, iirc, I am running it in "shared" flavor on my mac, and I do "BUILD=DEBUG gmake -j 4 && make install" in open-zwave to test HASS (and other programs, they link to the shared library). But I am mostly using either MinOZW and open-zwave-control-panel to verify and debug... But that is not important in this case, only saying that MinOZW should link with the correct version.

I'll have a detailed look at your data, probably tomorrow.

Fishwaldo, 2019-05-22:
@chpesc there was a separate issue fixed where labels were getting messed up. I fixed that with a recent commit to master. Have you upgraded to the latest?

Please remove any ozwcache cache file after updating and you should be good.

Fishwaldo, 2019-05-22:
@ChPeSc this was the commit: https://github.com/OpenZWave/open-zwave/commit/6cda09338dffe975f7a5f0788b42f52274a98dde

ChPeSc, 2019-05-22:
Thanks a lot Fishwaldo, I will try as soon as I will have time (il will take a while, sorry...). Scanning quickly over your commit it seems probable that this will fix my problem... My setup does not contain this fix yet. Therefore Peter, do not use too much time with my logfiles, I would say!
Sorry for having mixed up the topics... 
Thanks a lot !

ChPeSc, 2019-05-24:
I confirm that my issue was solved with Fishwaldos commit : 6cda093 .
Thanks !

sailmich, 2019-06-02:
Rpi 3
Zstick
Domoticz
Fibaro FGRM222 and other switches

I also have an issue with a corrupt zwcfg file on my system. I replaced motors of two of my blinds. As I would make the calibration run I saw strange parameters in the configuration like a pir sensor!
Even after rename zwcfg file I don't get a new zwcfg file. Domoticz is running with it's newest beta and open zwave version 1.66. 
How can I help to fix this issue.

petergebruers, 2019-06-02:
@sailmich thank your for reporting your observation(s).

At the moment, 2019-06-02, please do not run the Domoticz beta "in production", because it is based on OZW 1.6 and the maintainers are still updating the code. See for example this issue about SceneActivation: 

https://github.com/domoticz/domoticz/issues/3240

Also the configuration file format(s) differ...

It is good for testing though.

This also answers your question "I don't get a new zwcfg file" because in 1.6 it is no longer called like that (it stars with ozwcache). But if you revert to running Domoticz stable, you'll be back on OZW 1.4 and it should recreate your cwcfg...

sailmich, 2019-06-02:
> domoticz/domoticz#3240 <https://github.com/domoticz/domoticz/issues/3240>

Thanks for your answer, but unfortunately I don't have a backup. With 30+
switches it isn't funny to start from scratch. It seems that all my old
switches still working as they should. I cross my fingers that the issue
will fix. Better make a backup next time🙄

Fishwaldo, 2019-06-08:
Looks like this is a issue with the Silibs SDK and not OZW. Just found a user on Veralight that is reporting his sensor also randomly gets strange Values:

> Oh really? Right now I've got an aeotec 6in1 sensor that keeps sprouting child devices, _Smoke Sensor _Motion Sensor1, _Motion Sensor2, 3 4 5 etc. and _GET LANG or something along those lines. Deleting a child device removes all the others and the parent.
It is the furthest device from my Vera, which I've wondered if it could be related.

koush, 2019-06-08:
Could these sensors acting as zwave repeaters cause these ghost Values on other devices? I'm seeing strange Values manifesting on my GE dimmers.

petergebruers, 2019-06-09:
I would say that purely theoretically a "repeater node" could mess with the packet, and if that happened it would be a SDK bug because all protocol TX, RX, retransmisson, routing and more) is handled by the SDK aka the firmware. What I like about that theory is that it could produce perfectly valid packets, indistinguishable from real, but with bogus contents. But then we have to know... which repeater?

It is *very* hard to diagnose such an issue without Zniffer. Only Zniffer can register the "raw" packets.
There is a little bit of hope if someone has the issue (on a somewhat regular basis) + a recent controller + he/she would be running OZW 1.6 - because in that case OZW can log which repeater was used.

I have been thinking, with a little luck there is a hint in the OZW_Log but some of you might notice the bogus node data only after a reboot, which restarts the log.

OZW does support appending the log file, but it is not the default, it is considered an option for debugging. Debug logging on a large production network will make these files grow fast, be careful.

You can enable this by editing the file Config/options.xml and adding this line:

```
<Option name="AppendLogFile" value="true" />
```

There is not a lot of data, but I get the idea, which is nothing more than an idea, that the "MultiSensor 6" might be involved somehow. This device has got plenty of FW updates, might be worth checking if you're having the latest, atm that's 1.11 https://aeotec.freshdesk.com/support/solutions/articles/6000036562-how-to-update-multisensor-6-z-wave-firmware-

It might be purely coincidence, I guess the MS6 sends plenty of packets per day (e.g. it can report humidity very frequently because the value toggles between n and n+1 unless you change some parameter, tested on Fibaro HC2) which increases the "hit rate".

sailmich, 2019-06-10:
I renamed the ozwcachexxxxx.xml file. Then refreshed all nodes in setup/Hardware. Reboot my Rpi and got the correct configuration:). I'm on newest domoticz beta 4.1088 and ozw 1.6.

@petergebruers I do have a multi sensor 6 I didn't change the firmware for more than a year. As I remember I use 1.13 it's a beta where you can turn of the led. I already get a log file in OZW_Log.txt is that what you're talking about? 
<Option name="AppendLogFile" value="true" /> isn't activated. The Log is growing how can I stop this.

petergebruers, 2019-06-13:
Five minutes ago someone with a Fibaro HC2 contacted me:

```
why HC2 sometimes confound temp & humidity report from my MS6?
so. I can see in log 70% humidity and just from a moment 70*С and Fibaro alarmed me - FIRE!!
```
So the controller logs "humidity" from a MS6 then logs "70 °C" probably from another device. I have seen similar reports, the numeric values match. Again this involves the MS6. Still does not prove anything though, bet we can say it happens on a controller based on the series 300 chip which is pretty old and the SDK has not been updated in a while. Also, like OZW, the controller accept the packet because the packet is valid (but the value does not make sense but the controller cannot decide this for you).

It is impossible to tell with certainty if the MS6 was the sender, the repeater, or even involved in the transaction of the "implausible packet"

cincuranet, 2019-08-28:
> bet we can say it happens on a controller based on the series 300 chip which is pretty old

I'm using Aeotec Z-Stick (`ZW090-C`), which is using 500 series chip, and happens in my setup too. I have bunch of MS6 too, all running as battery powered (hence cannot be repeater).

Navi2016, 2019-10-11:
Think i'm having a similar issue. using Domoticz 4.10717 and have AEON MS6 in the netwerk. My Fibaro smoke sensors temperature are sudden reporting as Temp + Humidity | WTGR800. It does not have humidity so its stuck at 32%. Temperature does update. Was Temp | LaCrosse TX3 before.

RicardP, 2019-10-16:
@Navi2016 Is Domoticz 4.10717 based on OZW 1.4? I had the same issues as you described before but since OZW1.6 (later betas) I dont experience mixed/ghost values, at least "not noticeable". 
Another improvement in 1.6 is that you can perform Refresh Node Info, this action first wipes selected Nodes Ghost/corrupt values/CCs and then rebuilds the Nodes capabilities. The Refresh can be done without loosing related Devices in Domoticz.

JanJaapKo, 2020-01-29:
Hello,

I have seen this issue quite some time ago, and now it is back. The system had been running for more than a year flawlessly before. The first occurrence was around the time of upodating to OZW 1.6 if I'm not mistaken (but have no proof).  I solved it back then by refreshing all the files as mentioned somewhere else, it was ok for some months and now the issue is back. What I see is 2 things:
1. on my Everspring ST814 Temp&Hum sensor suddenly reports 'Radon concentration' in the command class sensor multilevel.
2. on my Fibaro FGBS01 universal binary sensor (has 2 temperature sensors connected) the way it is reported changes from time to time between 'Temp + Humidity | WTGR800' and ' Temp | LaCrosse TX3 '.

As with all the other users, unfortunately I have no logs of the change. I hope this input helps in solving this issue.

I'm running on my Rpi 3:
Domoticz 4.11662
OZW *1.6-1019-g0d94c942
Aeon LABS ZW090 Gen 5 USB stick

closed 2021-07-17 by koush