OpenZWave/open-zwave #1698: OpenZwave incorrect notification in SWITCH_BINARY command class for Fibaro fgwpe

enhhn17, 2019-01-21:
I have some zwave devices and control them using OpenZWave.

In my code, i have a function named "OnNotification(...)" and i added this function to Manager as watcher (using Manager::Get()->AddWatcher(OnNotification, NULL)).

In the function OnNotification(), all notifications are correct except some notifications of my Fibaro fgwpe.
When i turn on my Fibaro using OpenZWave, the function OnNotification() is called but the value of the passed notification to the function is incorrect and has level=0 (for SWITCH_BINARY command class (0x25)). And when i turn it off the level = 1. Not turning on/off the Fibaro device manually has have no problem. Another note is that when i active console log (by Options::Get()->AddOptionBool("ConsoleOutput", true)), there is no problem and all notifications are correct.

Fishwaldo, 2019-01-23:
I can't follow - where are you getting the value from? 

You should be reading the changed value from your notifications by calling Manager::GetValueAs*(valueid sent by the notification)

Fishwaldo, 2019-01-24:
Thats the correct code. If its reporting the wrong value, its most likely what the device is sending (we don't "interpret" any values received. 

Can you show logs where its "wrong"?

enhhn17, 2019-01-26:
When i turned on the Fibaro wall plug:

21:31.174 Info, Node008, Sending (Send) message (Callback ID=0x5e, Expected Reply=0x13) - SwitchBinaryCmd_Set (Node=8): 0x01, 0x0a, 0x00, 0x13, 0x08, 0x03, 0x25, 0x01, 0xff, 0x25, 0x5e, 0x4d
2019-01-26 08:21:31.179 Detail, Node008,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2019-01-26 08:21:31.180 Detail, Node008,   ZW_SEND_DATA delivered to Z-Wave stack
2019-01-26 08:21:31.192 Detail, Node008,   Received: 0x01, 0x05, 0x00, 0x13, 0x5e, 0x00, 0xb7
2019-01-26 08:21:31.193 Detail, Node008,   ZW_SEND_DATA Request with callback ID 0x5e received (expected 0x5e)
2019-01-26 08:21:31.193 Info, Node008, Request RTT 18 Average Request RTT 17
2019-01-26 08:21:31.193 Detail,   Expected callbackId was received
2019-01-26 08:21:31.193 Detail,   Expected reply was received
2019-01-26 08:21:31.193 Detail,   Message transaction complete
2019-01-26 08:21:31.193 Detail, 
2019-01-26 08:21:31.193 Detail, Node008, Removing current message
2019-01-26 08:21:31.193 Detail, 
2019-01-26 08:21:31.194 Info, Node008, Sending (Send) message (Callback ID=0x5f, Expected Reply=0x04) - SwitchBinaryCmd_Get (Node=8): 0x01, 0x09, 0x00, 0x13, 0x08, 0x02, 0x25, 0x02, 0x25, 0x5f, 0xb2
2019-01-26 08:21:31.199 Detail, Node008,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2019-01-26 08:21:31.199 Detail, Node008,   ZW_SEND_DATA delivered to Z-Wave stack
2019-01-26 08:21:31.211 Detail, Node008,   Received: 0x01, 0x05, 0x00, 0x13, 0x5f, 0x00, 0xb6
2019-01-26 08:21:31.211 Detail, Node008,   ZW_SEND_DATA Request with callback ID 0x5f received (expected 0x5f)
2019-01-26 08:21:31.212 Info, Node008, Request RTT 17 Average Request RTT 17
2019-01-26 08:21:31.212 Detail,   Expected callbackId was received
2019-01-26 08:21:31.218 Detail, Node008,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x08, 0x03, 0x25, 0x03, 0x00, 0xdf
2019-01-26 08:21:31.218 Detail, 
2019-01-26 08:21:31.218 Info, Node008, Response RTT 24 Average Response RTT 24
2019-01-26 08:21:31.218 Info, Node008, Received SwitchBinary report from node 8: level=Off
2019-01-26 08:21:31.219 Detail, Node008, Refreshed Value: old value=true, new value=false, type=bool
2019-01-26 08:21:31.219 Detail, Node008, Changes to this value are not verified
2019-01-26 08:21:31.219 Detail, Node008,   Expected reply and command class was received
2019-01-26 08:21:31.219 Detail, Node008,   Message transaction complete
2019-01-26 08:21:31.219 Detail, 
2019-01-26 08:21:31.219 Detail, Node008, Removing current message
2019-01-26 08:21:31.219 Detail, Node008, Notification: ValueChanged
2019-01-26 08:21:31.927 Detail, Node008,   Received: 0x01, 0x0c, 0x00, 0x04, 0x00, 0x08, 0x06, 0x31, 0x05, 0x04, 0x22, 0x02, 0x1f, 0xf6
2019-01-26 08:21:31.928 Detail, 
2019-01-26 08:21:31.928 Info, Node008, Received SensorMultiLevel report from node 8, instance 1, Power: value=54.3W
2019-01-26 08:21:31.928 Detail, Node008, Refreshed Value: old value=0.0, new value=54.3, type=decimal
2019-01-26 08:21:31.928 Detail, Node008, Changes to this value are not verified
2019-01-26 08:21:31.928 Detail, Node008, Notification: ValueChanged
2019-01-26 08:21:36.192 Info, Node008, Value::Set - COMMAND_CLASS_SWITCH_BINARY - Switch - 0 - 1 - False
2019-01-26 08:21:36.193 Info, Node008, SwitchBinary::Set - Setting node 8 to Off
2019-01-26 08:21:36.193 Detail, Node008, Queuing (Send) SwitchBinaryCmd_Set (Node=8): 0x01, 0x0a, 0x00, 0x13, 0x08, 0x03, 0x25, 0x01, 0x00, 0x25, 0x60, 0x8c
2019-01-26 08:21:36.193 Detail, Node008, Queuing (Send) SwitchBinaryCmd_Get (Node=8): 0x01, 0x09, 0x00, 0x13, 0x08, 0x02, 0x25, 0x02, 0x25, 0x61, 0x8c

When i turned it off:
2019-01-26 08:21:36.194 Info, Node008, Sending (Send) message (Callback ID=0x60, Expected Reply=0x13) - SwitchBinaryCmd_Set (Node=8): 0x01, 0x0a, 0x00, 0x13, 0x08, 0x03, 0x25, 0x01, 0x00, 0x25, 0x60, 0x8c
2019-01-26 08:21:36.199 Detail, Node008,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2019-01-26 08:21:36.199 Detail, Node008,   ZW_SEND_DATA delivered to Z-Wave stack
2019-01-26 08:21:36.212 Detail, Node008,   Received: 0x01, 0x05, 0x00, 0x13, 0x60, 0x00, 0x89
2019-01-26 08:21:36.212 Detail, Node008,   ZW_SEND_DATA Request with callback ID 0x60 received (expected 0x60)
2019-01-26 08:21:36.212 Info, Node008, Request RTT 18 Average Request RTT 17
2019-01-26 08:21:36.212 Detail,   Expected callbackId was received
2019-01-26 08:21:36.212 Detail,   Expected reply was received
2019-01-26 08:21:36.212 Detail,   Message transaction complete
2019-01-26 08:21:36.213 Detail, 
2019-01-26 08:21:36.213 Detail, Node008, Removing current message
2019-01-26 08:21:36.213 Detail, 
2019-01-26 08:21:36.213 Info, Node008, Sending (Send) message (Callback ID=0x61, Expected Reply=0x04) - SwitchBinaryCmd_Get (Node=8): 0x01, 0x09, 0x00, 0x13, 0x08, 0x02, 0x25, 0x02, 0x25, 0x61, 0x8c
2019-01-26 08:21:36.218 Detail, Node008,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2019-01-26 08:21:36.218 Detail, Node008,   ZW_SEND_DATA delivered to Z-Wave stack
2019-01-26 08:21:36.231 Detail, Node008,   Received: 0x01, 0x05, 0x00, 0x13, 0x61, 0x00, 0x88
2019-01-26 08:21:36.231 Detail, Node008,   ZW_SEND_DATA Request with callback ID 0x61 received (expected 0x61)
2019-01-26 08:21:36.231 Info, Node008, Request RTT 17 Average Request RTT 17
2019-01-26 08:21:36.231 Detail,   Expected callbackId was received
2019-01-26 08:21:36.238 Detail, Node008,   Received: 0x01, 0x09, 0x00, 0x04, 0x00, 0x08, 0x03, 0x25, 0x03, 0xff, 0x20
2019-01-26 08:21:36.238 Detail, 
[... cut ...]
2019-01-26 08:21:36.948 Detail, Node008,   Received: 0x01, 0x0c, 0x00, 0x04, 0x00, 0x08, 0x06, 0x31, 0x05, 0x04, 0x22, 0x00, 0x00, 0xeb
2019-01-26 08:21:36.948 Detail, 
2019-01-26 08:21:36.948 Info, Node008, Received SensorMultiLevel report from node 8, instance 1, Power: value=0.0W
2019-01-26 08:21:36.948 Detail, Node008, Refreshed Value: old value=54.3, new value=0.0, type=decimal
2019-01-26 08:21:36.949 Detail, Node008, Changes to this value are not verified
2019-01-26 08:21:36.949 Detail, Node008, Notification: ValueChanged

Note that the value of SensorMultiLevel  is correct but the value of SwitchBinary is incorrect.

Fishwaldo, 2019-01-27:
If you refresh the value - is it correct?

if not, then nothing wrong with OZW here.... this is either a bug with the device, or "intentional" by Fibaro.... check the config options for the device and if nothing stands out, or open a case with Fibaro support.

If it is correct, then you need to enable polling on the device. Its obviously a bit too slow to update the current value.

closed 2019-02-10 by no-response[bot]