OpenZWave/open-zwave #1700: Greenwave Powernode 6

jduquennoy, 2019-01-26:
Hi OpenZWave community,
The ticket is about problems with measurements reported by libOpenZWave for a Greenwave PowerNode 6 device.

I found 2 tickets from 2015 referring to a powerNode device, but they did not answer that specific problem, thus the creation of that new ticket.

## Situation description

I am using HomeAssistant, that relies on libOpenZWave for ZWave handling.
On my zwave network, I have a powerNode6 device. This device has 6 plugs, each of them with an independent switch, and reporting measurements for power and energy.

## Issue description

Power measurements seems to be reported identically, whichever one of the 6 plugs sent the report.
Here are the logs I have from homeAssistant for two power events on two different plugs:

- power changing from 0 to 1.8W on plug 3:
```
[openzwave] zwcallback args=[{'homeId': 25482965, 'valueId': {'value': 1.7999999523162842, 'homeId': 25482965, 'units': 'W', 'index': 8, 'commandClass': 'COMMAND_CLASS_METER', 'readOnly': True, 'type': 'Decimal', 'instance': 1, 'id': 72057594093273218, 'nodeId': 3, 'genre': 'User', 'label': 'Power'}, 'notificationType': 'ValueChanged', 'nodeId': 3}]
```

- power changing from 0 to 9.4W on plug 1:
```
[openzwave] zwcallback args=[{'homeId': 25482965, 'valueId': {'value': 9.399999618530273, 'homeId': 25482965, 'units': 'W', 'index': 8, 'commandClass': 'COMMAND_CLASS_METER', 'readOnly': True, 'type': 'Decimal', 'instance': 1, 'id': 72057594093273218, 'nodeId': 3, 'genre': 'User', 'label': 'Power'}, 'notificationType': 'ValueChanged', 'nodeId': 3}]
```

Here is the node declaration in the zwave xml config file (a bit verbose, I couldn't find a way to fold it. Sorry for that). It seems to declare 6 instances (one per plug I guess)
```
	<Node id="3" name="" location="" basic="4" generic="16" specific="1" type="Binary Power Switch" listening="true" frequentListening="false" beaming="true" routing="true" max_baud_rate="40000" version="4" query_stage="Complete">
		<Manufacturer id="99" name="GreenWave">
			<Product type="3" id="4" name="PowerNode 6 port" />
		</Manufacturer>
		<CommandClasses>
			<CommandClass id="32" name="COMMAND_CLASS_BASIC" version="1" request_flags="5" innif="true" mapping="37">
				<Instance index="1" endpoint="1" />
				<Instance index="2" endpoint="2" />
				<Instance index="3" endpoint="3" />
				<Instance index="4" endpoint="4" />
				<Instance index="5" endpoint="5" />
				<Instance index="6" endpoint="6" />
			</CommandClass>
			<CommandClass id="37" name="COMMAND_CLASS_SWITCH_BINARY" version="1" request_flags="5" innif="true">
				<Instance index="1" endpoint="1" />
				<Instance index="2" endpoint="2" />
				<Instance index="3" endpoint="3" />
				<Instance index="4" endpoint="4" />
				<Instance index="5" endpoint="5" />
				<Instance index="6" endpoint="6" />
				<Value type="bool" genre="user" instance="1" index="0" label="Switch" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="True" />
				<Value type="bool" genre="user" instance="2" index="0" label="Switch" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="True" />
				<Value type="bool" genre="user" instance="3" index="0" label="Switch" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="True" />
				<Value type="bool" genre="user" instance="4" index="0" label="Switch" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="True" />
				<Value type="bool" genre="user" instance="5" index="0" label="Switch" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="True" />
				<Value type="bool" genre="user" instance="6" index="0" label="Switch" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="True" />
			</CommandClass>
			<CommandClass id="39" name="COMMAND_CLASS_SWITCH_ALL" version="1" request_flags="5" innif="true">
				<Instance index="1" />
				<Value type="list" genre="system" instance="1" index="0" label="Switch All" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" vindex="3" size="1">
					<Item label="Disabled" value="0" />
					<Item label="Off Enabled" value="1" />
					<Item label="On Enabled" value="2" />
					<Item label="On and Off Enabled" value="255" />
				</Value>
			</CommandClass>
			<CommandClass id="50" name="COMMAND_CLASS_METER" version="2" request_flags="3" innif="true">
				<Instance index="1" endpoint="1" />
				<Instance index="2" endpoint="2" />
				<Instance index="3" endpoint="3" />
				<Instance index="4" endpoint="4" />
				<Instance index="5" endpoint="5" />
				<Instance index="6" endpoint="6" />
[... 133 lines cut ...]
						<Node id="1" />
					</Group>
					<Group index="2" max_associations="1" label="Relay health" auto="false" />
					<Group index="3" max_associations="1" label="Power level" auto="false">
						<Node id="1" />
					</Group>
					<Group index="4" max_associations="1" label="Overcurrent protection" auto="false" />
				</Associations>
			</CommandClass>
			<CommandClass id="134" name="COMMAND_CLASS_VERSION" version="1" request_flags="5" innif="true">
				<Instance index="1" />
				<Value type="string" genre="system" instance="1" index="0" label="Library Version" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="3" />
				<Value type="string" genre="system" instance="1" index="1" label="Protocol Version" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="3.41" />
				<Value type="string" genre="system" instance="1" index="2" label="Application Version" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="4.28" />
			</CommandClass>
		</CommandClasses>
	</Node>
```

But as you can see, in the callback log traces, the instance is always 1.
Have you already heard about such a problem ?

Thanks for your attention,

Jerome

jduquennoy, 2019-01-29:
Hi Jean Frqnçois,

I have only one xml file with all the devices.
I added the line to the commandClasses element of the powernode 6 Node element.

I think I should mention that there was already an entry with this ID, with the mapping=endpoint element:
			<CommandClass id="96" name="COMMAND_CLASS_MULTI_INSTANCE/CHANNEL" version="3" request_flags="1" innif="true" mapping="endpoints">
				<Instance index="1" />
			</CommandClass>

I removed that entry, and replaced it with yours.

Sadly, it makes no change: the measurement reports are always on instance 1 :-(.

Jerome

> 
> Yes I already heard this problem.
> 
> can you try this:
> Modify your config/greenwave/powernode6.xml file, add this
> 
> 
> <CommandClass id="96" mapping="endpoints" />
> before the
> 
> exclude your device and include again.
> 
> check if you still get same behaviour.

Jérôme Duquennoy

------------------------------------
clef GPG : http://jerome.duquennoy.fr/gpg/jduquennoy.gpg
md5 : c0ff5a8bca033eade512374db18ecb38

sarakha63, 2019-02-01:
Nothing in xml will help

In fact 0x0d, 0x01, 0x04, 0x32

0d = encap
01 = source endpoint
04 = dest endpoint
32 = command class meter

And the device itself always report the good endpoint but in the destination bitmap. And source endpoint is always 01 that explians everything.

The only thing i see would be a new xml parameter saying sourceIsdest = 1 and handling that directly  

with a if in multiinstance.cpp 

at the end 

and replacing 

	uint8 endPoint = _data[1] & 0x7f;

by 

if sourceisDest

jduquennoy, 2019-02-16:
Hi back (and sorry for the delay !).
Thanks for this in depth investigation !
I will try to take some time to apply the changes to mention, and open a PR on the github if I can have it working correctly with all my devices.
I cannot make any promise though, especially not on a date: I feel like my days are shorter than 24 hours those day, it is hard to find available slots :-).

Thanks a lot !

Jerome

> Nothing in xml will help
>
> In fact 0x0d, 0x01, 0x04, 0x32
>
> 0d = encap
> 01 = source endpoint
> 04 = dest endpoint
> 32 = command class meter
>
> And the device itself always report the good endpoint but in [...]

Jérôme Duquennoy

------------------------------------
clef GPG : http://jerome.duquennoy.fr/gpg/jduquennoy.gpg
md5 : c0ff5a8bca033eade512374db18ecb38

jduquennoy, 2019-03-06:
I have the fix on a fork on the repo. I'm trying to find a way to integrate the fixed version of the lib in my home automation system to smoke test it with my different zwave devices.
As soon as I can confirm that it works as expected, I'll open a pull request.

Fishwaldo, 2019-03-18:
Please open a PR when you have a fix.

closed 2019-03-18 by Fishwaldo

jduquennoy, 2019-03-21:
Hi Fishwaldo,
I created a pull request for that ticket: https://github.com/OpenZWave/open-zwave/pull/1750.

cbho, 2020-09-18:
Hi,
I know this issue is closed - don't know if writing here or posting another issue is the best, but:

I see the exact same behaviour as desribed in the original post:
All sensor values are reported to instance 1 - even though I can read, that it should be fixed.

System:
qt-openzwave (All-In-One Image) (with OpenZWave Version 1.6.1185, qt-openzwave Version: 1.2.0, and QT Version: 5.12.8)
Home Assistant (HassOS 4.13, Home Assistant 0.115.0)

I'd be happy to post relevant logs etc., but as I'm not sure which will be useful?
Any advice will be greatly appreciated (@jduquennoy?)

Regards,
Chr.