OpenZWave/open-zwave #1707: Yale Keyfree with ZWave Plus/S2 module (blue)

Dinth, 2019-02-02:
A while ago i've made a simple PR for adding support for Yale Keyfree with S2 module: https://github.com/OpenZWave/open-zwave/pull/1619.
Unfortunately, while the device seems to be recognized after this PR and it's mostly supported, there's still few things missing compared to same lock with old Zwave non-Plus module - for example "lock status" sensor is missing and i believe that few configuration items are missing too.

Fishwaldo, 2019-03-05:
Logs?

Dinth, 2019-03-05:
Here's my whole OZW_Log.txt
The lock is Node040.

Fishwaldo, 2019-03-18:
For the current state of the Door, we don't support that atm. Only what state the lock is in. 

For missing configuration params: https://github.com/OpenZWave/open-zwave/wiki/Adding-Devices

norrland1, 2019-07-06:
@Fishwaldo Will there be support for pulling state from the locks? Most locks are used by other devices (eg BLE) and not knowing if a lock is locked or not is fatal for its usage. 

I believe it worked in older versions of open-zwave, Am I wrong? Which versions does support it?

Fishwaldo, 2019-07-10:
State was never reported... needs either someone with time to update the CC to the latest version, or I'll get to it eventually.

Dinth, 2019-10-19:
I dont think that #3397 fixed the issue. 

Im have just paired my lock again and it still doesnt work. Here is the log of pairing process from OZW: https://pastebin.com/Mz9fJR1H

Home Assistant (which im using for controlling OZW) errors:

```
Z-Wave entity Unknown Node 52 Battery Level (node_id: 52) not ready after 30 seconds, continuing anyway
9:37 AM components/zwave/__init__.py (WARNING)
Z-Wave entity Unknown Node 52 Locked (node_id: 52) not ready after 30 seconds, continuing anyway
9:36 AM components/zwave/__init__.py (WARNING)
Z-Wave node 52 not ready after 30 seconds, continuing anyway
9:36 AM components/zwave/__init__.py (WARNING)
```

petergebruers, 2019-10-19:
> I dont think that #3397 fixed the issue.

Why are you referring to an issue reported on Domoticz tracker? You say you are running "Home Assistant"?

> Home Assistant (which im using for controlling OZW) errors:

HASS is not based on "OpenZWave Master", it is based on a fork of an older version called OZW 1.4. Have a look at the first lines in your log, does it say something like:

```
OpenZwave Version 1.6-942
```

It probably does not, it very likely mentions: "1.4-..." which is an older version. Your previous log says: "OpenZwave Version 1.4.3322"

HASS compatibility with OZW 1.6 is "Work In Progress".