OpenZWave/open-zwave #1710: Remotec ZXT-120 and ZXT-600 not reporting temperature setting to application

austwhite, 2019-02-07:
### Issue Guidelines

Before opening a new issue, please review the following:
* The Github Issues is for OZW Bugs only. Please do not use this for general questions or howto type conversations. If you have a question, please post to our mailing list at https://groups.google.com/forum/#!forum/openzwave
* Please use the search feature to see if your issues have been raised or addressed first.
* Try the OpenZWave Log Analyzer at http://www.openzwave.com/log-analyzer if you are experiencing problems with a device on your network.
* Often we will require Log Files to help diagnose the issue. Please stop any application you are using, remove zwcfg_*.xml/ozwcache_*.xml files, and restart your application. After experiencing your issue, stop your application and attach the OZW_Log.txt file to this issue (please do not filter the log file, we often need the full log file to diagnose issues)
* If you want to add a new device to the OZW Database, please consult https://github.com/OpenZWave/open-zwave/wiki/Adding-Devices

I am not suer if this is the right place for this.
There seems to be an error in the configuration for both ZXT-120 and ZXT-600 (Remotec devices).
These are Zwave to IR controllers for HVAC units that use IR remote controls.
The devices both support the ThermostatSetpointCmd_Get command, but this option is set to false in OpenZWave.

There are also a couple of additional lines that show Fan Speed as "unknown 4" and "unknown 5" where "unknown 4" seems to be an "Auto" mode and "unknown 5" seems to be a "medium" speed mode.

If this is in the wrong place, please let me know.

Fishwaldo, 2019-03-05:
I’m not sure what you are asking?

austwhite, 2019-03-05:
Thanks for looking at this.
I am wondering if the configuration for the device is correct in OpenZWave.
If I change the options manually in the zwcfgxxxx.xml file then the device behaves as expected, but my understanding is that this file is a cache file and that OZW pulls the configuration data from other files when a device is included on the ZWave network.
It seems the configuration for the device may be incorrect, hence why it shows the 'unknown' options for Fan speeds and also has the "getsupported"option  at  id ="67" set as false, when it looks like it should be true.

Fishwaldo, 2019-03-18:
Please open a Pull Request with the fixes needed then.

closed 2019-03-18 by Fishwaldo