OpenZWave/open-zwave #1716: BeNext Energy Switch Plus produced energy

dodepier, 2019-02-19:
Hello,

I believe that the Energy Switch Plus (not the Energy Switch, see manual for difference) is not fully supported by OZW. The produced energy is not reported in home-assistant. I believe this is due to some missing configs in OZW. 

This is in the manual:
> Meter V4 cmd class details for third party controllers:
> The meter cmd class exists out of a rate type, meter type, scale, precision and size. The rate type for import (consumed) = 0x01 and for export (produced) = 0x02 Meter type is 0x01 = Electricity meter
> Scale is 0x00 = kwh and 0x02 is Watt
> **All energy measured for solar production will be reported/requested on rate type 0x02.**

Manual:
https://www.benext.eu/static/manual/energyswitchrs.pdf 

This is in the zwcfg_*.xml of homeassistant I miss a line where the produced energy read:
```
			<CommandClass id="50" name="COMMAND_CLASS_METER" version="4" request_flags="2" innif="true">
				<Instance index="1" />
				<Value type="decimal" genre="user" instance="1" index="0" label="Energy" units="kWh" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="0.0246" />
				<Value type="decimal" genre="user" instance="1" index="1" label="Previous Reading" units="kWh" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="0.0246" />
				<Value type="int" genre="user" instance="1" index="2" label="Interval" units="seconds" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="-2147483648" max="2147483647" value="1200" />
				<Value type="decimal" genre="user" instance="1" index="8" label="Power" units="W" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="0.0" />
				<Value type="bool" genre="user" instance="1" index="32" label="Exporting" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="True" />
				<Value type="button" genre="system" instance="1" index="33" label="Reset" units="" read_only="false" write_only="true" verify_changes="false" poll_intensity="0" min="0" max="0" />
			</CommandClass>
```

Fishwaldo, 2019-03-05:
I’m not sure what your asking?

dodepier, 2019-03-05:
Hi,

I would like to add the produced energy now it is only reporting used energy however the Energy Switch does also report also production see manual. Can I used add this to the config with the information of the manual? something like this:

```
			<CommandClass id="50" name="COMMAND_CLASS_METER" version="4" request_flags="2" innif="true">
				<Instance index="1" />
				<Value type="decimal" genre="user" instance="1" index="0" label="Energy" units="kWh" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="0.0246" />
				<Value type="decimal" genre="user" instance="1" index="1" label="Previous Reading" units="kWh" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="0.0246" />
				<Value type="int" genre="user" instance="1" index="2" label="Interval" units="seconds" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="-2147483648" max="2147483647" value="1200" />
				<Value type="decimal" genre="user" instance="1" index="8" label="Power" units="W" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="0.0" />
				<Value type="bool" genre="user" instance="1" index="32" label="Exporting" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="True" />
				<Value type="button" genre="system" instance="1" index="33" label="Reset" units="" read_only="false" write_only="true" verify_changes="false" poll_intensity="0" min="0" max="0" />
                                <Instance index="2" />
				<Value type="decimal" genre="user" instance="2" index="0" label="Production Energy" units="kWh" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="0" />
				<Value type="decimal" genre="user" instance="2" index="1" label="Production Previous Reading" units="kWh" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="0" />
				<Value type="int" genre="user" instance="2" index="2" label="Production Interval" units="seconds" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="-2147483648" max="2147483647" value="1200" />
			</CommandClass>
```

Fishwaldo, 2019-03-18:
Would need to see the OZW log files. Please shutdown your app, remove zwcfg_*.xml and restart. Then send the logs over.

dodepier, 2019-04-08:
Sorry for the late response see log. Energy Switch Plus is connected with node 20 
[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/3055503/OZW_Log.txt)

Fishwaldo, 2019-04-12:
Its already there. The Exporting ValueID is true, and index 8 contains the produced power:

> 2019-04-08 19:05:11.130 Info, Node021, Sending (Send) message (Callback ID=0x74, Expected Reply=0x04) - MeterCmd_Get (Node=21): 0x01, 0x0a, 0x00, 0x13, 0x15, 0x03, 0x32, 0x01, 0x10, 0x25, 0x74, 0x82
2019-04-08 19:05:11.139 Detail, Node021,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2019-04-08 19:05:11.141 Detail, Node021,   ZW_SEND_DATA delivered to Z-Wave stack
2019-04-08 19:05:11.158 Detail, Node021,   Received: 0x01, 0x05, 0x00, 0x13, 0x74, 0x00, 0x9d
2019-04-08 19:05:11.158 Detail, Node021,   ZW_SEND_DATA Request with callback ID 0x74 received (expected 0x74)
2019-04-08 19:05:11.158 Info, Node021, Request RTT 27 Average Request RTT 34
2019-04-08 19:05:11.158 Detail,   Expected callbackId was received
2019-04-08 19:05:11.175 Detail, Node021,   Received: 0x01, 0x12, 0x00, 0x04, 0x00, 0x15, 0x0a, 0x32, 0x02, 0x41, 0x34, 0x00, 0x00, 0x00, 0x6e, 0x00, 0x00, 0xa5, 0x00, 0x78
2019-04-08 19:05:11.175 Detail, 
2019-04-08 19:05:11.175 Info, Node021, Response RTT 44 Average Response RTT 87
2019-04-08 19:05:11.175 Detail, Node021, Refreshed Value: old value=false, new value=true, type=bool
2019-04-08 19:05:11.175 Detail, Node021, Changes to this value are not verified
2019-04-08 19:05:11.175 Info, Node021, Received Meter report from node 21: Exporting Power=11.0W
2019-04-08 19:05:11.175 Detail, Node021, Initial read of value
2019-04-08 19:05:11.175 Detail, Node021,   Expected reply and command class was received
2019-04-08 19:05:11.175 Detail, Node021,   Message transaction complete

As the manual states - it only exposes this information as Power, not KW/h

closed 2019-04-12 by Fishwaldo