OpenZWave/open-zwave #1720: Segmentation Fault: Can't Find TimerEvent to Delete in TimerDelEvent

hoschid, 2019-02-24:
I have the following problem with the latest version of Openzwave from the dev branch (commit e991af5). Openzwave crashes with a **segmentation fault** when I press and hold any button of my Aeotec WallMote Quad:

```
zwave:app:debug node 19: value changed. value_id=19-91-1-1, value=Key Held down, old value=Inactive +9s
m2019-01-25 13:28:49.327 Detail, Node019,   Received: 0x01, 0x08, 0x00, 0x04, 0x00, 0x13, 0x02, 0x98, 0x40, 0x3a
2019-01-25 13:28:49.327 Info, Node019, Received SecurityCmd_NonceGet from node 19
2019-01-25 13:28:49.327 Info, NONCES: 0xe2, 0x2f, 0x28, 0x80, 0xd4, 0xac, 0xe7, 0x30
2019-01-25 13:28:49.327 Info, NONCES: 0x65, 0xb4, 0xde, 0x8c, 0xbd, 0xee, 0x3b, 0xed
2019-01-25 13:28:49.327 Info, NONCES: 0x8d, 0xee, 0x7e, 0x8d, 0xf0, 0xcc, 0xd0, 0x98
2019-01-25 13:28:49.327 Info, NONCES: 0xa0, 0xd6, 0x4e, 0x38, 0x32, 0x9c, 0x1c, 0xac
2019-01-25 13:28:49.327 Info, NONCES: 0x88, 0x27, 0x8e, 0x03, 0x61, 0x61, 0x4e, 0xbc
2019-01-25 13:28:49.327 Info, NONCES: 0xa5, 0x01, 0x88, 0xd8, 0x9e, 0xa4, 0x84, 0x66
2019-01-25 13:28:49.327 Info, NONCES: 0x30, 0x9e, 0xb3, 0x91, 0x00, 0x01, 0x4e, 0x42
2019-01-25 13:28:49.327 Info, NONCES: 0xa8, 0xdb, 0x2e, 0x57, 0xaa, 0xe1, 0xa7, 0x50
2019-01-25 13:28:49.327 Info, Node019, Sending (WakeUp) message (Callback ID=0x01, Expected Reply=0x00) - Nonce_Report - 0x01, 0x11, 0x00, 0x13, 0x13, 0x0a, 0x98, 0x80, 0x8d, 0xee, 0x7e, 0x8d, 0xf0, 0xcc, 0xd0, 0x98, 0x05, 0x01, 0x1c:
2019-01-25 13:28:49.332 Detail,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2019-01-25 13:28:49.332 Detail,   ZW_SEND_DATA delivered to Z-Wave stack
2019-01-25 13:28:49.349 Detail,   Received: 0x01, 0x07, 0x00, 0x13, 0x01, 0x00, 0x00, 0x02, 0xe8
2019-01-25 13:28:49.349 Detail,   ZW_SEND_DATA Request with callback ID 0x01 received (expected 0x01)
2019-01-25 13:28:49.367 Detail, Node019,   Received: 0x01, 0x1f, 0x00, 0x04, 0x00, 0x13, 0x19, 0x98, 0x81, 0x4f, 0xcb, 0x2c, 0x2b, 0xdf, 0x57, 0x3a, 0xe4, 0xf0, 0x78, 0x4d, 0x99, 0x5b, 0x1a, 0x8d, 0x62, 0x9d, 0x0f, 0x77, 0x84, 0x38, 0x92, 0x38, 0x23
2019-01-25 13:28:49.367 Info, Raw: 0x98, 0x81, 0x4f, 0xcb, 0x2c, 0x2b, 0xdf, 0x57, 0x3a, 0xe4, 0xf0, 0x78, 0x4d, 0x99, 0x5b, 0x1a, 0x8d, 0x62, 0x9d, 0x0f, 0x77, 0x84, 0x38, 0x92, 0x38, 0x23
2019-01-25 13:28:49.367 Detail, Node019, Decrypted Packet: 0x00, 0x5b, 0x03, 0x63, 0x02, 0x01
2019-01-25 13:28:49.367 Detail, 
2019-01-25 13:28:49.367 Info, Node019, Received Central Scene set from node 19: scene id=1 with key Attribute 2. Sending event notification.
2019-01-25 13:28:49.367 Detail, Node019, Refreshed Value: old value=3, new value=3, type=list
2019-01-25 13:28:49.367 Detail, Node019, Changes to this value are not verified
2019-01-25 13:28:49.367 Info, Node019, Automatically Clearing Scene 1 in 1000ms
2019-01-25 13:28:49.368 Warning, Cant Find TimerEvent to Delete in TimerDelEvent
11336 segmentation fault (core dumped)
```

When I press the button only shortly everything works fine:

```
2019-01-25 13:28:39.077 Detail, Node019,   Received: 0x01, 0x08, 0x00, 0x04, 0x00, 0x13, 0x02, 0x98, 0x40, 0x3a
2019-01-25 13:28:39.077 Info, Node019, Received SecurityCmd_NonceGet from node 19
2019-01-25 13:28:39.078 Info, NONCES: 0xe2, 0x2f, 0x28, 0x80, 0xd4, 0xac, 0xe7, 0x30
2019-01-25 13:28:39.078 Info, NONCES: 0x1f, 0x96, 0x94, 0x87, 0x98, 0x5c, 0x4d, 0xe3
2019-01-25 13:28:39.078 Info, NONCES: 0x12, 0x19, 0xec, 0x2b, 0x7b, 0x39, 0xd3, 0x4a
2019-01-25 13:28:39.078 Info, NONCES: 0xa0, 0xd6, 0x4e, 0x38, 0x32, 0x9c, 0x1c, 0xac
2019-01-25 13:28:39.078 Info, NONCES: 0x88, 0x27, 0x8e, 0x03, 0x61, 0x61, 0x4e, 0xbc
2019-01-25 13:28:39.078 Info, NONCES: 0xa5, 0x01, 0x88, 0xd8, 0x9e, 0xa4, 0x84, 0x66
2019-01-25 13:28:39.078 Info, NONCES: 0x30, 0x9e, 0xb3, 0x91, 0x00, 0x01, 0x4e, 0x42
2019-01-25 13:28:39.078 Info, NONCES: 0xa8, 0xdb, 0x2e, 0x57, 0xaa, 0xe1, 0xa7, 0x50
2019-01-25 13:28:39.078 Info, Node019, Sending (WakeUp) message (Callback ID=0x01, Expected Reply=0x00) - Nonce_Report - 0x01, 0x11, 0x00, 0x13, 0x13, 0x0a, 0x98, 0x80, 0xe2, 0x2f, 0x28, 0x80, 0xd4, 0xac, 0xe7, 0x30, 0x05, 0x01, 0x32:
2019-01-25 13:28:39.083 Detail,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2019-01-25 13:28:39.083 Detail,   ZW_SEND_DATA delivered to Z-Wave stack
2019-01-25 13:28:39.100 Detail,   Received: 0x01, 0x07, 0x00, 0x13, 0x01, 0x00, 0x00, 0x02, 0xe8
2019-01-25 13:28:39.100 Detail,   ZW_SEND_DATA Request with callback ID 0x01 received (expected 0x01)
2019-01-25 13:28:39.117 Detail, Node019,   Received: 0x01, 0x1f, 0x00, 0x04, 0x00, 0x13, 0x19, 0x98, 0x81, 0x3f, 0x2c, 0xbf, 0xf4, 0x10, 0x66, 0xba, 0xc4, 0xb2, 0x36, 0x03, 0xbf, 0x8d, 0x63, 0xe2, 0xcc, 0xb4, 0x0c, 0xe3, 0x39, 0x0e, 0xc8, 0x05, 0xfe
2019-01-25 13:28:39.118 Info, Raw: 0x98, 0x81, 0x3f, 0x2c, 0xbf, 0xf4, 0x10, 0x66, 0xba, 0xc4, 0xb2, 0x36, 0x03, 0xbf, 0x8d, 0x63, 0xe2, 0xcc, 0xb4, 0x0c, 0xe3, 0x39, 0x0e, 0xc8, 0x05, 0xfe
2019-01-25 13:28:39.118 Detail, Node019, Decrypted Packet: 0x00, 0x5b, 0x03, 0x61, 0x00, 0x02
2019-01-25 13:28:39.118 Detail, 
2019-01-25 13:28:39.118 Info, Node019, Received Central Scene set from node 19: scene id=2 with key Attribute 0. Sending event notification.
2019-01-25 13:28:39.118 Detail, Node019, Refreshed Value: old value=0, new value=1, type=list
2019-01-25 13:28:39.118 Detail, Node019, Changes to this value are not verified
2019-01-25 13:28:39.118 Info, Node019, Automatically Clearing Scene 2 in 1000ms
2019-01-25 13:28:39.118 Info, Timer: adding event in 1000 ms
2019-01-25 13:28:39.118 Detail, Timer: waiting with timeout 1000 ms
2019-01-25 13:28:39.118 Detail, Node019, Notification: ValueChanged
zwave:app:debug node 19: value changed. value_id=19-91-1-2, value=Pressed 1 Time, old value=Inactive +2s
m2019-01-25 13:28:40.118 Info, Timer: delayed event
2019-01-25 13:28:40.118 Detail, Node019, Refreshed Value: old value=1, new value=0, type=list
2019-01-25 13:28:40.118 Detail, Node019, Changes to this value are not verified
2019-01-25 13:28:40.118 Detail, Timer: waiting with timeout -1 ms
2019-01-25 13:28:40.119 Detail, Node019, Notification: ValueChanged
zwave:app:debug node 19: value changed. value_id=19-91-1-2, value=Inactive, old value=Pressed 1 Time +1s
```

Fishwaldo, 2019-03-05:
Thanks. I’ll look into this soon. Probably a silly typo somewhere.

dhylands, 2019-03-14:
I ran into the same problem. The issue is in this function:https://github.com/OpenZWave/open-zwave/blob/e991af56e429947e1fe556cd889b1c4326869a42/cpp/src/TimerThread.cpp#L308-L315

Calling m_timerList.erase invalidates the list, but the for loop continues to try to use it.

If there is only one entry in the list with the given id, then putting a break after the call to erase would be fine. Otherwise this needs to be converted into a while loop something like this:
```
		list<TimerThread::TimerEventEntry *>::iterator it = m_timerEventList.begin();
		while( it != m_timerEventList.end() ) {
			if ((*it)->id == id) {
				m_driver->GetTimer()->TimerDelEvent((*it));
				it = m_timerEventList.erase(it);
			} else {
				++it;
			}
		}
```

closed 2019-03-18 by Fishwaldo