OpenZWave/open-zwave #1724: -21474836.480 kWh value

Beddie, 2019-02-28:
When using NAS-WR01ZE (type=0200) device kWh readings are not correct, see image below:

[image]

After device inclusion:
kWh = 0.000 kWh
After some (undefined) time (the same result on 4 different devices):
kWh = -21474836.480 kWh

RicardP, 2019-03-02:
I have the same issues with nodes experience bad signal/link/mesh quality. Also other issues can be that those nodes suddenly shows a new "ghost" CC, ghost because the CC is not supported by the specific node. It looks like CRC check of packets does not work as expected, but I am not into coding so it is just a guess :)

Beddie, 2019-03-03:
More likely device related, see device manual (page 4)
"The max accumulate energy is 21474836.47kWh, if it is over this value, it will be back to 0kWh automatically."
[nas-wr01z.S2.SmartPlug.Manual.pdf](https://github.com/OpenZWave/open-zwave/files/2923060/nas-wr01z.S2.SmartPlug.Manual.pdf)

RicardP, 2019-03-04:
This negative, obvious corrupt, value shown in OZW is not grabbed "as is" from the Node. It was most likely sent OK but went corrupt on its travel to the controller or by OZW bug.

Sometimes (most times) Performing "Refresh Node Info" will restore all Node data, except presence of ghost CCs that needs removal of the specific Node in zwcfg_xxx.xml (stop application using OZW if editing).

Fishwaldo, 2019-03-05:
How do you deduce it’s corrupted?

As mentioned in the ghost CC issue, so far the only person that has managed to get logs from this shows that the device sent a perfectly valid Z-Wave message (so the big was in the firmware and not OZW). I’ve verified the CRC code and it’s working so I’m inclined to say this is what the device send us. (We don’t “interpret” anything in OZW, we present the values as OZW recieves them from the network)

Fishwaldo, 2019-03-05:
Please provide log files for me to check

RicardP, 2019-03-05:
Hi Fishwaldo, I found this searching on zwave and CRC 

https://forum.z-wavepublic.com/t/use-of-crc16-class-encoded-messages/115

I think there is an interesting writing in the last post, about 40KBaud issues and "funny" readings, and my own experiences with "funny" readings within OZW is from Slave Nodes. 

Many thanks for your support!!

Fishwaldo, 2019-03-05:
That’s the CRC16 command class, which I think I’ve only ever seen one device implement.  It’s not the same as the CRC checksum appended to each message sent/recieved. 

The Ghost CC issue is “interesting” in that the packet that created a new CC on a node is complex, and with several fields that must have specific values/formats. Getting random corruption that so perfectly represents a new CC is like winning the lottery!

As mentioned, only 1 person has been able to provide logs that show this so far, so until I can see more logs, I’m going with the “device firmware bug” and possibly even a bug in the Sigma SDK. (Which would explain why different devices from manufacturers show this behavior)

In this case - as mentioned, we would need to see logs to deduce why this is happening.

Beddie, 2019-03-06:
@Fishwaldo I did look into my log files but saw nothing abnormal, then decided after reading about open-zwave to throw away my zwcfg_XXXX.xml. This solved it for me. If you have an answer or a reference where i can find the answer to the solution i am interested. For now i am glad it is solved!

closed 2019-03-06 by Beddie