OpenZWave/open-zwave #1744: Fibaro Roller Shutter 2: statuses update

0rsa, 2019-03-15:
Hello,

Using domoticz, I have issues with my 8 Fibaro roller shutter 2.
Sometimes it works well, sometimes, fibaro module doesn't send an update status when his move is done.

Loot at this example in domoticz log files:
I send 4 commands to open 4 differents blinds (blinds are completely closed):

2019-03-15 07:31:31.570 Status: User: Admin initiated a switch command (197/Volet salle de bain/Set Level)
2019-03-15 07:31:31.574 Status: User: Admin initiated a switch command (218/Volet salle de jeux/Set Level)
2019-03-15 07:31:31.575 Status: LUA: Device based event fired on 'Volet salle de bain', value 'Open'
2019-03-15 07:31:31.578 Status: User: Admin initiated a switch command (226/Volet chambre/Set Level)
2019-03-15 07:31:31.579 Status: LUA: Device based event fired on 'Volet salle de jeux', value 'Open'
2019-03-15 07:31:31.583 Status: User: Admin initiated a switch command (221/Volet chambre amis/Set Level)
2019-03-15 07:31:31.583 Status: LUA: Device based event fired on 'Volet chambre', value 'Open'
2019-03-15 07:31:31.584 Status: LUA: Device based event fired on 'Volet chambre amis', value 'Open'
2019-03-15 07:31:31.643 Status: LUA: Device based event fired on 'Volet salle de bain', value 'Closed'
2019-03-15 07:31:31.826 Status: LUA: Device based event fired on 'Volet salle de jeux', value 'Closed'

-----------------------------

2 modules send immediately a status update:

2019-03-15 07:31:32.277 Status: LUA: Device based event fired on 'Volet chambre', value 'Closed'
2019-03-15 07:31:32.561 Status: LUA: Device based event fired on 'Volet chambre amis', value 'Closed'

---------------------------

Only one send a status update when it's done (opened):

2019-03-15 07:31:53.420 Status: LUA: Device based event fired on 'Volet chambre', value 'Open'

-> It's totally random, this is never the same blind which don't send a correct update.

0rsa, 2019-03-15:
additional info:
- I use OpenZWave 1.4-3402-g6bf498ce-dirty
- Using domoticz OpenZWave Control Panel, values are correct if I force a refresh of nodes.

0rsa, 2019-03-15:
I use UZB from zave.me.
Today I upgraded firmware from 5.25 to 5.36 (after a lot of pain), I reseted and reassociated all my modules. I will see if it's better in coming days and I will do a report here.

RicardP, 2019-03-15:
There is a lot of reports of nodes values getting mixed up and/or corrupt. I see it randomly on my nodes having the relative lowest reception/transmission quality. I'm afraid your FW update wont help, but please keep us posted.

0rsa, 2019-03-15:
@RicardP You're right.
I just closed my blinds tonight, 4 / 8 have a wrong update status (still opened)

0rsa, 2019-03-15:
I will try to add 1s delay between orders.

Fishwaldo, 2019-03-18:
Need to see the OZW logs.

closed 2019-04-01 by no-response[bot]