OpenZWave/open-zwave #1745: Many duplicated list items showing up in ozwcache file

dhylands, 2019-03-15:
While running on the Dev branch (e991af56e429947e1fe556cd889b1c4326869a42), I've noticed that many of the config lists in the ozwcache file have duplicated items. For example (the corresponding config file is config/linear/WA00Z-1.xml
```
                        <CommandClass id="115" name="COMMAND_CLASS_POWERLEVEL" version="1" request_flags="4" innif="true">
                                <Instance index="1" />
...snip...
                                <Value type="list" genre="system" instance="1" index="8" label="Test Status" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" vindex="0" size="1">
                                        <Help>The Current Status of the last PowerNode Test Executed</Help>
                                        <Item label="Failed" value="0" />
                                        <Item label="Success" value="1" />
                                        <Item label="In Progress" value="2" />
                                        <Item label="Failed" value="0" />
                                        <Item label="Success" value="1" />
                                        <Item label="In Progress" value="2" />
                                        <Item label="Failed" value="0" />
                                        <Item label="Success" value="1" />
                                        <Item label="In Progress" value="2" />
                                        <Item label="Failed" value="0" />
                                        <Item label="Success" value="1" />
                                        <Item label="In Progress" value="2" />
                                </Value>
```
or this one (the correponding config file is config/aeotec/zw130.xml)
```
                        <CommandClass id="112" name="COMMAND_CLASS_CONFIGURATION" version="1" request_flags="5" innif="true">
                                <Instance index="1" />
                                <Value type="list" genre="config" instance="1" index="1" label="Touch sound" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="1" vindex="1" size="1">
                                        <Help>This will enable or disable the sound effects when you press or touch the sensing area</Help>
                                        <Item label="Disable" value="0" />
                                        <Item label="Enable" value="1" />
                                        <Item label="Disable" value="0" />
                                        <Item label="Enable" value="1" />
                                </Value>
                                <Value type="list" genre="config" instance="1" index="2" label="Touch vibration" units="" read_only="false" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="1" vindex="1" size="1">
                                        <Help>This will enable or disable the vibration effects when you press or touch the sensing area</Help>
                                        <Item label="Disable" value="0" />
                                        <Item label="Enable" value="1" />
                                        <Item label="Disable" value="0" />
                                        <Item label="Enable" value="1" />
                                </Value>
```
I haven't found any particular pattern yet as to what might cause this.

I've never seen this issue on master with the zwcfg file.

chilicheech, 2019-03-17:
I noticed the same thing. It's been happening for months now but I haven't done any debugging.

lukescott, 2019-03-18:
I've also noticed that some labels or help entires are also wrong, like they got picked up from a different device. Some sort of pointer issue?

Fishwaldo, 2019-03-18:
Only for List ValueID’s? It’s probably related to the localization work.

RicardP, 2019-03-18:
I've seen same duplicates in the past, two years ago, on mainly nodes having bad RF properties, but it was also connected to a slow Raspberry running Domoticz. It seemed that re-transmits and slow hardware grown the ozwcache with duplicates, that caused bigger an even bigger/trashier ozwcache and led to unresponsive system.
When I moved my Domoticz Backup and Z-USB Stick to Windows platform the duplicates problem was gone. However, since I still get some false CCs and corrupt data on weak signal/meshed nodes.
Just sharing my experience I don't know if it is providing any useful info though... :)

dhylands, 2019-03-18:
For me, I'm running on my development linux laptop, which is reasonably powerful and all of the devices are withina few feet (less than six) from the dongle. So I doubt that RF signal strength is an issue. I've seen 5 or six duplicates show up. This is all using a fairly recent version of the Dev branch (i.e. I only started using the Dev branch in the past couple weeks). For now, I just filter the duplicates in my application.

Fishwaldo, 2019-03-28:
Fixed. Was due to the Localization Work. Thanks

closed 2019-03-28 by Fishwaldo