OpenZWave/open-zwave #1746: Devolo Smoke Detector MT2651 Wrong ID

streilu, 2019-03-16:
https://github.com/OpenZWave/open-zwave/blob/c7420ef8dfb9706bf0aca2a47111c7dc3897bde2/config/manufacturer_specific.xml#L516

I was just struggling with my Devolo MT2651 smoke detectors and found out that there's the wrong product-id in the manufacturer_specific.xml.
It should be "04a4" instead of "2001".

See screenshot for reference.

BR Luke
[image]

nechry, 2019-03-16:
it can be another version, I have the 2004 version in my hand and this is the right id. I will add your as same product

nechry, 2019-03-16:
close https://github.com/OpenZWave/open-zwave/commit/986c8f3f4446a40b3eeab7018867e13f5f66ae1b

closed 2019-03-16 by nechry