OpenZWave/open-zwave #1755: ZW116 Nano Switch gets double responses after changing switch state

sjorge, 2019-03-30:
I was suggested to open an issue upstream for https://github.com/robertsLando/Zwave2Mqtt/issues/12

My ZW116 seems to get multiple results after changing the switch state.
- immediately after with the old value
- one with a small delay (after the physical switch has been updates?) with the correct new value

Although the bug could also be in node-openzwave-shared... attached is the log without any zwcfg/zwscene xml.

```
2019-03-30 13:34:56.378 Always, OpenZwave Version 1.4.3405 Starting Up
2019-03-30 13:34:56.378 Info, Setting Up Provided Network Key for Secure Communications
2019-03-30 13:34:56.379 Info, mgr,     Added driver for controller /dev/cua/0
2019-03-30 13:34:56.379 Info,   Opening controller /dev/cua/0
2019-03-30 13:34:56.379 Info, Trying to open serial port /dev/cua/0 (attempt 1)
2019-03-30 13:34:56.380 Info, Serial port /dev/cua/0 opened (attempt 1)
2019-03-30 13:34:56.381 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9
2019-03-30 13:34:56.381 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_MEMORY_GET_ID: 0x01, 0x03, 0x00, 0x20, 0xdc
2019-03-30 13:34:56.381 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES: 0x01, 0x03, 0x00, 0x05, 0xf9
2019-03-30 13:34:56.381 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_GET_CAPABILITIES: 0x01, 0x03, 0x00, 0x07, 0xfb
2019-03-30 13:34:56.381 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_SUC_NODE_ID: 0x01, 0x03, 0x00, 0x56, 0xaa
2019-03-30 13:34:56.381 Detail,
2019-03-30 13:34:56.381 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x15) - FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9
2019-03-30 13:34:56.384 Detail, contrlr,   Received: 0x01, 0x10, 0x01, 0x15, 0x5a, 0x2d, 0x57, 0x61, 0x76, 0x65, 0x20, 0x36, 0x2e, 0x30, 0x32, 0x00, 0x01, 0x92
2019-03-30 13:34:56.384 Detail,
2019-03-30 13:34:56.384 Info, contrlr, Received reply to FUNC_ID_ZW_GET_VERSION:
2019-03-30 13:34:56.384 Info, contrlr,     Static Controller library, version Z-Wave 6.02
2019-03-30 13:34:56.384 Detail, Node045,   Expected reply was received
2019-03-30 13:34:56.384 Detail, Node045,   Message transaction complete
2019-03-30 13:34:56.384 Detail,
2019-03-30 13:34:56.384 Detail, contrlr, Removing current message
2019-03-30 13:34:56.384 Detail,
2019-03-30 13:34:56.384 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x20) - FUNC_ID_ZW_MEMORY_GET_ID: 0x01, 0x03, 0x00, 0x20, 0xdc
2019-03-30 13:34:56.387 Detail, contrlr,   Received: 0x01, 0x08, 0x01, 0x20, 0xfc, 0x20, 0xd0, 0x90, 0x01, 0x4b
2019-03-30 13:34:56.387 Detail,
2019-03-30 13:34:56.387 Info, contrlr, Received reply to FUNC_ID_ZW_MEMORY_GET_ID. Home ID = 0xfc20d090.  Our node ID = 1
2019-03-30 13:34:56.387 Detail, Node032,   Expected reply was received
2019-03-30 13:34:56.387 Detail, Node032,   Message transaction complete
2019-03-30 13:34:56.387 Detail,
2019-03-30 13:34:56.387 Detail, contrlr, Removing current message
2019-03-30 13:34:56.387 Detail,
2019-03-30 13:34:56.387 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x05) - FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES: 0x01, 0x03, 0x00, 0x05, 0xf9
2019-03-30 13:34:56.389 Detail, contrlr,   Received: 0x01, 0x04, 0x01, 0x05, 0x1c, 0xe3
2019-03-30 13:34:56.389 Detail,
2019-03-30 13:34:56.389 Info, contrlr, Received reply to FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES:
2019-03-30 13:34:56.389 Info, contrlr,     There is a SUC ID Server (SIS) in this network.
2019-03-30 13:34:56.389 Info, contrlr,     The PC controller is an inclusion static update controller (SUC) and was the original primary before the SIS was added.
2019-03-30 13:34:56.389 Detail, Node227,   Expected reply was received
2019-03-30 13:34:56.389 Detail, Node227,   Message transaction complete
2019-03-30 13:34:56.389 Detail,
2019-03-30 13:34:56.389 Detail, contrlr, Removing current message
2019-03-30 13:34:56.389 Detail,
2019-03-30 13:34:56.389 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x07) - FUNC_ID_SERIAL_API_GET_CAPABILITIES: 0x01, 0x03, 0x00, 0x07, 0xfb
2019-03-30 13:34:56.393 Detail, contrlr,   Received: 0x01, 0x2b, 0x01, 0x07, 0x05, 0x24, 0x01, 0x15, 0x04, 0x00, 0x00, 0x01, 0xfe, 0x87, 0x7f, 0x88, 0xcf, 0x3f, 0xc0, 0x47, 0xfb, 0xdf, 0xfd, 0xe0, 0x67, 0x00, 0x80, 0x80, 0x00, 0x80, 0x86, 0x00, 0x00, 0x00, 0xe8, 0x73, 0x00, 0x80, 0x0f, 0x00, 0x00, 0x60, 0x5a, 0x00, 0x6d
2019-03-30 13:34:56.393 Detail,
2019-03-30 13:34:56.393 Info, contrlr,  Received reply to FUNC_ID_SERIAL_API_GET_CAPABILITIES
2019-03-30 13:34:56.393 Info, contrlr,     Serial API Version:   5.36
2019-03-30 13:34:56.393 Info, contrlr,     Manufacturer ID:      0x0115
2019-03-30 13:34:56.393 Info, contrlr,     Product Type:         0x0400
2019-03-30 13:34:56.393 Info, contrlr,     Product ID:           0x0001
2019-03-30 13:34:56.394 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_RANDOM: 0x01, 0x04, 0x00, 0x1c, 0x20, 0xc7
2019-03-30 13:34:56.394 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_GET_INIT_DATA: 0x01, 0x03, 0x00, 0x02, 0xfe
2019-03-30 13:34:56.394 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_SET_TIMEOUTS: 0x01, 0x05, 0x00, 0x06, 0x64, 0x0f, 0x97
2019-03-30 13:34:56.394 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_APPL_NODE_INFORMATION: 0x01, 0x07, 0x00, 0x03, 0x01, 0x02, 0x01, 0x00, 0xf9
2019-03-30 13:34:56.394 Detail, Node036,   Expected reply was received
2019-03-30 13:34:56.394 Detail, Node036,   Message transaction complete
2019-03-30 13:34:56.394 Detail,
2019-03-30 13:34:56.394 Detail, contrlr, Removing current message
2019-03-30 13:34:56.394 Detail,
2019-03-30 13:34:56.394 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x56) - FUNC_ID_ZW_GET_SUC_NODE_ID: 0x01, 0x03, 0x00, 0x56, 0xaa
2019-03-30 13:34:56.396 Detail, contrlr,   Received: 0x01, 0x04, 0x01, 0x56, 0x01, 0xad
[... 2156 lines cut ...]
2019-03-30 13:35:46.567 Detail, Node004,   ZW_SEND_DATA delivered to Z-Wave stack
2019-03-30 13:35:46.586 Detail, Node004,   Received: 0x01, 0x18, 0x00, 0x13, 0x4f, 0x00, 0x00, 0x02, 0x00, 0xc1, 0x7f, 0x7f, 0x7f, 0x7f, 0x00, 0x00, 0x03, 0x00, 0x00, 0x00, 0x00, 0x03, 0x01, 0x00, 0x00, 0x79
2019-03-30 13:35:46.586 Detail, Node004,   ZW_SEND_DATA Request with callback ID 0x4f received (expected 0x4f)
2019-03-30 13:35:46.586 Info, Node004, Request RTT 26 Average Request RTT 26
2019-03-30 13:35:46.586 Detail,   Expected callbackId was received
2019-03-30 13:35:46.599 Detail, Node004,   Received: 0x01, 0x0c, 0x00, 0x04, 0x00, 0x04, 0x04, 0x81, 0x06, 0xce, 0x23, 0xbf, 0x00, 0x22
2019-03-30 13:35:46.599 Detail,
2019-03-30 13:35:46.599 Info, Node004, Response RTT 39 Average Response RTT 40
2019-03-30 13:35:46.599 Info, Node004, Received Clock report: Saturday 14:35
2019-03-30 13:35:46.599 Detail, Node004, Refreshed Value: old value=5, new value=5, type=list
2019-03-30 13:35:46.599 Detail, Node004, Changes to this value are not verified
2019-03-30 13:35:46.599 Detail, Node004, Refreshed Value: old value=14, new value=14, type=byte
2019-03-30 13:35:46.599 Detail, Node004, Changes to this value are not verified
2019-03-30 13:35:46.599 Detail, Node004, Refreshed Value: old value=35, new value=35, type=byte
2019-03-30 13:35:46.599 Detail, Node004, Changes to this value are not verified
2019-03-30 13:35:46.599 Detail, Node004,   Expected reply and command class was received
2019-03-30 13:35:46.599 Detail, Node004,   Message transaction complete
2019-03-30 13:35:46.599 Detail,
2019-03-30 13:35:46.599 Detail, Node004, Removing current message
2019-03-30 13:35:46.599 Detail, Node004, Notification: ValueChanged
2019-03-30 13:35:46.599 Detail, Node004, Notification: ValueChanged
2019-03-30 13:35:46.600 Detail, Node004, Notification: ValueChanged
```

Node4 is the node with the issue.

sjorge, 2019-03-30:
As per the original ticket, we also tried with getsupport=false for the switch value, that resulted in no value change messages at all.

Fishwaldo, 2019-04-01:
I assume your referring to the messages around 13:35:42.

Without knowing this device, there is no "BUG" in such. In OZW, we always to a GET after a SET to make sure we have the most recent version. The additional report we receive is probably due to the lifeline association sending the update to all nodes in the lifeline association group. 

In addition, we get a HAIL message (thats ancient!) that forces us to query all dynamic valueid's for updates. 

Check the config of the device if you can turn off HAIL messages (shouldn't be needed if Association work well on this device) and as said, nothing wrong with the second report...

Fishwaldo, 2019-04-01:
regarding the "old value" coming in then the updated value - that's what the device sends... we have no control over that...

closed 2019-04-01 by Fishwaldo

sjorge, 2019-04-01:
The hail cc is a configuration option 4-112-1-80 according to the configuration, but before wiping zwcfg that was set to basic.

Fishwaldo, 2019-04-01:
zwcfg is a cache file. Changing things there does not propagate the changes to to the device. You shouldn't need to ever edit it (if you *need* to its a bug that should be addressed via a device config file instead). 

So you need to use the configuration ValueID's to change the Hail to something else. Hail was mainly used as a workaround when there was a patent on sending instant updates from Switches. That patent expired a long time ago and I've not seen Hail in any recent devices which implies to me you might be using a older device/older firmware. (when zwave was a bit more wild west!). 

GetSupport as you found out, disables GET requests all together. Its intended use is for devices that dont support GET commands and thus have timeout issues (mainly for Basic CommandClasses). Its not for supressing messages. 

IMHO most devices do REPORT the correct value in the initial GET request... the biggest issue comes from MultiLevelSwitches that are ramping up/down and report their current level rather than the final level. That is handled by a updated version of the CommandClass (that has a target value) but a BinarySwitch should be a "instant" change (its either on/off). 

If the device reports its original state, and then almost immediately (via the lifeline association I assume) its final state - sorry - nothing OZW can do about it. It could be for a multitude of reasons. Perhaps if your network is congested, you will get a longer pause between the SET and GET commands and get the right report. 
The only change you can try is [Manager::SetChangeVerified](http://www.openzwave.com/dev/classOpenZWave_1_1Manager.html#ac26cc3c1b37cf068ac9eee23ef343c30) on the specific ValueID, but that will increase the network utilization, and potentially latency and most likely, you would have recieved the Assocation Report before the change is verified anyway.

sjorge, 2019-04-02:
It's on the newest firmware listed at the Aeotec website from 2018. I did notice I can reset the device without excluding/including it again (nice because it is inside a wall) I will do that and then capture another log, for completeness if other people have the same issue some day. Or with a bit of luck that might solve it (unlikely) I'll also while capturing that log change the default from Hail CC to basic and see and repeat so everything is in the log. Probably won't have time to do this for another week or so.

sjorge, 2019-04-06:
When calling SetChangeVerified(value_id, true) the problem is gone. So that is the proper fix.