OpenZWave/open-zwave #1762: Use of non-reentrant time functions (i.e  localtime()).

nihilus, 2019-04-11:
In TimeParameters.cpp, LogImpl.cpp and  TimeStampImpl.cpp there are calls to localtime() which should be replaced by calls to localtime_r() to ensure thread safety.

Fishwaldo, 2019-04-12:
Patches Welcome!

nihilus, 2019-04-12:
`Index: openzwave-1.4.164/cpp/src/platform/unix/TimeStampImpl.cpp
===================================================================
--- openzwave-1.4.164/cpp/src/platform/unix/TimeStampImpl.cpp	(revision 23000)
+++ openzwave-1.4.164/cpp/src/platform/unix/TimeStampImpl.cpp	(working copy)
@@ -112,8 +112,9 @@
 )
 {
 	char str[100];
-	struct tm *tm;
-	tm = localtime( &m_stamp.tv_sec );
+	struct tm *tm, xtm;
+	memset(&xtm, 0, sizeof(xtm));
+	tm = localtime_r( &m_stamp.tv_sec, &xtm );
 
 	snprintf( str, sizeof(str), "%04d-%02d-%02d %02d:%02d:%02d:%03d ", 
 		  tm->tm_year + 1900, tm->tm_mon + 1, tm->tm_mday,
Index: openzwave-1.4.164/cpp/src/platform/unix/LogImpl.cpp
===================================================================
--- openzwave-1.4.164/cpp/src/platform/unix/LogImpl.cpp	(revision 23000)
+++ openzwave-1.4.164/cpp/src/platform/unix/LogImpl.cpp	(working copy)
@@ -261,8 +261,9 @@
 	// Get a timestamp
 	struct timeval tv;
 	gettimeofday(&tv, NULL);
-	struct tm *tm;
-	tm = localtime( &tv.tv_sec );
+	struct tm *tm, xtm;
+	memset(&xtm, 0, sizeof(xtm));
+	tm = localtime_r( &tv.tv_sec, &xtm);
 
 	// create a time stamp string for the log message
 	char buf[100];
Index: openzwave-1.4.164/cpp/src/command_classes/TimeParameters.cpp
===================================================================
--- openzwave-1.4.164/cpp/src/command_classes/TimeParameters.cpp	(revision 23000)
+++ openzwave-1.4.164/cpp/src/command_classes/TimeParameters.cpp	(working copy)
@@ -176,13 +176,14 @@
 	if ( (ValueID::ValueType_Button == _value.GetID().GetType()) && (_value.GetID().GetIndex() == TimeParametersIndex_Set) )
 	{
 		time_t rawtime;
-		struct tm *timeinfo;
+		struct tm *timeinfo, xtm;
 		time(&rawtime);
 #ifdef WINAPI_FAMILY_APP
 #pragma warning(push)
 #pragma warning(disable:4996)
 #endif
-		timeinfo = localtime(&rawtime);
+		memset(&xtm, 0, sizeof(xtm));
+		timeinfo = localtime_r(&rawtime, &xtm);
 #ifdef WINAPI_FAMILY_APP
 #pragma warning(pop)
 #endif
`

nihilus, 2019-04-12:
This solution works for me.

petergebruers, 2019-04-27:
Hi @nihilus. I have turned your patch into a PR. I have no doubt, this is a sane advice, but I was wondering if you saw issues caused by this. Or did you review the code and and thought "let's fix that". I am interested because I am threading and locking used in OZW.

nihilus, 2019-04-27:
I did scan through our codebase at work for unsafe non-reentantrant functions and stumbled over it and since I knew openzwave uses threading I fixed it.

nihilus, 2019-04-27:
As I had issues in another threaded program using non-reentantrant variants of functions.

petergebruers, 2019-04-28:
@nihilus thank you for your explanation and thank you for reporting this!

petergebruers, 2019-04-29:
Fix merged in OpenZwave Version V1.5-3776-g0079066f

closed 2019-05-03 by Fishwaldo