OpenZWave/open-zwave #1764: Issue with ZMNKID (new thermostat device from Goap/Qubino)

petrovcicklemen, 2019-04-12:
Hi.

I'm writing from the company Goap, which sells various Zwave devices under the brandname Qubino.

We've developed a new thermostat device (ZMNKID), which we also want to use in OpenZwave. A configuration file was created and tested, to avoid integration problems in the future.

We discovered one issue with setpoint configuration (command THERMOSTATE_SETPOINT_SET), other functionalities appear to be working correctly.

The problem:
The device is not a multi channel device, like our older thermostats. The device supports two thermostate modes: heat and cool (its in idle mode by default).
When we select "Heat" mode in the GUI, the gateway sends a THERMOSTATE_SETPOINT_SET(Mode=Cooling) instead of THERMOSTATE_SETPOINT_SET(Mode=Heat). 
Gateway sends:            0x43, 0x01, 0x02, 0x01, 0x21
Gateway should send:  0x43, 0x01, 0x01, 0x01, 0x21

Since this functionality appeared to be working correctly in other gateways, this is probably a gateway related issue. 

Would anybody know, if this can be fixed in the configuration xml? I'm also attaching a picture of the GUI and can provide a zniffer trace, if neede.

The edited configuration xml file, manufacturer_specific.xml, a screenshot of the gui and a zniffer log, of the problem, can be downloaded from here:
https://www.dropbox.com/s/eatg6zhtkqzfnu0/ZMNKID.7z?dl=0

Also: We're using OpenZwave with the Control panel GUI. We've added <help> elements at every parameter, but cant seem to find, where these descriptions are shown on the GUI. Does anybody know, if Control panel even shows them?

Thank you.

Regards.

Fishwaldo, 2019-04-12:
Hi,
I think you might be talking about the note in SDS13781 - 4.109.2 Interoperability Considerations. Can you confirm?

If so, then adding the option `typeInterpretation="A"` to the <CommandClass id="67"> section in the config file should fix this. (it seems most vendors were going with the second interpretation, so type B is the default in master. - I'll flip it around in the Dev Branch). Can you test and confirm that fixes it?

As for the Config files, I would appreciate if you can open a PR for them so proper attribution of your changes is made :)

The control panel (ozwcp for short) doesn't always seem to display the help messages etc. It seems to be a HTML/Javascript etc but and unfortunately I'm weak in those languages... But a lot of applications to make use of the Help sections either as a tooltip/mouseover or as a help dialog.

petrovcicklemen, 2019-04-12:
Hi.

It appears both interpretations have the same encoding for the Heat (byte 1, bit 1) and Cool (byte 1, bit 2) mode, which are the modes, that our device supports.

The configuration file currently doesn't have the section  <CommandClass id="67"/>. I was thinking that maybe the "base" attribute, which is commonly found in <CommandClass id="67"/>, could be an issue? Could you, please, tell me what it does? I'll be able to test this on monday.

Thanks.

petrovcicklemen, 2019-04-16:
Hi.

I've added the `<CommandClass id="67" base="0" />` section into the configuration xml. I kindly ask you, to include the

Fishwaldo, 2019-04-17:
Ahhh. Apologies, the TypeInterpretation is implemented in the Dev Branch. Can you try that version if you didn't already?

As for base, some manufacturers started the "index" for each mode at 0, some started at 1. Base just adds to the index.

petrovcicklemen, 2019-04-17:
I've added the line `<CommandClass id="67" base="0" />` to the configuration file and it appears to be working now.

A [pull request](https://github.com/OpenZWave/open-zwave/pull/1766) was created for the configuration file and the changed `manufacturer_specific.xml`. Is this ok, or should we test the dev branch anyway?

Thank you.

Fishwaldo, 2019-04-17:
I'll merge the pull request soon. No need to submit one for the Dev branch, I'll eventually sync the two together. 

Glad to know its working! I'll close this issue then.

closed 2019-04-17 by Fishwaldo