OpenZWave/open-zwave #1765: Compile Error with ubuntu 18.10

Fishwaldo, 2019-04-13:
/usr/src/packages/BUILD/cpp/src/command_classes/DoorLockLogging.cpp: In member function 'virtual bool OpenZWave::DoorLockLogging::HandleMsg(const uint8*, uint32, uint32)':
[  268s] /usr/src/packages/BUILD/cpp/src/command_classes/DoorLockLogging.cpp:313:78: error: '%d' directive output may be truncated writing between 1 and 3 bytes into a region of size between 0 and 253 [-Werror=format-truncation=]
[  268s]                                          snprintf(usercode, sizeof(usercode), "%s %d", tmpusercode, (int)_data[12+i]);
[  268s]                                                                               ^~~~~~~
[  268s] /usr/src/packages/BUILD/cpp/src/command_classes/DoorLockLogging.cpp:313:78: note: directive argument in the range [0, 255]
[  268s] In file included from /usr/include/stdio.h:873,
[  268s]                  from /usr/include/c++/8/cstdio:42,
[  268s]                  from /usr/include/c++/8/ext/string_conversions.h:43,
[  268s]                  from /usr/include/c++/8/bits/basic_string.h:6391,
[  268s]                  from /usr/include/c++/8/string:52,
[  268s]                  from /usr/src/packages/BUILD/cpp/src/command_classes/CommandClasses.h:31,
[  268s]                  from /usr/src/packages/BUILD/cpp/src/command_classes/DoorLockLogging.cpp:28:
[  268s] /usr/include/x86_64-linux-gnu/bits/stdio2.h:67:35: note: '__builtin___snprintf_chk' output between 3 and 258 bytes into a destination of size 254
[  268s]    return __builtin___snprintf_chk (__s, __n, __USE_FORTIFY_LEVEL - 1,
[  268s]           ~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
[  268s]         __bos (__s), __fmt, __va_arg_pack ());
[  268s]         ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

closed 2019-04-20 by Fishwaldo

mikefranken, 2019-04-23:
Now it does not compile on MacOs or Raspberry Pi:
`In function ‘char* strncat(char*, const char*, size_t)’,
    inlined from ‘virtual bool OpenZWave::DoorLockLogging::HandleMsg(const uint8*, uint32, uint32)’ at /tmp/open-zwave-aebdb28d830d2c0b48cb8be38aba3d6ee621ee1c.6Ap/build/open-zwave-aebdb28d830d2c0b48cb8be38aba3d6ee621ee1c/cpp/src/command_classes/DoorLockLogging.cpp:313:89:
/usr/include/arm-linux-gnueabihf/bits/string3.h:150:71: error: call to char* __builtin___strncat_chk(char*, const char*, unsigned int, unsigned int) might overflow destination buffer [-Werror]
   return __builtin___strncat_chk (__dest, __src, __len, __bos (__dest));
`

caseywebdev, 2019-04-24:
I got the same error as @mikefranken on my RPi. Building from 75bcd082e478fc08bff872e62fc8122bae14eb86 worked though.

petergebruers, 2019-04-24:
Hi! You're probably using gcc 8 to compile and it has extra tests on string functions. We're fixing things to be compatible. #1767 is also needed (not same bug but related), it is waiting to get approval.

ThomasAnonymous, 2019-06-26:
I got the same error during make on release 1.6 on a Raspberry Pi 2 running alpine 3.10.0 and gcc 8.3.0. 
I changed the usercode str length to 255 per https://github.com/OpenZWave/open-zwave/commit/bd9de2c7b9d051427e55e4de3002be05406429e6#commitcomment-33353393
and it looks like an acceptable solution.