OpenZWave/open-zwave #1769: PREFIX still tries to install stuff in //usr/local/lib/x86_64-linux-gnu

m-p-goodwin, 2019-04-17:
I have used the following commands:
`git clean -xdf`
`make -j10`
then I would like to install it in a local folder:
`PREFIX=/home/martin/opt/ make`
but I get the following output:
```
LDFLAGS="" CPPFLAGS="" make -C /home/martin/src/open-zwave/cpp/build/ -
make[1]: Entering directory '/home/martin/src/open-zwave/cpp/build'
Building OpenZWave Version 1.4-3449-g75bcd082-dirty
make[1]: Leaving directory '/home/martin/src/open-zwave/cpp/build'
LDFLAGS="" CPPFLAGS="" make -C /home/martin/src/open-zwave/cpp/examples/MinOZW/ -
make[1]: Entering directory '/home/martin/src/open-zwave/cpp/examples/MinOZW'
make[1]: Nothing to be done for 'default'.
make[1]: Leaving directory '/home/martin/src/open-zwave/cpp/examples/MinOZW'

```

m-p-goodwin, 2019-04-17:
I can see that it tries to make the pkg-config file libopenzwave.pc in the //usr/local/lib/x86_64-linux-gnu/pkgconfig/ (now that I have let it do what it wanted). Everything else is installed as requested with the PREFIX variable.

ffbenedetti, 2019-07-10:
This is the same root cause that it is also creating the issue reported here: #1416
In https://github.com/OpenZWave/open-zwave/blob/master/cpp/build/support.mk#L127
it is set a target directory that is not considering at all PREFIX variable.

ffbenedetti, 2019-07-10:
I think the fix is to change line: https://github.com/OpenZWave/open-zwave/blob/master/cpp/build/support.mk#L127

from 

`pkgconfigdir ?= $(shell test -d "$(instlibdir)/pkgconfig" && echo "$(instlibdir)/pkgconfig" || pkg-config --variable pc_path pkg-config | awk -F: '{ print $$1 }')`

to 

`pkgconfigdir ?= $(instlibdir)/pkgconfig`

Fishwaldo, 2019-07-10:
If the OpenZWave.pc file does not go into the directory where pkg-config is configured for, then it’s useless. So just installing it into a random directory is not going to work. Hence why one of the tests is the pc_path. 

So that fix is not correct from a pkg-config perspective.

ffbenedetti, 2019-07-10:
that is consistent with the way how the `configure` set the target for .pc file relative to the prefix directory

Fishwaldo, 2019-07-10:
> that is consistent with the way how the `autoconf` set the target for .pc file relative to the prefix directory

Then that is broken. 
pkg_config will only look in specific directories on the system for pc files. If you install openzwave.pc into a "random" directory, then you might as well not install it.

oandrew, 2019-07-10:
> Then that is broken.
> pkg_config will only look in specific directories on the system for pc files. If you install openzwave.pc into a "random" directory, then you might as well not install it.

You could say the same about shared libraries - why install them at all  if they won't be picked up by default (e.g. installing to custom prefix)? 

The idea of PREFIX is that everything goes under PREFIX. 
Also relying on the output of `pkg-config --variable pc_path pkg-config` is dangerous as it can contain more than 1 path. 

 
https://www.gnu.org/prep/standards/html_node/Directory-Variables.html

> GNU packages should not try to guess which value should be appropriate for these variables on the system they are being installed onto: use the default settings specified here so that all GNU packages behave identically, allowing the installer to achieve any desired layout.

> All installation directories, and their parent directories, should be created (if necessary) before they are installed into.

Fishwaldo, 2019-07-10:
> You could say the same about shared libraries - why install them at all if they won't be picked up by default (e.g. installing to custom prefix)?

which, ironically, is one of the things that pkg-config is meant to overcome. From the homepage:
>[than hard-coding values on where to find glib (or other libraries)](https://www.freedesktop.org/wiki/Software/pkg-config/)

>Also relying on the output of pkg-config --variable pc_path pkg-config is dangerous as it can contain more than 1 path.

Keep reading on that support.mk line. We take the first entry: 
` awk -F: '{ print $$1 }'`

and hey, surprise on my mac with homebrew installed:
[image]

which just happens to be the first directory listed in pkg-config

In the Ops case, he should just override the pkgconfigdir variable to install it locally or to a throw away location, as he seems to be installing OZW as a normal user (and will have forementioned ldconfig settings to deal with if he is using OZW as a shared library.)

Your proposed changes will break pkg-config for OZW on most of the distributions it currently works for, for the sake of
1) Homebrew which uses non-standard path (according to the LSB) all over the place
2) a usecase, that 95% of the users never use (in the years since pkg-config has been in OZW, this is the first report we got about installing outside of prefix and running into permission issues). 

As homebrew does things "differently" - Your only option is to override the pkgconfigdir variable with whatever Cellar location the openzwave library is being installed (and then ensure that its symlinked back to the /usr/local/lib/pkgconfig directory by whatever mechanism homebrew uses)

ffbenedetti, 2019-07-10:
Then also autoconf is broken. All packages that use autoconf calling configure and passing a different prefix then the default is putting the .pc files under the directory <prefix>/lib/pkgconfig.
autoconf is the standard for installing linux packages. It is fully integrated with pkg-config and vice versa.
Homebrew is designed to install packages that use autoconf conventions. 
It uses the <prefix> parameter used by configure (generated by autoconf) to force the installation in a separate path then the MacOS then the global /usr/local.
This is confirmed by the fact that for all packages that use autoconf and that use pkg-config the .pc files are installed under <prefix>/lib/pkgconfig and the Homebrew formula  is just doing:

def install
    args = %W[
      --prefix=#{prefix}
    ]
     system "./configure", *args
     system "make"
     system "make", "install"

oandrew, 2019-07-10:
> Keep reading on that support.mk line. We take the first entry:
> ` awk -F: '{ print $$1 }'`
>
> and hey, surprise on my mac with homebrew installed:
> [image]
>
> which just happens to be the first directory [...]
Exactly, it just happens to be the first directory. 
On my other machine (ubuntu xenial):
```
$ pkg-config --variable pc_path pkg-config
/usr/local/lib/x86_64-linux-gnu/pkgconfig:/usr/local/lib/pkgconfig:/usr/local/share/pkgconfig:/usr/lib/x86_64-linux-gnu/pkgconfig:/usr/lib/pkgconfig:/usr/share/pkgconfig

$ find /usr/local/lib/pkgconfig -name '*.pc' | wc -l
41

$ find /usr/local/lib/x86_64-linux-gnu/pkgconfig -name '*.pc' | wc -l
0
```

`/usr/local/lib/x86_64-linux-gnu/pkgconfig` is the first entry in pkg-config search list and yet none of the software that I manually compiled installed it's .pc files in that directory. They all went into `/usr/local/lib/pkgconfig`.

Sure, in my case  open-zwave would get installed in the right place  because `/usr/local/lib/pkgconfig` directory already exists.
But what if it didn't? 

here is what grpc does:
https://github.com/grpc/grpc/blob/master/Makefile#L3234

ffbenedetti, 2019-07-10:
I have found where all come from: the GNU Coding Standards

https://www.gnu.org/prep/standards/html_node/Directory-Variables.html

It mandates that the `make install` must support the `prefix` variable that forces all the other variables, like `libdir` `bindir` etc., to be based on the path specified `prefix`.
Autoconf, pkg-conf etc. are all utility designed based on the GNU Coding Standards. Homebrew is doing the same.

The root problem is that the `make install` is not following the GNU Coding Standards: uses a variable `PREFIX` instead of `prefix` and for the `.pc` files it is installing them not applying the path `$(libdir)/pkgconfig` as required by  a GNU and implemented by autoconf

Fishwaldo, 2019-07-11:
As *ALREADY* mentioned - you fix is to override the pkgconfigdir variable.

and as ALREADY mentioned:
>Your proposed changes will break pkg-config for OZW on most of the distributions it currently works for

So please go file a bug with all the distributions and versions that don't use the "standard" location for pc files (due to multiarch, different packaging requirements, old versions of pkg-config etc), and let me know when they fix their distribution.

Quoting a bunch of standards when distributions don't always follow them is pointless. I'm speaking from practical experience here when I'm telling you this, as I've packaged up OZW (and others apps over the years) and found this. 

This conversation is toxic for a minor issue that has a fix provided for you to override the defaults if you don't like it. Use it. Closing and Locking the conversation.

btw, just for clarity sake:
```
root@nanopim4:/usr/lib/aarch64-linux-gnu/pkgconfig# cd /usr/lib/aarch64-linux-gnu/pkgconfig/
root@nanopim4:/usr/lib/aarch64-linux-gnu/pkgconfig# ls *pc | wc -l
125
```

```
root@deadpan:/root# cd /usr/share/pkgconfig/
root@deadpan:/usr/share/pkgconfig# ls *pc | wc -l
86
```

```
root@tallyo:/root# cd /usr/lib64/pkgconfig/
root@tallyo:/usr/lib64/pkgconfig# ls *pc | wc -l
161
root@tallyo:/usr/lib64/pkgconfig# cd /usr/lib32/pkgconfig/
root@tallyo:/usr/lib32/pkgconfig# ls *pc | wc -l
181
root@tallyo:/usr/lib32/pkgconfig# pkg-config --variable pc_path pkg-config
/usr/lib/pkgconfig:/usr/lib64/pkgconfig
```

Notice in that last line, there is *NO* /usr/local etc. So.... again, why should I break distributions where pkg-config for OZW is working for the sake of following a standard that it seems a large amount of distributions ignore, or do differently?

@m-p-goodwin you should invoke the make command with:
`PKGCONFIGDIR=/home/martin/opt/lib/pkgconfig PREFIX=/home/martin/opt/ make install` to solve the issue.

closed 2019-07-11 by Fishwaldo

Fishwaldo, 2019-07-11:
by the way, autoconf makes no effort to even check if the directory where pc files are installed is even a valid pkg-config directory:
```
AC_DEFUN([PKG_INSTALLDIR],
[m4_pushdef([pkg_default], [m4_default([$1], ['${libdir}/pkgconfig'])])
m4_pushdef([pkg_description],
    [pkg-config installation directory @<:@]pkg_default[@:>@])
AC_ARG_WITH([pkgconfigdir],
    [AS_HELP_STRING([--with-pkgconfigdir], pkg_description)],,
    [with_pkgconfigdir=]pkg_default)
AC_SUBST([pkgconfigdir], [$with_pkgconfigdir])
m4_popdef([pkg_default])
m4_popdef([pkg_description])
])dnl PKG_INSTALLDIR
```