OpenZWave/open-zwave #1783: Compiling from source is failing since today

AlCalzone, 2019-05-03:
I guess this has to do with the recent merge, but I don't know where to start. For a couple of hours now, our builds are failing:

```
---> cd open-zwave-master && make && sudo make install
/home/travis/build/ioBroker/ioBroker.zwave/open-zwave-master/cpp/src/CompatOptionManager.cpp: In member function ‘void OpenZWave::CompatOptionManager::WriteXML(TiXmlElement*)’:
/home/travis/build/ioBroker/ioBroker.zwave/open-zwave-master/cpp/src/CompatOptionManager.cpp:259:33: error: ‘text’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
   valElement->LinkEndChild(text);
                                 ^
cc1plus: all warnings being treated as errors
make[1]: *** [/home/travis/build/ioBroker/ioBroker.zwave/open-zwave-master/.lib/CompatOptionManager.o] Error 1
make: *** [all] Error 2
---> Error: Command failed: cd open-zwave-master && make && sudo make install
/home/travis/build/ioBroker/ioBroker.zwave/open-zwave-master/cpp/src/CompatOptionManager.cpp: In member function ‘void OpenZWave::CompatOptionManager::WriteXML(TiXmlElement*)’:
/home/travis/build/ioBroker/ioBroker.zwave/open-zwave-master/cpp/src/CompatOptionManager.cpp:259:33: error: ‘text’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
   valElement->LinkEndChild(text);
                                 ^
cc1plus: all warnings being treated as errors
make[1]: *** [/home/travis/build/ioBroker/ioBroker.zwave/open-zwave-master/.lib/CompatOptionManager.o] Error 1
make: *** [all] Error 2
/home/travis/build/ioBroker/ioBroker.zwave/lib/preinstall.js:16
    if (!suppressException) throw e;
                            ^
Error: Command failed: cd open-zwave-master && make && sudo make install
/home/travis/build/ioBroker/ioBroker.zwave/open-zwave-master/cpp/src/CompatOptionManager.cpp: In member function ‘void OpenZWave::CompatOptionManager::WriteXML(TiXmlElement*)’:
/home/travis/build/ioBroker/ioBroker.zwave/open-zwave-master/cpp/src/CompatOptionManager.cpp:259:33: error: ‘text’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
   valElement->LinkEndChild(text);
                                 ^
cc1plus: all warnings being treated as errors
make[1]: *** [/home/travis/build/ioBroker/ioBroker.zwave/open-zwave-master/.lib/CompatOptionManager.o] Error 1
make: *** [all] Error 2
```

Fishwaldo, 2019-05-03:
https://groups.google.com/d/msg/openzwave/a26mAKMam58/KZnH6wQ0AgAJ

what OS?

Its been building fine for me: http://bamboo.my-ho.st/bamboo/browse/OZW-OZWLNX/latest

Fishwaldo, 2019-05-03:
Should be fixed now.

closed 2019-05-03 by Fishwaldo

Fishwaldo, 2019-05-08:
@pascal-mormin please confirm exactly which version of 1.6 is failing.

ThomasAnonymous, 2019-06-26:
I also just got a make error here. 

```
/home/key-per/open-zwave-1.6/cpp/src/CompatOptionManager.cpp: In member function 'void OpenZWave::CompatOptionManager::WriteXML(TiXmlElement*)':
/home/key-per/open-zwave-1.6/cpp/src/CompatOptionManager.cpp:259:27: error: 'text' may be used uninitialized in this function [-Werror=maybe-uninitialized]
   valElement->LinkEndChild(text);
   ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
```

Using 1.6 from https://github.com/OpenZWave/open-zwave/commit/890f24b7e88f488eee464ed14c01fbceb276cf2a
on a Raspberry Pi 2, running alpine-3.10.0, gcc-8.3.0

ThomasAnonymous, 2019-06-26:
> I also just got a make error here.
> 
> ```
> /home/key-per/open-zwave-1.6/cpp/src/CompatOptionManager.cpp: In member function 'void OpenZWave::CompatOptionManager::WriteXML(TiXmlElement*)':
> /home/key-per/open-zwave-1.6/cpp/src/CompatOptionManager.cpp:259:27: error: 'text' may be used uninitialized in this function [-Werror=maybe-uninitialized]
>    valElement->LinkEndChild(text);
>    ~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~
> ```
> 
> Using 1.6 from [890f24b](https://github.com/OpenZWave/open-zwave/commit/890f24b7e88f488eee464ed14c01fbceb276cf2a)
> on a Raspberry Pi 2, running alpine-3.10.0, gcc-8.3.0

It looks like adding a default condition on the switch allows the build to proceed, I just duped the int condition for now - maybe these values can be cast to int to avoid the switch altogether?

Fishwaldo, 2019-06-26:
Are you using the master branch? These should have all been fixed.

petergebruers, 2019-06-26:
@ThomasAnonymous as @Fishwaldo said those gcc issues were fixed after that build... But to be 100% certain: I am on a Mac (using clang by default) but I can select gcc, then build master with:

gcc version 8.3.0 (Homebrew GCC 8.3.0) 

I am able to build master and run MinOZW, this is OpenZwave Version 1.6-788-g52db0e91

I am nut sure if that would be different on a Pi 2

Please start a new issue if you can't build master.

ThomasAnonymous, 2019-06-26:
I was using 1.6 stable linked above. Master came back with a clean build ( some piecewise construct notes, nbd) and overall looks much more organized. You guys have been very productive in the past month! Thanks for your quick responses and all your hard work.