OpenZWave/open-zwave #1785: Manager.cpp:1768 - 101 - Invalid ValueID passed to GetValueLabel

AlCalzone, 2019-05-08:
This is a followup to https://github.com/OpenZWave/open-zwave/issues/1783

Now that we got 1.6 to work on Linux and OSX (thanks for the quick solution), we're getting new reports of our zwave adapter not working. This time OZW crashes with an exception during execution:
`Manager.cpp:1768 - 101 - Invalid ValueID passed to GetValueLabel`

I've added the log file so you can examine:
[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/3155858/OZW_Log.txt)
It seems to have to do with an EUROtronic thermostat - I remember there was some work required to make it functional. Maybe it has to do with that?

Fishwaldo, 2019-05-08:
This is a application error. Its attempted to pass a invalid ValueID (one that doesn't exist) to OZW. 

I doubt very much that your going to have much success with migrating to 1.6 without having a updated wrapper and updating your application code. While I've done my best to maintain API compability with existing methods, new methods/classes etc have been introduced (Such as BitSet ValueID) that will become more and more common as we update the config files. 

For instance - you need to wrap all calls to the manager class in a try..catch block to catch potential errors (such as above).

closed 2019-05-08 by Fishwaldo

AlCalzone, 2019-05-08:
@Fishwaldo From what I've heard this did work prior to the update. I'm currently assuming that the users who are affected manually merged the [corresponding PR](https://github.com/OpenZWave/open-zwave/pull/1387) into their 1.4/1.5 installation. 

As we are using the `node-openzwave-shared` wrapper, we cannot wrap the calls ourselves. Do you have an idea how to avoid the error altogether? Could this come from incompatible data cached somewhere? Or do we need to change the installation routine so it compiles an old version of OZW?

Fishwaldo, 2019-05-08:
1.4 would have logged a error and returned something undefined for some calls. Hence one of the reasons why we introduced exceptions. (as there was no clean way to indicate a error previously in some calls). 
1.5 has had exceptions enabled *forever*

1.6 has a new config file format. Dropping a 1.4 (or 1.5 config file from a few weeks ago) into 1.6 wont work. You need to update it. 

As I mentioned, until node-openzwave-shared is updated properly with 1.6, I wouldn't run it. You can use the 1.4 branch in OZW repository, but that will not receive any further updates.