OpenZWave/open-zwave #1788: Missing entry for an MH9-CO2-WD CO2 Monitor

gdave321, 2019-05-10:
My CO2 sensor is missing in manufacturer_specific.xml, however, it works just fine with an existing config file. This works for me, fully tested and working:

    --- manufacturer_specific.xml.ORIG  2019-05-10 13:24:09.048257876 +0200
    +++ manufacturer_specific.xml       2019-05-10 13:25:08.964118480 +0200
    @@ -1113,6 +1113,7 @@
         <Product config="mcohome/mh8fceu.xml" id="3102" name="MH8-FC-EU Thermostat" type="0801"/>
         <Product config="mcohome/mh8fceu.xml" id="3102" name="MH8-FC4-EU Thermostat" type="0802"/>
         <Product config="mcohome/mh9co2.xml" id="0201" name="MH9-CO2-WD CO2 Monitor" type="0905"/>
    +    <Product config="mcohome/mh9co2.xml" id="3102" name="MH9-CO2-WD CO2 Monitor" type="0901"/>
         <Product config="mcohome/a8-9.xml" id="1352" name="A8-9 Multi-sensor" type="a800"/>
         <Product config="mcohome/mh7h.xml" id="5102" name="MH7H Water/Electrical Heating Thermostat" type="0701"/>
         <Product config="mcohome/mh7h.xml" id="5102" name="MH7H Water/Electrical Heating Thermostat" type="0702"/>

closed 2019-05-11 by nechry

chris-cjk, 2020-04-22:
I am trying to get this working with Hubitat... no luck so far.  Do you have whole device handler in its entirety?  Could you kindly send me what you have?  I would be really pleased if you could,

chris-cjk, 2020-04-22:
I just have the temperature and humidity NOT the C02 PPM.

commit 0ab45e29 (2019-05-11): Update manufacturer_specific.xml

fix #1788