OpenZWave/open-zwave #1801: Setting config values for Aeon ZW090-A always fails

michael-fadely, 2019-05-16:
Attempting to change a configuration variable with the Aeon ZW090-A always fails. In the attached log, you can see that the option I tried to configure was `LED indicator configuration`.

[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/3189252/OZW_Log.txt)

I confirmed that the device is fully functional by testing this with Aeon's backup software. The LED indicator toggle in their software works every time.

Fishwaldo, 2019-05-17:
Unfortunately we can't change configuration values on the controller. Its a limitation of the SerialAPI protocol. The Aeotec utility has non-zwave packet it sends to the serial port to do this.

closed 2019-05-17 by Fishwaldo

michael-fadely, 2019-05-17:
That would certainly explain it. Thank you!