OpenZWave/open-zwave #1804: Remove CommandClasses from MultiInstance Endpoints

Fishwaldo, 2019-05-17:
Some devices (eg, WallMote) incorrectly advertise CommandClasses on endpoints. Need a way to remove those commandclasses.

GravityRZ, 2019-05-22:
is this related

v1.6 format in a lot of modules, causing problems, not working modules and errors

`<CommandClass id="96"> <Compatibility> <MapRootToEndpoint>true</MapRootToEndpoint> </Compatibility> </CommandClass>`

v1.4 format in a lot of modules
WORKING when you apply this format on the v1.6 file
`<CommandClass id="96" mapping="endpoints" /> </CommandClass>`

petergebruers, 2019-05-22:
@GravityRZ you are talking about an issue have with Domoticz, right?

https://github.com/domoticz/domoticz/issues/3253

This is 99% certain not related with the issue @Fishwaldo opened, because this issue here is something quite specific to this WallMote.

The "MapRootToEndpoint" is a flag that adds an additional data, and in 1.4 that was called "mapping="endpoints". It is difficult for me to tell if your device need that flag or not, but as the files were converted from 1.4 to the new 1.6 format, everything was based on the old files, so theoretically they are equivalent.

GravityRZ, 2019-05-22:
@petergebruers 
yes i was talking about an issue in Domoticz.

i think something went wrong when translating the 1.4 files to 1.6
MapRootToEndpoint and mapping="endpoints might be equivalents but the qubino modlu ehas problems with the new 1.6 command and not the 1.4 command.

Fishwaldo, 2019-06-25:
This is done.

closed 2019-06-25 by Fishwaldo