OpenZWave/open-zwave #1807: Compiler errors.

kdschlosser, 2019-05-17:
OK folks. I have one for ya.

I am trying to compile the current code in the master branch and I am encountering compiler errors.

This is the setup. 
openzwave is being compiled from inside python. I updated the windows build portion of the library. the code has been added to python_openzwave it will be in the next release. The build system works fine with the previous release of openzwave.

BUILD ENVIRONMENT
```
Machine architecture: x64
Build architecture: x86

== Windows SDK ================================================
   version:     10.0.17763.0
   sdk version: 10.0.17763.0\
   path:        C:\Program Files (x86)\Windows Kits\10

   -- Universal CRT -------------------------------------------
      path: C:\Program Files (x86)\Windows Kits\10\
   -- ATLMFC --------------------------------------------------
      path:         C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\VC\Tools\MSVC\14.16.27023\ATLMFC
      include path: C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\VC\Tools\MSVC\14.16.27023\ATLMFC\include
      lib path:     C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\VC\Tools\MSVC\14.16.27023\ATLMFC\lib\x86

== Extension SDK ==============================================
   path: C:\Program Files (x86)\Microsoft SDKs\Windows Kits\10\ExtensionSDKs

== TypeScript =================================================
   path: None

== HTML Help ==================================================
   path: C:\Program Files (x86)\HTML Help Workshop

== .NET =======================================================
   version:    v4.0.30319

   -- x86 -----------------------------------------------------
      version: v4.0.30319
      path:    C:\Windows\Microsoft.NET\Framework
   -- x64 -----------------------------------------------------
      version: v4.0.30319
      path:    C:\Windows\Microsoft.NET\Framework64
   -- NETFX ---------------------------------------------------
      path:         C:\Program Files (x86)\Microsoft SDKs\Windows\v10.0A\bin\NETFX 4.6.1 Tools\
      x86 exe path: C:\Program Files (x86)\Microsoft SDKs\Windows\v10.0A\bin\NETFX 4.6.1 Tools\
      x64 exe path: C:\Program Files (x86)\Microsoft SDKs\Windows\v10.0A\bin\NETFX 4.6.1 Tools\x64

== Visual C ===================================================
   version: 14.0
   path:    C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\VC

   -- Tools ---------------------------------------------------
      version:     14.16.27023
      path:        C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\VC\Tools\MSVC\14.16.27023
      redist path: C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\VC\Redist\MSVC\14.16.27012
   -- F# ------------------------------------------------------
      path: None
   -- DLL -----------------------------------------------------
      version: v140-14.16.27012.6
      path:    C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\VC\Redist\MSVC\14.16.27012\x86\Microsoft.VC141.CRT

== MSBuild ====================================================
   version: 14.0
   path:    C:\Program Files (x86)\MSBuild\14.0\bin

== Python =====================================================
  version: 2.7.15.final.0
  architecture: x86
  library: Python27.lib
  libs: ['c:\\stackless27\\libs']
[... 241 lines cut ...]
tr_fun<int,int>(_Result (__fastcall *)(_Arg))'
with
[
_Result=int,
_Arg=int
]
C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\VC\Tools\MSVC\14.16.27023\include\functional(576): note: or       'std::pointer_to_unary_function<int,int,int (__cdecl *)(int)> std::ptr
fun<int,int>(_Result (__cdecl *)(_Arg))'
with
[
_Result=int,
_Arg=int
]
C:\Users\Administrator\Documents\GitHub\python-openzwave\openzwave\cpp\src\Utils.cpp(99): note: while trying to match the argument list '(overloaded-function)'
C:\Users\Administrator\Documents\GitHub\python-openzwave\openzwave\cpp\src\Utils.cpp(98): error C2672: 'std::find_if': no matching overloaded function found
compiling ValueStore.cpp...
```

The compiler hangs when compiling Utils.cpp. This error does not show until I perform a keyboard interrupt. then the compiler starts to move again spits out this error and of course the program exists because i did a keyboard interrupt. 

The new build system for python_openzwave does not use the solution foles included with openzwave. instead I create an identical build environment that would be created if using visual studio or vcvarsall. I do this without using vcvarsall (it has issues). I then call cl.exe directly passing the compiler arguments outlined above. 

If there is someone that is running a Windows box and has Visual Studio, Visual C, or the Visual Studio Build Tools and is willing to test to make sure that the problem is not isolated to only my machine. That would be a HUGE help.

To do that you need to download the 0.5.x branch and run `python setup.py build --flavor=dev` the program should take care of everything.

kdschlosser, 2019-05-17:
never mind.. after trying to figure this out for the past 3 hours. only when i finally post an issue do i find the issue and fix it.. 

sorry folks!

closed 2019-05-17 by kdschlosser

Fishwaldo, 2019-05-17:
You can always check http://bamboo.my-ho.st/bamboo/browse/OZW-OZWLNX when you have compile issues. 

As you can see we have a compile failure on older versions of VS right now. I need to investigate and fix that.

kdschlosser, 2019-05-17:
I am having a booger of a time with one thing. and it doesn't make any sense. I am getting the error below even tho I do have an overloaded method declared. It beats the hell out of me why it's doing that.

```
Error compiling Cython file:
------------------------------------------------------------
...
                cret = self.manager.SetValueListSelection(values_map.at(id), type_string)
                logger.debug("SetValueListSelection %s", cret)
                ret = 1 if cret else 0
            elif datatype == "BitSet":
                type_bool = value
                cret = self.manager.SetValue(values_map.at(id), pos, type_bool)
                                           ^
------------------------------------------------------------
```

Fishwaldo, 2019-05-18:
What is type_bool from the compiler perspective? It’s probably not sure if it’s a integer or ???

You could wrap it in a static_cast<bool>(value) when passing it to the SetValue call so the compiler is sure it’s a bool value. 

Or do something like if Val = python_true ? true : false in the setvalue call if it can’t be cast.

kdschlosser, 2019-05-18:
ok so this code is Cython code. Cython code = C++ code and python code smashed into a single language.

It is similar to C++ code in terms of the verbage and having to declare variable types. the syntax and statements are python. there needs to be a header file of sorts in Cython where you declare classes methods and class variables. 

there are some datatypes that would need to be cast into a python version of it. and some that really do not.

`bool` is not one of the types that needs to be cast. and `int` in python pretty much covers all of your `uint*` and `int*` data types. as well as `ushort*` and `short*` there is more then that. but to keep it simple we will leave it there. There is only checking does for min/max when converting from python to C++ and nothing is done when passing from C++ to python. I believe that is how it works. I have not dove into it. But from my work with PyWinAPI and porting the Windows SDK to python this is what I have come up with. I doubt it is going to be much different because of Cython.  

you have a "header" file of corts as well as the source file. very similiar to what is done in C++

file extensions
header = .pyd
source = .pyx

This code is from the "header" (.pxd) file
Cython does allow for overloading. You can see that done for SetValue. and you can also see the declaring of the overload SetValue(value id, pos, value) for the BitSet value type.

```cython
cdef extern from "Manager.h" namespace "OpenZWave":

    cdef cppclass Manager:
        # // Destructor
        void Destroy()
        # // Configuration
        # deprecated void WriteConfig(uint32_t homeid)
        Options* GetOptions()
        # // Drivers
        bool AddDriver(string serialport)
        bool RemoveDriver(string controllerPath)
        uint8_t GetControllerNodeId(uint32_t homeid)
        uint8_t GetSUCNodeId(uint32_t homeid)
        bool IsPrimaryController(uint32_t homeid)
        bool IsStaticUpdateController(uint32_t homeid)
        bool IsBridgeController(uint32_t homeid)
        string getVersionAsString()
        string getVersionLongAsString()
        string GetLibraryVersion(uint32_t homeid)
        string GetLibraryTypeName(uint32_t homeid)
        int32_t GetSendQueueCount( uint32_t homeId )
        void LogDriverStatistics( uint32_t homeId )
        void GetDriverStatistics( uint32_t homeId, DriverData* data )
        void GetNodeStatistics( uint32_t homeId, uint8_t nodeid, NodeData* data )
        ControllerInterface GetControllerInterfaceType( uint32_t homeId )
        string GetControllerPath( uint32_t homeId )
        # // Network
        void TestNetworkNode( uint32_t homeId, uint8_t nodeId, uint32_t count )
        void TestNetwork( uint32_t homeId, uint32_t count )
        void HealNetworkNode( uint32_t homeId, uint32_t nodeId, bool _doRR )
        void HealNetwork( uint32_t homeId, bool doRR)
        # // Polling
        uint32_t GetPollInterval()
        void SetPollInterval(uint32_t milliseconds, bIntervalBetweenPolls)
        bool EnablePoll(ValueID& valueId, uint8_t intensity)
        bool DisablePoll(ValueID& valueId)
        bool isPolled(ValueID& valueId)
        void SetPollIntensity( ValueID& valueId, uint8_t intensity)
        uint8_t GetPollIntensity(ValueID& valueId)
        # // Node Information
        bool RefreshNodeInfo(uint32_t homeid, uint8_t nodeid)
        bool RequestNodeState(uint32_t homeid, uint8_t nodeid)
        bool RequestNodeDynamic( uint32_t homeId, uint8_t nodeId )
        bool IsNodeListeningDevice(uint32_t homeid, uint8_t nodeid)
        bool IsNodeFrequentListeningDevice( uint32_t homeId, uint8_t nodeId )
        bool IsNodeBeamingDevice( uint32_t homeId, uint8_t nodeId )
        bool IsNodeRoutingDevice(uint32_t homeid, uint8_t nodeid)
        bool IsNodeSecurityDevice( uint32_t homeId, uint8_t nodeId )
        bool IsNodeZWavePlus( uint32_t homeId, uint8_t nodeId )
        uint32_t GetNodeMaxBaudRate(uint32_t homeid, uint8_t nodeid)
        uint8_t GetNodeVersion(uint32_t homeid, uint8_t nodeid)
        uint8_t GetNodeSecurity(uint32_t homeid, uint8_t nodeid)
[... 192 lines cut ...]
                else:
                    type_string = str_to_cppstr(string(value))
                cret = self.manager.SetValue(values_map.at(id), type_string)
                ret = 1 if cret else 0
            elif datatype == "Button":
                type_bool = value
                cret = self.manager.SetValue(values_map.at(id), type_bool)
                ret = 1 if cret else 0
            elif datatype == "List":
                logger.debug("SetValueListSelection %s", value)
                if six.PY3:
                    type_string = str_to_cppstr(value)
                else:
                    type_string = str_to_cppstr(string(value))
                cret = self.manager.SetValueListSelection(values_map.at(id), type_string)
                logger.debug("SetValueListSelection %s", cret)
                ret = 1 if cret else 0
            elif datatype == "BitSet":
                type_bool = value
                cret = self.manager.SetValue(values_map.at(id), pos, type_bool)
                ret = 1 if cret else 0

        return ret

```

kdschlosser, 2019-05-18:
never mind. I now see what the problem is.. HOLY COW!!!

I am such a dum dum i need to cast the pos into a uint8_t and i am not doing that.  OYE VAE!

well this is what helped me to figure it out. sometimes it is just explaining it that will make the problem reveal it's self.