OpenZWave/open-zwave #1814: Find a way to inform applications about important ValueID's

Fishwaldo, 2019-05-23:
Valid Feedback from a HA user - When looking at the entities for his Notification Based device, there was a lot of "advanced" or ancillary ValueID's exposed. (like SourceNode, or the Old Alarm V1 interface). 

We should look at a way to tag ValueID's as "Primary" for a CommandClass, and the others as secondary, or Advanced. Then GUI's can easily filter out advanced ValueID's for users (unless they want to see them I guess).

petergebruers, 2019-05-23:
I like the idea. A certain brand of gateways uses xml config files (who would have guessed) They are on the system not accessible to users. But a little bird tells me they use a property "visible". This gateway creates a "device" for each interesting property (not to be confused with a node), but hides the non-interesting ones. Their "device" is more or less equivalent to a ValueID I would say.

The config file of an older, non-Z-Wave plus double relay looks like this:

```
<endP id="0" enabled="true" visible="true">
...
<endP id="1" enabled="false" visible="false">
...
<endP id="2" enabled="true" visible="true">
```

This way, the BUI shows only 2 relays, the first one controls the root and the 2nd one endpoint 2. Endpoint one is available in a special section "hidden". I don't know why they selected root instead of endpoint one.

This is the config of a Z-Wave Plus version of their double relay.

```
<endP id="0" enabled="true" visible="true" hideVirt0="true">
...
<endP id="1" enabled="true" visible="true">
...
<end id="2" enabled="true" visible="true">
```

The BUI shows 2 relays, like the previous example, but uses multichannel. I am not sure what the hideVirt0 is about but I guess it has to do with CCs on root device.

Fishwaldo, 2019-05-24:
I would want to automate this rather than leave it upto a config file. The idea of updating 900+ config files does not sound appealing right now haha.

I'm debating the following approach:
1) each CommandClass tags a ValueID(s) as "primary"
2) Node Class "filters" the primary ValueID's based on device types etc
3) Manager Class exposes a Manager::IsValuePrimary(ValueID vid)

Item 2 might screw around with devices that do everything, but from a "user" perspective, if I buy a doorbell, I might only be interested in the doorbell, and rest is "meh". 

The other option is at Step 2 - Those ValueID's we filter out, instead retag them as "Secondary" and everything else as "ancillary" 

I also wonder 1) how many applications filter the Genre's? 2) if we are using the right Genre's on some ValueID's in commandclasses. 2 would be a audit activity I guess.

Fishwaldo, 2019-06-13:
So here is my train of thought so far... 
1) For Every Specific and DeviceType Entry in devices_classes.xml we add a "Primary" CommandClass. (eg a light switch gets SwitchBinary as the Primary CC)
2) For Every ValueID in each commandclass we add either a "Primary", "Secondary", or "Ancillary" flag to those in the User Genre
3) Add 2 Methods to the Manager Class - 
* GetPrimaryCC(Nodeid) - Returns the Primary CC in Device_Classes.xml
* GetValueIDRank(Nodeid, valueid) returns the flag Primary/Secondary/Ancillary for the ValueID

Applications can then filter ValueID's by DeviceType/Specific, and then further filter down by the Priority of the ValueID. 

So a Dimmer Switch might end up with:
GetPrimaryCC - Returns MultiLevelSwitch
ValueID's would be set such as Level would be primary, Bright/Dim/Increment/Decrement/Duration as Secondary and Step size, Ignore Start Level, Start Level and Target Value as Ancillary

Other Command Classes would also still have the Rank assigned to each ValueID, but the GUI can choose to not display them, or put them in a "advanced" tab, potentially "sorted" by ranking etc. 

Thoughts?