OpenZWave/open-zwave #1820: Qubino module ZMNHADx does not work in zwave 1.6 build

GravityRZ, 2019-05-30:
using domoticz  V4.10848

with zwave 1.6 a new mapping command is used.
The ZMNHADx modlue can not cope with this.
as a result S1 does not switch on and gives an error about an unknown ID.
 temperature reading from external sensor does work

i noticed when i comment out the new mapping command the module DOES WORK.
S1 switches on/off
temp reading still works.

this module also has S2 and S3 functionality but i do not know if these get affected as well wjhen you comment out the mapping command.

quick fix below

```
<!-- Map endpoints to instances -->
  <CommandClass id="96">
   <!-- <Compatibility>
   <!--    <MapRootToEndpoint>true</MapRootToEndpoint>
   <!--  </Compatibility>
  </CommandClass>
```

mdklapwijk, 2019-06-03:
Seem to have run into the same here, running the beta channel of Domoticz. Thought the module had failed, replaced it resulting in the same behaviour:
- configuration from Domoticz/OpenZWave Control Panel works as expected,
- power meters report correctly, when switching the relay via i1 and push button,
- 'node not found' in the Domoticz web interface, when switching the relay via the interface,
- in the 'OpenZWave Control Panel' within Domoticz, the module has an empty 'current' page.

Removing, the compatibility and maprootendpoint as suggested above did not resolve this (above example also seems to use incomplete comment-tags).

Switching back to the stable channel of Domoticz (./updaterelease) seems to fix the issue, as the relay is controllable from software again and the 'current' page is also working/showing properly again.

Fishwaldo, 2019-06-03:
@mdklapwijk Can you provide full log files of it working/notworking (and remove zwcfg* on the Stable Version and ozwcache* on the beta version before starting up the application).

also - if you could provide the Firmware Version the device reports. I wonder if we are dealing with different firmware bugs here.

mdklapwijk, 2019-06-03:
@Fishwaldo, as it is my main domotica system at this moment I'd rather not. Will check if I can do a test setup with spare parts though. Only that will have to wait till next weekend.

GravityRZ, 2019-06-03:
here are mine but i think you already have them. look for an exception error
module 11 is the qubino module

[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/3248117/OZW_Log.txt)
[ozwcache_0xcea80b03.txt](https://github.com/OpenZWave/open-zwave/files/3248118/ozwcache_0xcea80b03.txt)

Fishwaldo, 2019-06-04:
Please try leaving the Compatibility Block for CommandClass 96, but remove all the CommandClass Blocks below, remove your ozwcache file and restart
`  <CommandClass id="48">
    <Instance index="2"/>
    <Value genre="user" index="0" instance="1" label="Int2" read_only="true" type="bool" value="false" write_only="false"/>
    <Value genre="user" index="0" instance="2" label="Int3" read_only="true" type="bool" value="false" write_only="false"/>
    <SensorMap index="0" type="8"/>
    <Compatibility>
      <CreateVars>false</CreateVars>
    </Compatibility>
  </CommandClass>
  <CommandClass id="37">
    <Instance index="1"/>
    <Value genre="user" index="0" instance="1" label="Switch" read_only="false" type="bool"/>
    <Compatibility>
      <CreateVars>false</CreateVars>
    </Compatibility>
  </CommandClass>
`

GravityRZ, 2019-06-05:
ok, will try it as soon as they get the scene activation fixed for the fgd212

these 2 problems  are the reason i keep instaling v1.4 zwave versions.

petergebruers, 2019-06-05:
I think SceneActivation is functional now. See https://github.com/domoticz/domoticz/issues/3240#issuecomment-498744136

GravityRZ, 2019-06-07:
i am not sure if it is fixed. i still see discussions regarding delays. also i the thread youm ention i do not see anybody stating that it is permanently fixed

GravityRZ, 2019-06-08:
@petergebruers

tested it with build v4.10880 and it seems to work
S1 switch works
temp reading works (took  a while before a reading came in)
can you modify the config file in the build so we do not havr to change it all the times

petergebruers, 2019-06-09:
Thank you for testing that new build. If you post your working, modified, ZMNHADx.xml here I can take care of the "pull request". Because I'm rather new to this I'll double check with @Fishwaldo to get it merged.

One small thing though... What about Input 2 and 3? I don't own this module so I cannot test it...

GravityRZ, 2019-06-09:
Here it is(renamed to .txt)

i do not physically use the 3 switch connections (I1,I2,I3)
i am only using 1 switch in domoticz which i gues is I1
i tried other devices but they do not seems to respond

in the past the config file for this module has been buggy. several things were tried at that time resulting in extra switches, temp sensors which were created but did not work

at the moment nobody else is complaining about this module so i think for now we can at least fix it so both I1 and temp input works. 

[ZMNHADx.txt](https://github.com/OpenZWave/open-zwave/files/3269785/ZMNHADx.txt)

closed 2019-06-11 by Fishwaldo

petergebruers, 2019-06-24:
Possibly related.

https://github.com/domoticz/domoticz/issues/3348

Fishwaldo, 2019-06-26:
Works for me! :)

Fishwaldo, 2019-06-26:
I just tested a ZMNHAD1 H1S6P2 and it worked fine (although I don't have a Temp Sensor Probe). You need to ensure that the controller is part of Group 9 according to the Manual.

GravityRZ, 2019-06-29:
tested with build 935
working also
no more errors
checked config file and indeed the last part is deleted
temp working correctly and indeed you need to add it to group 9 for this

PR #1832 (2019-06-11): Update Qubino ZMNHADx
Fixes #1820
commit baeeb8dd (2019-06-11): Update Qubino ZMNHADx (#1832)

Fixes #1820