OpenZWave/open-zwave #1830: FGMS001-ZW5 only Lux reported and possibly Association Set issue

petergebruers, 2019-06-07:
Reported here:

https://groups.google.com/d/msg/openzwave/fCeEnEpFmsQ/6XR3bsGyBgAJ

This gist:
* used to work on OZW 1.4
* Logs analyzed by @Fishwaldo and problem triaged as "buggy device"
* Device is FGMS-001 ZW5 running latest firmware
* Log files available in google groups topic

Briefly discussed on Discord, Home-Assistant Z-Wave channel.

User is running Domoticz.

Status: need a device, and need time to test...

petergebruers, 2019-06-07:
At least 2 separate issues:
* Get always returns Lux. But this is not the main issue, because the device will report temp based on device parameters. This is a red herring.
* Lifeline group is empty and does not seem to accept setting. This disables unsolicited reporting. Which causes the "no temp" issue. 

Tested on Fibaro HC2: device does not accept MCA in lifeline, only SCA. If I set MCA +SCA the device in the BUI only shows SCA.

Discussed with @Fishwaldo - TODO: try ForceInstances false. Association worked on 1.4 because SCA was the default.

petergebruers, 2019-06-08:
The "only reporting Lux issue" is a bug in OZW affecting devices having COMMAND_CLASS_SENSOR_MULTILEVEL version >=5

Spec:

SDS13781 Z-Wave Application Command Class Specification
4.64.6 Multilevel Sensor Get Command

Get has 2 parameter bytes, the first is "Sensor Type" the second has "scale" parameter.  The total size is 4 bytes. At present, OZW sends only 3 bytes. I've implemented a quick fix in test, always sending scale 0. After this, SENSOR_MULTILEVEL_GET during startup retrieves the correct sensors:

```
2019-06-08 11:02:17.253 Info, Node002, Received SensorMultiLevel report from node 2, instance 1, Temperature: value=25.3C
2019-06-08 11:02:17.296 Info, Node002, Received SensorMultiLevel report from node 2, instance 1, Luminance: value=22lux
2019-06-08 11:02:17.339 Info, Node002, Received SensorMultiLevel report from node 2, instance 1, Seismic Intensity: value=0.0mercalli
2019-06-08 11:02:17.382 Info, Node002, Received SensorMultiLevel report from node 2, instance 1, Acceleration X-axis: value=0.0m/s2
2019-06-08 11:02:17.425 Info, Node002, Received SensorMultiLevel report from node 2, instance 1, Acceleration Y-axis: value=0.0m/s2
2019-06-08 11:02:17.467 Info, Node002, Received SensorMultiLevel report from node 2, instance 1, Acceleration Z-axis: value=0.0m/s2
```
I have to think about a proper fix.
I shall now investigate the multichannel association issue.

Fishwaldo, 2019-06-08:
I’m going to argue the “bug in OZW” statement. It should still work with Versions Prior to 5. 
The Get Message is perfectly valid for Versions upto that.

As mentioned on the code - I think we need to re-write the CommandClass (already planned). Hardcoding the Scale type to 0 gives me jitters on what it will break :)

So yes - we need to properly implement the Supported_Scale_Get/Report commands etc. But I’m not going to waste my time doing that on the existing codebase.

petergebruers, 2019-06-08:
I totally understand, I see you already have an issue for a rework of the sensor CC.

https://github.com/OpenZWave/open-zwave/issues/1569

petergebruers, 2019-06-11:
> I’m going to argue the “bug in OZW” statement. It should still work with Versions Prior to 5.

I have reworded my statement, the word "bug" is indeed not appropriate...

petergebruers, 2019-06-13:
FGMS-001 should work after merge of:

https://github.com/OpenZWave/open-zwave/pull/1839
https://github.com/OpenZWave/open-zwave/pull/1837

petergebruers, 2019-06-26:
FGMS-001 is fully functional in OZW, all sensors and alarms reported.

closed 2019-06-26 by petergebruers