OpenZWave/open-zwave #1847: Door lock cause OZW driver fatal crash

lily148, 2019-06-17:
Using the latest OZW 1.6 self compile version.
Can't add door lock, door lock successfully added but when refresh information. OZW driver crash at 

```
2019-06-17 22:40:43.828 Info, Node034, Received SecurityCmd_NonceReport from node 34
2019-06-17 22:40:43.828 Info, Node034, Encrypted Flag is 1
2019-06-17 22:40:43.828 Info, Node034, Sending (Query) message (Callback ID=0x78, Expected Reply=0x04) - UserNumberCmd_Get (Node=34): 0x01, 0x09, 0x00, 0x13, 0x22, 0x02, 0x63, 0x04, 0x25, 0x78, 0xff
2019-06-17 22:40:43.840 Detail, Node034,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2019-06-17 22:40:43.840 Detail, Node034,   ZW_SEND_DATA delivered to Z-Wave stack
2019-06-17 22:40:43.869 Detail, Node034,   Received: 0x01, 0x18, 0x00, 0x13, 0x78, 0x00, 0x00, 0x03, 0x00, 0xd0, 0x7f, 0x7f, 0x7f, 0x7f, 0x01, 0x01, 0x03, 0x00, 0x00, 0x00, 0x00, 0x02, 0x01, 0x00, 0x00, 0x5f
2019-06-17 22:40:43.869 Detail, Node034,   ZW_SEND_DATA Request with callback ID 0x78 received (expected 0x78)
2019-06-17 22:40:43.869 Info, Node034, Request RTT 93 Average Request RTT 84
2019-06-17 22:40:43.869 Detail, Node034, Extended TxStatus: Time: 30, Hops: 0, Rssi:  -48 --- --- --- ---, AckChannel: 1, TxChannel: 1, RouteScheme: Last Working Route, Route: 0 0 0 0, RouteSpeed: 40K, RouteTries: 1, FailedLinkFrom: 0, FailedLinkTo: 0
2019-06-17 22:40:43.870 Detail, Node034,   Expected callbackId was received
2019-06-17 22:40:43.909 Detail, Node034,   Received: 0x01, 0x0a, 0x00, 0x04, 0x00, 0x22, 0x02, 0x98, 0x40, 0xca, 0x00, 0xc3
2019-06-17 22:40:43.909 Info, Node034, Received SecurityCmd_NonceGet from node 34
2019-06-17 22:40:43.909 Info, NONCES: 0x2e, 0x04, 0xd6, 0x9c, 0x52, 0xe0, 0x3d, 0x0c
2019-06-17 22:40:43.909 Info, NONCES: 0x5c, 0xee, 0x53, 0xda, 0x01, 0x16, 0x49, 0xde
2019-06-17 22:40:43.909 Info, NONCES: 0xa0, 0x6f, 0x5a, 0xf2, 0x50, 0x98, 0xff, 0x5a
2019-06-17 22:40:43.910 Info, NONCES: 0x88, 0x6b, 0x84, 0x88, 0x82, 0xcd, 0x67, 0x32
2019-06-17 22:40:43.910 Info, NONCES: 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
2019-06-17 22:40:43.910 Info, NONCES: 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
2019-06-17 22:40:43.910 Info, NONCES: 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
2019-06-17 22:40:43.910 Info, NONCES: 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00
2019-06-17 22:40:43.911 Info, Node034, Sending (Query) message (Callback ID=0x01, Expected Reply=0x04) - Nonce_Report - 0x01, 0x11, 0x00, 0x13, 0x22, 0x0a, 0x98, 0x80, 0x88, 0x6b, 0x84, 0x88, 0x82, 0xcd, 0x67, 0x32, 0x05, 0x01, 0x3c:
2019-06-17 22:40:43.919 Detail, Node034,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2019-06-17 22:40:43.920 Detail, Node034,   ZW_SEND_DATA delivered to Z-Wave stack
2019-06-17 22:40:43.946 Detail, Node034,   Received: 0x01, 0x18, 0x00, 0x13, 0x01, 0x00, 0x00, 0x03, 0x00, 0xd0, 0x7f, 0x7f, 0x7f, 0x7f, 0x01, 0x01, 0x03, 0x00, 0x00, 0x00, 0x00, 0x02, 0x01, 0x00, 0x00, 0x26
2019-06-17 22:40:43.946 Detail, Node034,   ZW_SEND_DATA Request with callback ID 0x01 received (expected 0x01)
2019-06-17 22:40:43.946 Info, Node034, Request RTT 170 Average Request RTT 127
2019-06-17 22:40:43.947 Detail, Node034, Extended TxStatus: Time: 30, Hops: 0, Rssi:  -48 --- --- --- ---, AckChannel: 1, TxChannel: 1, RouteScheme: Last Working Route, Route: 0 0 0 0, RouteSpeed: 40K, RouteTries: 1, FailedLinkFrom: 0, FailedLinkTo: 0
2019-06-17 22:40:43.970 Detail, Node034,   Received: 0x01, 0x1f, 0x00, 0x04, 0x00, 0x22, 0x17, 0x98, 0x81, 0x96, 0xc7, 0x0c, 0xcb, 0xbd, 0x84, 0x0e, 0xdb, 0x1c, 0x83, 0xbe, 0x8a, 0x88, 0xcf, 0xd9, 0x1e, 0x2c, 0xbd, 0x02, 0x39, 0xa5, 0xcf, 0x00, 0x59
2019-06-17 22:40:43.970 Info, Raw: 0x98, 0x81, 0x96, 0xc7, 0x0c, 0xcb, 0xbd, 0x84, 0x0e, 0xdb, 0x1c, 0x83, 0xbe, 0x8a, 0x88, 0xcf, 0xd9, 0x1e, 0x2c, 0xbd, 0x02, 0x39, 0xa5, 0xcf
2019-06-17 22:40:43.971 Detail, Node034, Decrypted Packet: 0x00, 0x63, 0x05, 0x1e
2019-06-17 22:40:43.971 Detail,
2019-06-17 22:40:43.971 Info, Node034, Response RTT 194 Average Response RTT 169
2019-06-17 22:40:43.971 Info, Node034, Received User Number report from node 34: Supported Codes 30 (30)
2019-06-17 22:40:43.971 Detail, Node034, Initial read of value
*** Error in `/opt/domoticz/domoticz': malloc(): memory corruption: 0xa7a4ea48 ***
Fatal Python error: Aborted

Thread 0xb5034430 (most recent call first):
2019-06-17 22:40:43.973  Error: Domoticz(pid:17233, tid:17257('OZW-driver')) received fatal signal 6 (Aborted)
2019-06-17 22:40:43.974  Error: siginfo address=0x4351, address=0xb6c754b9

```

Fishwaldo, 2019-06-17:
Can you reproduce this crash with ozwcp please? if so, I need a full OZW_Log.txt file to diagnose.

Fishwaldo, 2019-06-17:
And if possible, a GDB Backtrace.

lily148, 2019-06-18:
@Fishwaldo I can reproduce the crash because it crash every time I start the system. Because of that, I exclude it from the network already. I will include into the network again when I back from work and send you a complete log.

lily148, 2019-06-18:
@Fishwaldo below is the full crash from ozwcp.  Besides the crash, I notice there is one warning in the log too.
`2019-06-19 00:11:28.875 Warning, Node032, MAC Authentication of Packet Failed. Dropping`

```
2019-06-19 00:10:38.039 Info, Loading Localization File /opt/domoticz/Config/Localization.xml
2019-06-19 00:10:38.044 Info, Loaded /opt/domoticz/Config/Localization.xml With Revision 7
2019-06-19 00:10:38.045 Always, Using Language Localization 
2019-06-19 00:10:38.051 Info, Loading NotificationCCTypes File /opt/domoticz/Config/NotificationCCTypes.xml
2019-06-19 00:10:38.052 Info, Loaded /opt/domoticz/Config/NotificationCCTypes.xml With Revision 9
2019-06-19 00:10:38.053 Info, Setting Up Provided Network Key for Secure Communications
2019-06-19 00:10:38.100 Info, Manufacturer_Specific.xml file Revision is 29
2019-06-19 00:10:38.991 Info, Product name collision: FGR223 Roller Shutter Controller 3 type 303 id 1000 manufacturerid 10f, collides with FGRM223 Roller Shutter Controller 3, type 303 id 1000 manufacturerid 10f
2019-06-19 00:10:39.529 Info, Product name collision: ED2.0 Meter Adapter type 0 id 0 manufacturerid 128, collides with ED2.0 Meter Adapter, type 0 id 0 manufacturerid 128
2019-06-19 00:10:39.530 Info, Product name collision: ED2.0 Display type 128 id 0 manufacturerid 128, collides with ED2.0 Display, type 128 id 0 manufacturerid 128
2019-06-19 00:10:39.735 Info, Error in Product Config file at line 1 - missing Revision  attribute
2019-06-19 00:10:39.812 Info, Product name collision: GR105 Auto Valve type 202 id 511 manufacturerid 152, collides with GR105 Auto Valve, type 202 id 511 manufacturerid 152
2019-06-19 00:10:39.813 Info, Product name collision: GR105n Auto Valve type 3 id 512 manufacturerid 152, collides with GR105n Auto Valve, type 3 id 512 manufacturerid 152
2019-06-19 00:10:39.813 Info, Product name collision: ZSE02 Motion Sensor type 500 id 3 manufacturerid 152, collides with ZSE02 Motion Sensor, type 500 id 3 manufacturerid 152
2019-06-19 00:10:39.855 Info, Product name collision: ZWN-BPC Ceiling Mounted PIR Motion Sensor type 601 id 901 manufacturerid 11a, collides with ZWN-BPC Ceiling Mounted PIR Motion Sensor, type 601 id 901 manufacturerid 11a
2019-06-19 00:10:39.855 Info, Product name collision: ZWN-SC7 7-Button Scene Controller type 801 id b03 manufacturerid 11a, collides with ZWN-SC7 7-Button Scene Controller, type 801 id b03 manufacturerid 11a
2019-06-19 00:10:39.855 Info, Product name collision: ZW15S 15A On/Off Switch type 101 id 102 manufacturerid 11a, collides with ZW15S 15A On/Off Switch, type 101 id 102 manufacturerid 11a
2019-06-19 00:10:39.855 Info, Product name collision: ZW20R 20A TR Duplex Receptacle type 101 id 603 manufacturerid 11a, collides with ZW20R 20A TR Duplex Receptacle, type 101 id 603 manufacturerid 11a
2019-06-19 00:10:39.855 Info, Product name collision: ZWN-RSM1 PLUS-Smart Single Relay Switch Module type 111 id 605 manufacturerid 11a, collides with ZWN-RSM1 PLUSâ€”Smart Single Relay Switch Module, type 111 id 605 manufacturerid 11a
2019-06-19 00:10:39.855 Info, Product name collision: ZW20RM 20A TR Smart Meter Duplex Receptacle type 111 id 101 manufacturerid 11a, collides with ZW20RM 20A TR Smart Meter Duplex Receptacle, type 111 id 101 manufacturerid 11a
2019-06-19 00:10:39.856 Info, Product name collision: ZW500D 500W In-Wall Preset Dimmer Switch type 102 id 201 manufacturerid 11a, collides with ZW500D 500W In-Wall Preset Dimmer Switch, type 102 id 201 manufacturerid 11a
2019-06-19 00:10:39.856 Info, Product name collision: ZW15RM Plus 15A TR Smart Meter Duplex Receptacle type 111 id 105 manufacturerid 11a, collides with ZW15RM Plus 15A TR Smart Meter Duplex Receptacle, type 111 id 105 manufacturerid 11a
2019-06-19 00:10:39.957 Info, Queuing Lookup on mfs.db.openzwave.com for Node 0
2019-06-19 00:10:39.958 Info,   Opening controller /dev/serial/by-id/usb-0658_0200-if00
2019-06-19 00:10:39.958 Info, Trying to open serial port /dev/serial/by-id/usb-0658_0200-if00 (attempt 1)
2019-06-19 00:10:39.959 Info, mgr,     Added driver for controller /dev/serial/by-id/usb-0658_0200-if00
2019-06-19 00:10:39.959 Info, Starting DNSThread
2019-06-19 00:10:39.960 Info, LookupTxT Checking mfs.db.openzwave.com
2019-06-19 00:10:39.961 Info, Timer: thread starting
2019-06-19 00:10:39.961 Detail, Timer: waiting with timeout -1 ms
2019-06-19 00:10:39.962 Info, Serial port /dev/serial/by-id/usb-0658_0200-if00 opened (attempt 1)
2019-06-19 00:10:39.962 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9
2019-06-19 00:10:39.963 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_MEMORY_GET_ID: 0x01, 0x03, 0x00, 0x20, 0xdc
2019-06-19 00:10:39.963 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_CONTROLLER_CAPABILITIES: 0x01, 0x03, 0x00, 0x05, 0xf9
2019-06-19 00:10:39.963 Detail, contrlr, Queuing (Command) FUNC_ID_SERIAL_API_GET_CAPABILITIES: 0x01, 0x03, 0x00, 0x07, 0xfb
2019-06-19 00:10:39.963 Detail, contrlr, Queuing (Command) FUNC_ID_ZW_GET_SUC_NODE_ID: 0x01, 0x03, 0x00, 0x56, 0xaa
2019-06-19 00:10:39.977 Info, Lookup for mfs.db.openzwave.com returned 29
2019-06-19 00:10:39.988 Info, ManufacturerSpecificDB Initialized
2019-06-19 00:10:39.988 Detail, Notification: ManufacturerSpecificDB Ready
2019-06-19 00:10:39.989 Detail, 
2019-06-19 00:10:39.989 Info, contrlr, Sending (Command) message (Callback ID=0x00, Expected Reply=0x15) - FUNC_ID_ZW_GET_VERSION: 0x01, 0x03, 0x00, 0x15, 0xe9
[... cut ...]
2019-06-19 00:12:46.889 Info, Node032, Received User Number report from node 32: Supported Codes 30 (30)
2019-06-19 00:12:46.889 Detail, Node032, Initial read of value

```

Then crash with the message below:
```
*** Error in `./domoticz': malloc(): memory corruption: 0xb6b89708 ***
Fatal Python error: Aborted

Thread 0xb4fe4430 (most recent call first):
2019-06-19 00:12:46.891  Error: Domoticz(pid:2003, tid:2049('OZW-driver')) received fatal signal 6 (Aborted)
2019-06-19 00:12:46.891  Error: siginfo address=0x7d3, address=0xb6c254b9

```

Fishwaldo, 2019-06-18:
Could you please try removing lines 207 to 209 of cpp/src/UserCode.cpp, recompile and see if it still crashes?

lily148, 2019-06-19:
Okay, sure. Will test it today and feedback.

lily148, 2019-06-19:
@Fishwaldo tried removing those lines, no go... but I found out the crash is under line 174, during the for loop, it will crash at loop #19 seem like it is under silly compiler section. I tested the door lock with OZW v1.4, everything is okay.

```
if (Node* node = GetNodeUnsafe())
					{
						string data;

						for (uint16 i = 0; i <= m_dom.GetFlagByte(STATE_FLAG_USERCODE_COUNT); i++)
						{
							char str[16];
							if (i == 0)
							{
								snprintf(str, sizeof(str), "Enrollment Code");
								node->CreateValueString(ValueID::ValueGenre_User, GetCommandClassId(), _instance, i, str, "", true, false, data, 0);
							}
							else
							{
								snprintf(str, sizeof(str), "Code %d:", i);
								node->CreateValueString(ValueID::ValueGenre_User, GetCommandClassId(), _instance, i, str, "", false, false, data, 0);
							}
							m_userCode[i].status = UserCode_Available;
							/* silly compilers */
							for (int j = 0; j < 10; j++)
								m_userCode[i].usercode[i] = 0;
						}
						if (m_com.GetFlagBool(COMPAT_FLAG_UC_EXPOSERAWVALUE))
						{
							node->CreateValueRaw(ValueID::ValueGenre_User, GetCommandClassId(), _instance, ValueID_Index_UserCode::RawValue, "Raw UserCode", "", false, false, 0, 0, 0);
							node->CreateValueShort(ValueID::ValueGenre_User, GetCommandClassId(), _instance, ValueID_Index_UserCode::RawValueIndex, "Raw UserCode Index", "", false, false, 0, 0);
						}
					}
```

lily148, 2019-06-20:
@Fishwaldo It is confirm the silly compilers causing it crash. I removed the silly compilers, it went through the whole Node info refresh. But I can't set any passcode, is it something related to the silly compilers?
May I know what this silly compilers do?
Besides that, I try to increase or decrease the number of loop in silly compilers, still no go. Need your advice.

Fishwaldo, 2019-06-21:
Please try with the latest version. Stupid Typo.

lily148, 2019-06-22:
@Fishwaldo Yes, changing the value to j did solved the crash.
However, I still cannot set user code or remove user code from the OZW panel. But I believe it shouldn't related to this crash?

lily148, 2019-06-24:
Okay, I am able to set and change user code now.  What I did was just clearing the cache file and refresh node info again. One more problem is the remove user code button is not working.
When I click on the button, no response. Nothing work. No log or what so ever. I suspect it does not get trigger, either the button event is not register? or anything else I should look for? Need some advice.

Fishwaldo, 2019-06-25:
In Which Application Lily? If not ozwcp, please try that - as I can't be certian if domoticz has implemented the UserCode CC correctly after the 1.6 update.

lily148, 2019-06-25:
I trigger all my testing at ozwcp, the delete usercode button indeed dosen't fire up any action. No log, no error. Not sure is it, it doesn't match the if else statement or the event handler for that button is not working. Just to double confirm, the button type for clear usercode is "short" not "button" right?

Fishwaldo, 2019-06-25:
Its not a "button" per say. Its a Short ValueID. You enter the index of the UserCode you want to delete, and you should see some traffic when you submit. 

If you see traffic, but it doesn't delete the usercode, its possible this lock uses a "non-standard" or different way to clear usercodes (ASCII NULLS versus binary 0's versus clearing the Status etc etc etc). If thats the case, then we have to figure it out.

lily148, 2019-06-27:
I enter the UserCode index and click submit, no traffic at all.

Fishwaldo, 2019-06-28:
Sounds like a browser/cache issue. You should at least get a message from OZWCP about doing a SET.

lily148, 2019-06-28:
Okay, I did an inspection at the browser. The browser did post the value but no SET command from the log :(
[image]

petergebruers, 2019-06-30:
Hi! I'm helping Fishwaldo and I've been following the discussion. I am not sure what is going on with ozwcp but I've made an update a few minutes ago to fix "data passing". I don't have a door lock but I've tested with dimmers and switches and it is OK.

Would it be possible to try to build ozwcp from source? Then please check if you see activity in OZW_Log.txt, or at the bottom of ozwcp, as soon as you click on submit or toggle a value.

https://github.com/OpenZWave/open-zwave-control-panel

closed 2019-07-24 by no-response[bot]