OpenZWave/open-zwave #1849: Manager::IsPrimaryController, Manager::IsStaticUpdateController, Manager::IsBridgeController

kdschlosser, 2019-06-18:
This is a question leading to an addition suggestion.

By using Manager::GetNodeBasic we are able to tell what the node is.

BASIC_TYPE_CONTROLLER = 0x01 (Portable Controller)
BASIC_TYPE_STATIC_CONTROLLER = 0x02 (Static Controller)
BASIC_TYPE_SLAVE = 0x03 (Slave)
BASIC_TYPE_ROUTING_SLAVE = 0x04 (Routing Slave)

routing slave is everything that is not some form of a controller.

using this we are able to determine if a node is a controller or not.

These 3 methods exist but they only work on the USB Stick it's self.

* Manager::IsPrimaryController
* Manager::IsStaticUpdateController
* Manager::IsBridgeController

Is there any code in place that would provide the same information the 3 methods above do but for a node that is not the USB Stick and if it is not a routing slave?

If there is no mechanism in place is there a way to collect this information from a node and make it available via overloading of the above methods (adding a node id parameter)?

I know that it is not common place to see a network with more then a single controller unless there are portable controllers. In my use case I do happen to have more then one controller. I do not know if any additional commands can be run on the slave of a network or not. So at the moment it would be for informational purposes and possible addition of commands at a later date if any can be performed.

petergebruers, 2019-06-18:
Excellent question. I can explain why the current code does not have that per node. The code that determines those  IsPrimaryController and IsStaticUpdateController  properties uses a serial function called FUNC_ID_SERIAL_API_GET_INIT_DATA and it is described in "INS12350 Serial API Host Appl. Prg. Guide" under "7.4 Node List Command"

```
Capabilities flags:
Bit 0: 0 = Controller API; 1 = Slave API
Bit 1: 0 = Timer functions not supported; 1 = Timer functions supported. Bit 2: 0 = Primary Controller; 1 = Secondary Controller
Bit 3: 0 = Not SIS; 1 = Controller is SIS
Bit 4-7: reserved
```

This command does not accept a "node ID" so it's always referring to the attached (serial) controller.

The IsBridgeController uses FUNC_ID_ZW_GET_VERSION and that one is listed in INS13954-7, Z-Wave 500 Series Appl. Programmers Guide v6.81.0x under 4.3.2.18 ZW_Version

Again, this is a command that does not accept a Node ID and always applies to the attached controller.

> Is there any code in place that would provide the same information the 3 methods above do but for a node that is not the USB Stick and if it is not a routing slave?

I am not sure how to do that, but I guess an indirect way would be possible. For example, In Driver.cpp HandleGetSUCNodeIdResponse sets m_SUCNodeId and that field is retrieved through:

uint8 Manager::GetSUCNodeId(uint32 const _homeId)

I am not sure about "primary" - I must admit that I've only toyed a little bit with secondary static controllers (I do have portable ones), but I do know a little bit about SUC/SIS and primary controller shift is in INS13954-7, Z-Wave 500 Series Appl. Programmers Guide v6.81.0x under "5 APPLICATION NOTE: SUC/SIS IMPLEMENTATION" and "6 APPLICATION NOTE: CONTROLLER SHIFT IMPLEMENTATION"

Fishwaldo, 2019-06-18:
The Terminology "Primary" and "Secondary" doesn't really translate directly to the Z-Wave Roles. Is more of just a Friendly name we assign for interfaces. 

Essentially - Anything returning a Controller (Static or Portable) can manage the network. 

You then get into things like SUC and SIS. 
SUC - Static Update Controller - Contains the Routing Table for devices. Is necessary For Pre-ZWave+  devices so they can get a copy of how to reach different nodes. For Z-Wave Plus devices (or technically, nodes that support Explorer Frames) is not a requirement anymore. 
SIS - Basically just "hands out" new NodeID's when devices are added to a network. Again, a Pre-ZWave+ component thats not really used anymore. 

The SUC node is really the brains of the network.  That will tell you the "Primary" controller, but any other device that runs the Controller firmware can add/remove devices as well... No additional functionality is exposed by being a SUC or not. The only thing that might be relevent is "ControllerReplication" which basically moves the SUC role to another device.

closed 2019-06-22 by kdschlosser