OpenZWave/open-zwave #1852: Node ID of 0 for notification type ControllerCommand

kdschlosser, 2019-06-21:
I am not sure if this would be considered a bug or if it is by design. maybe it was simply missed and should be added.

If I use Manager::RequestNodeNeighborUpdate I supply a node ID to perform the operation on. Because commands can be queued that command can be called multiple times with different Node ID's. If this method and other methods that target specific nodes get called from different threads and each thread waits for the result via notification it is going to be impossible to tell which node the command completed for.

example:

thread 1 calls RequestNodeNeighborUpdate on node 10 and then waits for the notification that is of type ControllerCommand and has a Command type of RequestNodeNeighborUpdate

at the same (or close to the same) time thread 2 calls RequestNodeNeighborUpdate on node 12 and then waits for the notification that is of type ControllerCommand and has a Command type of RequestNodeNeighborUpdate

both threads will see the notification and not be able to distinguish what node the notification is for. 

I am sure that this issue can be handled in the application with having a thread safe queue that contains the node_ids and a thread lock to be able to access the container for different threads and so on and so forth. That is quite a bit of additional code that would need to be added to an application when OZW should be able to easily attach the node id that was supplied to the method to the notification

and for the methods that do not need a controller id supplied to them then the controller node id should be attached.

The reason I am asking about this is because in python-openzwave the OZW notifications have to get converted into python signals part of that conversion allows us to take that node id and get a python object instance that represents the node. this node then gets attached to the python signal as a sender. This will enable an application to register a callback for a signal and also supply a sender. it will filter the signals so only signals that are specific to that node and are of the correct notification type will make the callback.

petergebruers, 2019-06-21:
Not answering the question about the signaling (node 0 and matching notifications)... It is good practice (1) not to do this too often and (2) not to do anything else when doing RequestNodeNeighborUpdate on a single node. A node scan's all nodes in the network, then reports back as you have pointed out and this gets progressively slower when you have more nodes. Tests on my 90 node Fibaro network indicate many node's take 10 - 20 seconds to scan and report. Also, to complicate matters a bit, battery nodes will perform this when they wake up. On my network, a full mesh update would take about 20-30 minutes (also due to scheduling on the HC2 so might not apply to OZW) and during that the network is very, very sluggish.

kdschlosser, 2019-06-21:
I was only using RequestNodeNeighborUpdate as an example. there are other methods which use the ControllerCommand notification and also take a node id as a parameter

kdschlosser, 2019-08-11:
I believe we can close this now has it has been merged.

closed 2019-08-11 by kdschlosser