OpenZWave/open-zwave #1857: Qubino ZMNHIDx H1S2P1

jotakar, 2019-06-25:
Hello
I've just updated my OZWave controller. This thermostat worked fine, but the new config file has changed some parameters and now it doesn't work. 
Command Class 64: 
this model (H1S2P1) has two modes: off and Auto, and its value are 0 and 3 (not 1). Yes it's a strange value, but it works with this values, not with 1 (it looks like it uses the bit 1 and not bit 0)
Command Class 67:
It works for me with base 1 (default, I think). So I can set a thermostat temperature and auto mode set on/off the device.
Although there is a problem with units: it set ºF, although I set "C" in config run file for units (ozwcache_0xd15e57b9.xml). Some times it gets the right unit and value. 
Older versions had the same problem.
Command class 49 (this looks like a minor fail)
It returns "unknown" for label , although I set "Temperature" in config run file  
 
Anybody has using this device and had solve this problems?

Fishwaldo, 2019-06-25:
When you say you have updated you mean your running OZW 1.6?

Fishwaldo, 2019-06-25:
>  although I set "C" in config run file for units (ozwcache_0xd15e57b9.xml)
Editing the file has no effect. You must update the configuration via the ValueID's exposed by OZW
> It returns "unknown" for label , although I set "Temperature" in config run file
Again, its a cache file, not a config file. 

I have this device, I will get to testing it soon.

Lukas-19, 2019-06-25:
Thanks for your answer.
Yes, I updated to last OZW, 1.6. 
Oh!, I've just noticed that this is a cache file not a "config file". Then the right way is update via ValueID's exposed by OZW. I need study a bit ... ;)
Well, I wait for your news about test this product.
Thanks again.

Fishwaldo, 2019-06-26:
Sorry - I've just looked and I have a ZMNKID1, not HID1. (its a pre-release version). 

The HID version has two config files depending upon which Revision you have (S1 or S2). Please make sure your manufacturer_specific.xml file references the right version.

If your still having problems, please let us know what revision you have and provide a full OZW_Log file without the ozwcache present.

Thanks

jotakar, 2019-06-26:
This device is of 2016. 
My device is ZNMHID1 **H1S1P1**, I was confused because I saw S2 in config data. 
So I reviewed manufacturer_specific.xml as you suggested  and my device were disabled. I enabled it and restarted my domoticz without ozwcache.
Here is the file OZW_Log
[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/3331467/OZW_Log.txt)

And in ozwcfg_cache appears several  values **label="Unknown"** like
```
               <Value type="decimal" genre="user" instance="1" index="1" label="Unknown" units="" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="23.7">
                    <Help>Temperature Sensor Value</Help>
                </Value>
```
Not only for this device, also in others one.

Although help text and value are ok
And about units in Heating: at the begining I see units "F" for Thermostat Set Point, and after  submiting  several values in this field the units goes to C and values are valids. Some times units change again when I restarted my controlles (domoticz).
Thanks for your answer.

petergebruers, 2019-06-26:
> And in ozwcfg_cache appears several values **label="Unknown"** like

You are running Domoticz Dev version with a recent OZW version and it happens because your Domoticz Config folder is missing a few files. Yo can add them yourself. I have a PR for that but it wasn't merged yet:

https://github.com/domoticz/domoticz/pull/3341

Copy SensorMultiLevelCCTypes.xml from OZW to Domoticz Config/SensorMultiLevelCCTypes.xml. Also copy SensorMultiLevelCCTypes.xml

Then update the node (in Domoticz, hardware, Z-Wave panel).

It is not an OZW issue.

jotakar, 2019-06-26:
I've copied the files (SensorMultiLevelCCTypes.xml and .xsd) to the folder Config in domoticz. Now in ozw control panel I get a more complete info about nodes, but nodes values have unknown values yet. I see fields for temperature sensor with the help tip ok, but with label unknown. I dont know it the fault is in Domoticz or in OZW...

petergebruers, 2019-06-26:
Your ozwcache should look a lot better, less or no "Unknown" elements... but only after you do "Refresh Node Info". Alternatively you can stop Domoticz, delete ozwcache* file and start again, but then you'll have to wake up your battery devices. OZW will recreate the ozwcache based on settings of the device.

closed 2019-07-10 by Fishwaldo