OpenZWave/open-zwave #1861: clang 5 error: Node.h:566:32: error: ‘std::shared_ptr’ has not been declared

ekarak, 2019-06-29:
As seen in https://travis-ci.org/OpenZWave/node-openzwave-shared/jobs/552083887
I'm using clang for the travis build which errors out with:
```
In file included from /usr/local/include/openzwave/Driver.h:38:0,
                 from /usr/local/include/openzwave/Manager.h:39,
                 from ../src/openzwave.hpp:30,
                 from ../src/callbacks.cc:18:
/usr/local/include/openzwave/Node.h: At global scope:
/usr/local/include/openzwave/Node.h:566:32: error: ‘std::shared_ptr’ has not been declared
    void SetProductDetails(std::shared_ptr<Internal::ProductDescriptor> product);
                                ^
/usr/local/include/openzwave/Node.h:566:42: error: expected ‘,’ or ‘...’ before ‘<’ token
    void SetProductDetails(std::shared_ptr<Internal::ProductDescriptor> product);
                                          ^
/usr/local/include/openzwave/Node.h:621:4: error: ‘shared_ptr’ in namespace ‘std’ does not name a type
    std::shared_ptr<Internal::ProductDescriptor> m_Product;
    ^
```
gcc version
gcc (Ubuntu 4.8.4-2ubuntu1~14.04.3) 4.8.4

looks like [the 'memory' header from C++ std library](https://en.cppreference.com/w/cpp/memory/shared_ptr) is somehow being automatically brought in by gcc, but not with clang.

Solution might be to explicitly bring that in an #include statement.

Fishwaldo, 2019-06-29:
You need to ensure you are compiling with -std=c++11 passed to the compiler.

ekarak, 2019-06-29:
thank you Justin

closed 2019-06-29 by ekarak