OpenZWave/open-zwave #1862: FGMS001 ZW5 commandclass alarm messages flooding the log

GravityRZ, 2019-06-30:
ever since going from Zwave Version: 1.6-748-g339d53d7-dirty
to Version: 1.6-792-g711a83af-dirty

i have been getting zwave error messages on this module

the X,Y and Z axis messages were known but only appeared once in previous release.

now they appear regularly.
a new message which is triggered by the PIR sensor alarm is also logged repeatedly
i thought this was already a known issue but since i do not see an issue about this i am posting this.

log files included

`2019-06-29 21:09:20.958 Error: OpenZWave: Value_Changed: Unhandled value type ZDTYPE_SWITCH_NORMAL (2). Node: 16 (0x10), CommandClass: SENSOR MULTILEVEL, Label: X-Axis Acceleration, Instance: 1, Index: 52 (OpenZWave.cpp:2886)
2019-06-29 21:09:21.062 Error: OpenZWave: Value_Changed: Unhandled value type ZDTYPE_SWITCH_NORMAL (2). Node: 16 (0x10), CommandClass: SENSOR MULTILEVEL, Label: Y-Axis Acceleration, Instance: 1, Index: 53 (OpenZWave.cpp:2886)
2019-06-29 21:09:21.108 Error: OpenZWave: Value_Changed: Unhandled value type ZDTYPE_SWITCH_NORMAL (2). Node: 16 (0x10), CommandClass: SENSOR MULTILEVEL, Label: Y-Axis Acceleration, Instance: 1, Index: 53 (OpenZWave.cpp:2886)
2019-06-29 21:09:21.215 Error: OpenZWave: Value_Changed: Unhandled value type ZDTYPE_SWITCH_NORMAL (2). Node: 16 (0x10), CommandClass: SENSOR MULTILEVEL, Label: Z-Axis Acceleration, Instance: 1, Index: 54 (OpenZWave.cpp:2886)
2019-06-29 21:09:58.334 Error: OpenZWave: Value_Changed: Tried adding value, not succeeded!. Node: 16 (0x10), CommandClass: ALARM, Label: Previous Event Cleared, Instance: 1, Index: 256
2019-06-29 21:09:58.421 Error: OpenZWave: Value_Changed: Tried adding value, not succeeded!. Node: 16 (0x10), CommandClass: ALARM, Label: Previous Event Cleared, Instance: 1, Index: 256
2019-06-29 21:09:58.422 Error: OpenZWave: Value_Changed: Tried adding value, not succeeded!. Node: 16 (0x10), CommandClass: ALARM, Label: Previous Event Cleared, Instance: 1, Index: 256
2019-06-29 21:09:58.492 Error: OpenZWave: Value_Changed: Tried adding value, not succeeded!. Node: 16 (0x10), CommandClass: ALARM, Label: Previous Event Cleared, Instance: 1, Index: 256
2019-06-29 21:09:58.492 Error: OpenZWave: Value_Changed: Tried adding value, not succeeded!. Node: 16 (0x10), CommandClass: ALARM, Label: Previous Event Cleared, Instance: 1, Index: 256`

[domoticz.log](https://github.com/OpenZWave/open-zwave/files/3342619/domoticz.log)
[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/3342620/OZW_Log.txt)
[ozwcache_0xcea80b03.txt](https://github.com/OpenZWave/open-zwave/files/3342621/ozwcache_0xcea80b03.txt)

petergebruers, 2019-06-30:
Hi! Contrary to what you might think, this is a Domoticz issue, not an OZW problem. But I understand the confusion, Domoticz adds the prefix OpenZWave: Value_Changed: which sounds like it is OpenZWave having a problem.

https://github.com/domoticz/domoticz/issues/3360#issuecomment-506828741

What happens is OZW dutifully passes al alarms (Notifications) of the device to Domoticz, and not all cases are handled yet.

The reason you see this now is because, actually, both OZW and Domoticz have improved handling and logging! But the Notification class is a bit tricky (eg this device can report both tamper state and motion alarm).

Also the Acceleration should not be a switch, it is reported as a number. OZW does the right thing.

I've started a private conversation with https://github.com/gizmocuz about this.

closed 2019-06-30 by petergebruers

GravityRZ, 2019-06-30:
thanks  petergebruers

my mistake, did not look there and noticed you mentioned this.

just checked with Version: 1.6-800-g903c0c5b-dirty still there

petergebruers, 2019-06-30:
> just checked with Version: 1.6-800-g903c0c5b-dirty still there

yes, that is good, this means OZW is still working ;)