OpenZWave/open-zwave #1866: Neo Coolcam PIR Motion

poudenes, 2019-07-04:
Hi All,

I found out that the config for Neo Coolcam PIR Motion is not correct
`open-zwave/config/shenzen_neo/nas-pd02z.xml`

With a other user (he created the new XML) we have all the options for those motion sensors.

We changed the line in manufacturer_specific.xml

`<Product config="shenzen_neo/nas-pd02z.xml" id="108d" name="Battery Powered PIR Sensor V2" type="0003"/>`

to 

`<Product config="shenzen_neo/nas-pd02ze.xml" id="108d" name="Battery Powered PIR Sensor V2" type="0003"/>`

Added the new nas-pd02ze.xml in shenzen_neo/

```
<Product sourceFile="shenzen_neo/nas-pd02ze.xml">
  <CommandClass id="112">
    <Value type="byte" genre="config" instance="1" index="1" label="Sensitivity Level Setting" value="12" min="8" max="255" size="1">
       <Help>This parameter defines the sensitivity of PIR detector, it is recommended to test the detector with movements from a farthest end of the coverage area at first time of use.         If movements cannot be detected sensitively, simply adjust the sensitivity level with this parameter.         This Parameter can be configured with the value 8 trough 255, where 8 means high sensitivity and 255 means lowest sensitivity.       </Help>
    </Value>
    <Value type="short" genre="config" instance="1" index="2" label="On/Off Duration" units="second" value="30" min="5" max="600" size="2">
       <Help>This parameter can be determined how long the associated devices should stay ON status.         For instance, this parameter is set to 30(second), the PIR detector will send a BASIC_SET Command to an associated device with value basic set level if PIR detector is triggered and the associated device will be turned on 30(second) before it is turned off.         This Parameter value must be large than Parameter 6#.         If user set this parameter to default by Configure CC, the parameter #6 will be set to default value.         Available Settings:5 to 600(second).       </Help>
    </Value>
    <Value type="byte" genre="config" instance="1" index="3" label="Basic Set Level" value="255" min="0" max="255" size="1">
       <Help>Basic Set Command will be sent where contains a value when PIR detector is triggered, the receiver will take it for consideration; for instance, if a lamp module is received the Basic Set Command of which value is decisive as to how bright of dim level of lamp module shall be.         This Parameter is used to some associated devices.         Available Settings: 0, 1 to 99 or 255.       </Help>
    </Value>
    <Value type="list" genre="config" instance="1" index="4" label="PIR Detecting Function Enabled/Disabled" value="1" size="1">
       <Help>This parameter can be enabled or disabled the PIR detector detecting function.       </Help>
      <Item label="Disable" value="0" />
      <Item label="Enable" value="255" />
     </Value>
    <Value type="short" genre="config" instance="1" index="5" label="Ambient Illumination Lux Level" units="lux" value="100" min="0" max="1000" size="2">
       <Help>This parameter can be set a lux level value which determines when the light sensor is activated.         If the ambient illumination level falls below this value and a person moves across or within the detected area , PIR detector will send a Z-Wave ON command(i.e. BASIC_SET (value = parameter 3) to an associated device and activate it.         0 to 1000(Lux).       </Help>
    </Value>
    <Value type="byte" genre="config" instance="1" index="6" label="Re-trigger Interval Setting" units="second" value="8" min="1" max="8" size="1">
       <Help>This Parameter can be used to adjust the interval of being re-triggered after the PIR detector has been triggered.         This Parameter value must be less than Parameter 2#.         If user set this parameter to default by Configure CC, the parameter #2 will be set to default value.         Available Settings: 1 to 8(s).       </Help>
    </Value>
    <Value type="short" genre="config" instance="1" index="7" label="Light Sensor Polling Interval" units="second" value="180" min="60" max="36000" size="2">
       <Help>This Parameter can be set the light sensor measure ambient illumination level interval time.         NOTE: This Value Must Be less than Wakeup Interval Time.         Available Settings: 60 ~ 36000(second).       </Help>
    </Value>
    <Value type="list" genre="config" instance="1" index="8" label="Lux Level Function Enable" value="0" size="1">
       <Help>If this parameter is set to Enable, and when Lux level less than the value define by parameter #5, PIR detector will send a BASIC_SET command frame(i.e. BASIC_SET (value = parameter 3) to an associated device and activate it.         If Lux Level greater than the value define by parameter #5, PIR detector will not send a BASIC_SET command frame.</Help>
      <Item label="Disable" value="0" />
      <Item label="Enable" value="1" />
     </Value>
    <Value type="byte" genre="config" instance="1" index="9" label="Ambient Illumination Lux Level Report" units="lux" value="100" min="0" max="255" size="1">
      <Help>This parameter defines by how much Lux Level must change, in lux, to be reported to the main controller.</Help>
    </Value>
    <Value type="byte" genre="config" instance="1" index="10" label="Ambient Temperature Differential Level Report" units="0.1∞C" value="100" min="0" max="127" size="5">
      <Help>This parameter is configured the value that differential between current measured and previous report value. If the differential value larger than the settings, device will report this measured temperature value to nodes associated in lifeline.</Help>
    </Value>
    <Value type="list" genre="config" instance="1" index="11" label="Led Blink Enable" value="1" size="1">
      <Help>This parameter can enabled or disable the PIR led blinking function. </Help>
      <Item label="Disable" value="0" />
      <Item label="Enable" value="1" />
     </Value>
    <Value type="list" genre="config" instance="1" index="12" label="Motion Event Report One Time Enable" value="1" size="1">
      <Help>The motion detected event can be sent multiple times/only once until device report motion cleared event. </Help>
      <Item label="multiple times" value="0" />
      <Item label="only once" value="1" />
     </Value>
    <Value type="short" genre="config" instance="1" index="99" label="Ambient Light Intensity Calibration" value="1000" min="1" max="65536" size="2">
       <Help>This parameter defines the calibrated scale for ambient light intensity.</Help>
    </Value>
	 
  </CommandClass>
  <CommandClass id="133">
    <Associations num_groups="4">
      <Group index="1" max_associations="4" label="Lifeline" />
      <Group index="2" max_associations="4" label="Control Commands BASIC_SET" />
      <Group index="3" max_associations="4" label="Send Notification Report" />
      <Group index="4" max_associations="4" label="Send Sensor Binary Report" />
    </Associations>
  </CommandClass>
</Product>
```

petergebruers, 2019-07-04:
Thank you for reporting this. You can either wait for someone to pick up your request... Or you can try to open a "pull request" yourself. Detailed intsructions on those config files are here

https://github.com/OpenZWave/open-zwave/wiki/Adding-Devices

I had a quick look and because "type" and "id" do not change, I see no reason to introduce a new name for the config file, just edit the old one. Maybe I am missing something. Any reason to call it "pd02ze"?

poudenes, 2019-07-04:
Because the old one als use same xml file. If you check the manufacturer xml file you see that 2 product id's use same xml file.

petergebruers, 2019-07-04:
Thank you for clarifying that. I know the situation with the Neo Motion sensor is a bit "messy". There is an older version and a newer (they look different) with different IDs but then the new one exists in two variants, with or without temperature reporting. I am not sure but we would probably need to take into account the firmware version to differentiate. I own the second version with temp sensor. I noticed many documents float on the internet, all having slightly different parameters. Do you happen to know if they add parameters only, depending on revision? Then one config file could serve different variants. If they change the meaning or size of parameters then... we have a problem...

poudenes, 2019-07-04:
Oh I don't know.... I also have the newer version with temperature.

petergebruers, 2019-07-05:
Okay, as I suspected, we have a problem... There is indeed a "V2 with temperature sensor" and a "V2 without a temperature sensor" (older). Based on the current entries both devices use the same id="108d" and  type="0003" because they have different parameters, so OZW cannot see the difference.

The one without temp has p 10 = "Blink enable" and so the current config file is correct for all older devices without temp reporting.

As soon as they added temp, p 10 became "temperature diff reporting".

I have to think about it but there might be a solution: I could describe parameter 10 as "temperature diff reporting if the device supports temp and blink enable if it does not".

This device has no entry in the Alliance database https://products.z-wavealliance.org

And I cannot find a canonical source for the manual(s) either... This makes it more difficult to make (good) decisions.

> Because the old one als use same xml file. If you check the manufacturer xml file you see that 2 product id's use same xml file.

I've checked and I don't think we are looking at the same config/manufacturer_specific.xml file... the "master" version only has one reference to "pd02z.xml". Can you double check? BTW it is not unusual to have two different id/type pairs to point to the same xml file, they might be regional variants (frequencies) or hardware revisions using the same set of config options.

poudenes, 2019-07-06:
> I've checked and I don't think we are looking at the same config/manufacturer_specific.xml file... the "master" version only has one reference to "pd02z.xml". Can you double check? BTW it is not [...]

Youre right:
    <Product config="shenzen_neo/nas-pd02z.xml" id="108d" name="Battery Powered PIR Sensor V2" type="0003"/>

There is 1 line for the pd02z.xml for ID 108d

Here I have 1 old one and 6 new versions. I include the old one with old XML. After that change the line so it refer to new XML file and reboot my "Domoticz" and then add the new versions with new XML

For me problem was solved. But I can imagine when people have both versions and don't have the knowledge their stuck.

Fishwaldo, 2019-07-06:
I hope someone is complaining to the vendor about this. It’s not just a OZW issue. It’s going to affect every gateway out there.

poudenes, 2019-07-06:
You mean, if Neo Coolcam give every product a unique ID then the whole problem is solved?
Same PIR Motion but with a new feature, then we give the ID 1088d for example?

petergebruers, 2019-07-06:
@poudenes I may have solved the problem by crafting this config file... Would it be possible to revert your "manuf" file back to the original, then overwrite your nas-pd02z.xml with this version?

https://github.com/petergebruers/open-zwave/blob/nas-pd02z/config/shenzen_neo/nas-pd02z.xml

Then delete your cache file or refresh your nodes

> Here you can see the Product Identifier: NAS-PD01ZE with the E

So far I have not looked at the "PD01" device, based on your info and my experience this is about the "PD02" device which has the issues you describe. A similar problem might exist for the older "01" but I have no information. The PD01 and PD2 have different IDs so use different xml files...

I've noticed you started working on the PD02 file but then refer to the PD01 so not sure when that confusion happened...

> I hope someone is complaining to the vendor about this. It’s not just a OZW issue. It’s going to affect every gateway out there.

I am not sure how to contact them, this might be their official site

https://szneo.com

Not a lot info there...

The problem is not new, the PD02 was released "without temperature sensing" quite a while ago and then suddenly versions "with temperature sensor" appeared.

> You mean, if Neo Coolcam give every product a unique ID then the whole problem is solved?

Yes, when they change the meaning of parameters or functionality they should do that.

But this will not fix the devices that have been sold... I think the chances of getting firmware updates is pretty slim.

IMHO my version of the config, nas-pd02z.xml, based on your config suggestions (@poudenes ) still improves the situation. But it would be nice if you could test it!

petergebruers, 2019-07-06:
I suddenly realize all posts up to now do not properly explain the main issue and proposed fix.

So with same IDs there is a temperature model and a non-temperature model.

The device flashes a bright LED when it detects motion, and this can be disabled but they used a different parameter for this.

With the current nas-pd02z.xml there it is not possible to turn off the LED.

I propose a new config file with these parameters

```
index="10" label="Ambient Temperature Differential Level Report or Led Blink Enable (0 or 1 - non-temperature version)"
index="11" label="Led Blink Enable (temperature reporting version)"
```

Parameter 10 is the only "conflicting" parameter but imho that explanation should help. Parameters 11,12 and 99 are new

poudenes, 2019-07-06:
haha, correct. This explain the conflict correct!

petergebruers, 2019-07-07:
The issue referenced by @sincze is not related, his issue is having device ID 0000 which means the device has not been interviewed yet.

Here's what I will do. I'll wait another day to see if someone either has fundamental objections to the proposed fix, or if someone comes up with new information. If there is no objection, I'll merge it.

poudenes, 2019-07-07:
If there will be no merge. I know I can update the info myself when I need do a reinstall :)

closed 2019-07-10 by Fishwaldo