OpenZWave/open-zwave #1871: Stable release branches?

AlCalzone, 2019-07-09:
Now that everything seems to be back to normal, I would like to discuss how OZW works as a dependency for external libraries. 

Little reminder: The update of the master branch to version 1.6 broke external libraries that depend on openzwave-shared (https://github.com/OpenZWave/open-zwave/issues/1783). The reason is that this wrapper installs the master branch which had breaking changes in 1.6 and thus could not be compiled without changes to the wrapper.

For libraries that depend on openzwave-shared, this process is too fragile in my opinion. Is there a possibility that we can get stable release branches in the future that are guaranteed to not be updated with breaking changes? Going forward this would mean that there is a branch 1.6 that is kept in sync with master until the next breaking change. At that point, a new branch 1.8 is created and so on...
The wrapper should reflect this strategy in its versioning scheme, e.g. by switching to a new major version whenever a new OZW version is targeted.

Thoughts? @Fishwaldo

Fishwaldo, 2019-07-10:
Master is the stable branch. Only time things change is when I make a new minor version release (1.6, 1.7 etc). Dev Branch is usually where new versions are worked on.

We do not use GIT as a release platform. Anything of "production" quality should not be using HEAD on git for release builds. They should at least use a specific commit, or preferably, the downloads at http://old.openzwave.com/downloads/ which go through CI Build Tests etc etc etc. 

that being said - While I appreciate the inconvenience 1.6 caused on some - the Development of 1.5 sat in Dev branch for over 2 years - and despite many calls for testing - It didn't happen. They only way a lot of the bugs have shaken out of 1.6 is because its being used now. 

So Please work on a specific commit, or the release tarballs, and not on HEAD and you will not have any problems.

AlCalzone, 2019-07-10:
Fair enough. However we're dependent on the openzwave-shared wrapper from your org which installs the latest OZW master commit. Aside from forking it, we have no influence which commit is being targeted.

Fishwaldo, 2019-07-10:
For the sake of Clarity - I don't own anything! I just invite anyone developing something around OZW if they would like to host their repository under the OZW Organization (to make it easy for others to find related software etc). They maintain full admin access to their repositories etc and I don't interfere at all with their development.... so in this case, a ticket should be opened on the openzwave-shared repository and let @ekarak decide how he wishes to approach this.

closed 2019-07-10 by Fishwaldo