OpenZWave/open-zwave #1878: All NEO 3gangs and 2gangs switch missing individual switch control

lily148, 2019-07-13:
Hi, 
After the recent update, self compile the version on 12 Jul 2019. All my NEO 2 gangs and 3 gangs switch missing its individual control.
Not sure anything wrong..
I attached my log file....Node 37 is my 3 gangs switch
[OZW_Log.txt](https://github.com/OpenZWave/open-zwave/files/3388497/OZW_Log.txt)

lily148, 2019-07-13:
Checked on the log... it seem like it cannot get the less of the instance...
```
2019-07-13 13:19:35.719 Detail, Node037, AdvanceQueries queryPending=0 queryRetries=0 queryStage=CacheLoad live=1
2019-07-13 13:19:35.719 Detail, Node037, QueryStage_Associations
2019-07-13 13:19:35.719 Info, Node037, Number of association groups reported for node 37 is 4.
2019-07-13 13:19:35.719 Info, Node037, Get MultiChannelAssociation for group 1 of node 37
2019-07-13 13:19:35.719 Detail, Node037, Queuing (Send) MultiChannelAssociationCmd_Get (Node=37): 0x01, 0x0a, 0x00, 0x13, 0x25, 0x03, 0x8e, 0x02, 0x01, 0x25, 0xc1, 0xa9
2019-07-13 13:19:35.719 Detail, Node037, Queuing (Query) Query Stage Complete (CacheLoad)
.
.
.
.
.
2019-07-13 13:19:40.188 Info, Node037, Sending (Send) message (Callback ID=0xcf, Expected Reply=0x04) - MultiChannelAssociationCmd_Get (Node=37): 0x01, 0x0a, 0x00, 0x13, 0x25, 0x03, 0x8e, 0x02, 0x02, 0x25, 0xcf, 0xa4
2019-07-13 13:19:40.189 Info, Node037, Encrypted Flag is 0
2019-07-13 13:19:40.198 Detail, Node037,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2019-07-13 13:19:40.199 Detail, Node037,   ZW_SEND_DATA delivered to Z-Wave stack
2019-07-13 13:19:40.218 Detail, Node037,   Received: 0x01, 0x18, 0x00, 0x13, 0xcf, 0x00, 0x00, 0x02, 0x00, 0xbe, 0x7f, 0x7f, 0x7f, 0x7f, 0x00, 0x00, 0x03, 0x00, 0x00, 0x00, 0x00, 0x03, 0x01, 0x00, 0x00, 0x86
2019-07-13 13:19:40.218 Detail, Node037,   ZW_SEND_DATA Request with callback ID 0xcf received (expected 0xcf)
2019-07-13 13:19:40.218 Info, Node037, Request RTT 29 Average Request RTT 29
2019-07-13 13:19:40.218 Detail, Node037, Extended TxStatus: Time: 20, Hops: 0, Rssi:  -66 --- --- --- ---, AckChannel: 0, TxChannel: 0, RouteScheme: Last Working Route, Route: 0 0 0 0, RouteSpeed: 100K, RouteTries: 1, FailedLinkFrom: 0, FailedLinkTo: 0
2019-07-13 13:19:40.219 Detail, Node037,   Expected callbackId was received
2019-07-13 13:19:40.234 Detail, Node037,   Received: 0x01, 0x0d, 0x00, 0x04, 0x00, 0x25, 0x05, 0x8e, 0x03, 0x02, 0x01, 0x00, 0xbe, 0x00, 0xe6
2019-07-13 13:19:40.234 Detail, 
2019-07-13 13:19:40.234 Info, Node037, Response RTT 44 Average Response RTT 44
2019-07-13 13:19:40.234 Info, Node037, Received Multi Instance Association report from node 37, group 2
2019-07-13 13:19:40.234 Info, Node037,   The group contains:
2019-07-13 13:19:40.234 Info, Node037, Get MultiChannelAssociation for group 3 of node 37
2019-07-13 13:19:40.235 Detail, Node037, Queuing (Send) MultiChannelAssociationCmd_Get (Node=37): 0x01, 0x0a, 0x00, 0x13, 0x25, 0x03, 0x8e, 0x02, 0x03, 0x25, 0xdb, 0xb1
2019-07-13 13:19:40.235 Detail, Node037,   Expected reply and command class was received
2019-07-13 13:19:40.235 Detail, Node037,   Message transaction complete
2019-07-13 13:19:40.235 Detail, 
2019-07-13 13:19:40.235 Detail, Node037, Removing current message
2019-07-13 13:19:40.235 Detail, Node037, Notification: Group
.
.
.
.
2019-07-13 13:19:41.291 Info, Node037, Sending (Send) message (Callback ID=0xdb, Expected Reply=0x04) - MultiChannelAssociationCmd_Get (Node=37): 0x01, 0x0a, 0x00, 0x13, 0x25, 0x03, 0x8e, 0x02, 0x03, 0x25, 0xdb, 0xb1
2019-07-13 13:19:41.291 Info, Node037, Encrypted Flag is 0
2019-07-13 13:19:41.299 Detail, Node037,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2019-07-13 13:19:41.300 Detail, Node037,   ZW_SEND_DATA delivered to Z-Wave stack
2019-07-13 13:19:41.320 Detail, Node037,   Received: 0x01, 0x18, 0x00, 0x13, 0xdb, 0x00, 0x00, 0x02, 0x00, 0xbe, 0x7f, 0x7f, 0x7f, 0x7f, 0x00, 0x00, 0x03, 0x00, 0x00, 0x00, 0x00, 0x03, 0x01, 0x00, 0x00, 0x92
2019-07-13 13:19:41.320 Detail, Node037,   ZW_SEND_DATA Request with callback ID 0xdb received (expected 0xdb)
2019-07-13 13:19:41.320 Info, Node037, Request RTT 28 Average Request RTT 28
2019-07-13 13:19:41.320 Detail, Node037, Extended TxStatus: Time: 20, Hops: 0, Rssi:  -66 --- --- --- ---, AckChannel: 0, TxChannel: 0, RouteScheme: Last Working Route, Route: 0 0 0 0, RouteSpeed: 100K, RouteTries: 1, FailedLinkFrom: 0, FailedLinkTo: 0
2019-07-13 13:19:41.320 Detail, Node037,   Expected callbackId was received
2019-07-13 13:19:41.335 Detail, Node037,   Received: 0x01, 0x0d, 0x00, 0x04, 0x00, 0x25, 0x05, 0x8e, 0x03, 0x03, 0x01, 0x00, 0xbe, 0x00, 0xe7
2019-07-13 13:19:41.336 Detail, 
2019-07-13 13:19:41.336 Info, Node037, Response RTT 44 Average Response RTT 44
2019-07-13 13:19:41.336 Info, Node037, Received Multi Instance Association report from node 37, group 3
2019-07-13 13:19:41.336 Info, Node037,   The group contains:
2019-07-13 13:19:41.336 Info, Node037, Get MultiChannelAssociation for group 4 of node 37
2019-07-13 13:19:41.336 Detail, Node037, Queuing (Send) MultiChannelAssociationCmd_Get (Node=37): 0x01, 0x0a, 0x00, 0x13, 0x25, 0x03, 0x8e, 0x02, 0x04, 0x25, 0xe3, 0x8e
2019-07-13 13:19:41.336 Detail, Node037,   Expected reply and command class was received
2019-07-13 13:19:41.336 Detail, Node037,   Message transaction complete
2019-07-13 13:19:41.336 Detail, 
2019-07-13 13:19:41.336 Detail, Node037, Removing current message
2019-07-13 13:19:41.336 Detail, Node037, Notification: Group
.
.
.
.
```

petergebruers, 2019-07-13:
I've had a quick look and I am not sure about what's going on yet, but I remember we had a chat about one of your NEO device(s) a while ago:

https://github.com/domoticz/domoticz/issues/3324

To summarize, you have a "newer" version than me but then we were talking about Node 29, not 37.

So node 29 is ANZ frequency and is not in the database (I have to add it, it is still on my todo list).

Anyway, at that time you were unable to control that device because:

```
@lily148 your diagnosis is correct: your switch uses Version 2 of the Switch Binary command class and because of this, OpenZWave exposes additional properties.

V1:
ValueID_Index_SwitchBinary::Level
V2 adds:
ValueID_Index_SwitchBinary::Duration
ValueID_Index_SwitchBinary::TargetState

The way COpenZWave::AddValue is written won't work because of "bool" and "byte" are available.
```

So the problem was with Domoticz, I am not sure if that situation has changed.

Can you control the device(s) with ozwcp?

https://github.com/OpenZWave/open-zwave-control-panel

Also, can you please have a quick look at your ozwcache file, it should have been updated to version 4

```
<Driver xmlns="https://github.com/OpenZWave/open-zwave" version="4"
```

I did not find "WARNING: Driver::ReadCache - %s is from an older version of OpenZWave and cannot be loaded." in your log but OZW does that only once so I guess you've restarted.

petergebruers, 2019-07-13:
According to your log, both Node 29 and 37 do seem to report status with endpoints:

```
2019-07-13 13:20:04.795 Info, Node029, Received a MultiChannelEncap from node 29, endpoint 1 for Command Class COMMAND_CLASS_SWITCH_BINARY
2019-07-13 13:20:04.795 Info, Node029, Received SwitchBinary report from node 29: level=Off

2019-07-13 13:20:05.005 Info, Node029, Received a MultiChannelEncap from node 29, endpoint 2 for Command Class COMMAND_CLASS_SWITCH_BINARY
2019-07-13 13:20:05.005 Info, Node029, Received SwitchBinary report from node 29: level=On

2019-07-13 13:20:20.870 Info, Node037, Received a MultiChannelEncap from node 37, endpoint 1 for Command Class COMMAND_CLASS_SWITCH_BINARY
2019-07-13 13:20:20.870 Info, Node037, Received SwitchBinary report from node 37: level=On

2019-07-13 13:20:20.916 Info, Node037, Received a MultiChannelEncap from node 37, endpoint 2 for Command Class COMMAND_CLASS_SWITCH_BINARY
2019-07-13 13:20:20.916 Info, Node037, Received SwitchBinary report from node 37: level=Off

2019-07-13 13:20:20.963 Info, Node037, Received a MultiChannelEncap from node 37, endpoint 3 for Command Class COMMAND_CLASS_SWITCH_BINARY
2019-07-13 13:20:20.963 Info, Node037, Received SwitchBinary report from node 37: level=Off
```

So your issue is not about getting data from the switches, but about turning them on or off, right (you say "individual control")?

> it seem like it cannot get the less of the instance...

What is "the less" of an instance?

lily148, 2019-07-13:
For the earlier report, I fixed it by moving the node->CreateValueBool(ValueID::ValueGenre_User, G... 
to the top of version s2 code.
The problem, that I face now.... is different.... it does not create 3 instance of "on/off" switch for me.
Yes, the cache file Driver is version 4.
Below is what I get from ozwcp
[image]

lily148, 2019-07-13:
Could it be because the changing of the AddCommandClass, in commandclass.cpp and h?
This problem seem like happen after that commit...

petergebruers, 2019-07-13:
Not sure what commit you're referring to but probably related to merge of Dev.

I can reproduce your problem, but I'm not available to work on it right now. I've send data to @Fishwaldo 

I've also issued a statement and some explanation on the Domoticz forum about, you may be interested, see https://github.com/domoticz/domoticz/issues/3530