OpenZWave/open-zwave #1880: mcohome php220 not regognized

tonbor, 2019-07-13:
### Issue Guidelines

Before opening a new issue, please review the following:
* The Github Issues is for OZW Bugs only. Please do not use this for general questions or howto type conversations. If you have a question, please post to our mailing list at https://groups.google.com/forum/#!forum/openzwave
* Please use the search feature to see if your issues have been raised or addressed first.
* Try the OpenZWave Log Analyzer at http://www.openzwave.com/log-analyzer if you are experiencing problems with a device on your network.
* Often we will require Log Files to help diagnose the issue. Please stop any application you are using, remove zwcfg_*.xml/ozwcache_*.xml files, and restart your application. After experiencing your issue, stop your application and attach the OZW_Log.txt file to this issue (please do not filter the log file, we often need the full log file to diagnose issues)
* If you want to add a new device to the OZW Database, please consult https://github.com/OpenZWave/open-zwave/wiki/Adding-Devices

tonbor, 2019-07-13:
[image]
My config file (working) is changed from 1.4 to 1.6 merged with ozp 1.6 and added to manufacturer file many many thanks, but it will not load with new beta release Domoticz

petergebruers, 2019-07-14:
https://github.com/OpenZWave/open-zwave/pull/1853#issuecomment-511125937

tonbor, 2019-07-14:
See comment at my push request, please add the eu version id 5102

petergebruers, 2019-07-14:
https://github.com/OpenZWave/open-zwave/pull/1882

petergebruers, 2019-07-14:
Will be merged if test passes

petergebruers, 2019-07-14:
Merged. Please provide feedback here if you find the time to test it...

closed 2019-07-28 by no-response[bot]