OpenZWave/open-zwave #1883: Avoid saving 255 to groups

petergebruers, 2019-07-14:
Avoid adding (saving) the broadcast address to groups. It is unlikely that users add this deliberately but the address might appear on devices (reported by a few users, very rare, but causes great confusion).

petergebruers, 2019-07-28:
This was reported recently on the Hass ZWave Discord channel:

Zorkny Last Friday at 10:32 PM
Every now-and-the I have a qubino relay that gets associated with 255 (broadcast), this is not good as you can imagine as it calls all other zwave units when it turns on. Any idea why? I remove the association and it comes back again. Nuc, ubuntu, hassio-docker, aeon-zwave stick.

Also

Riemers 06/26/2019
So i have a weird issue where my motion sensor just turns the whole house on / off
now when i look at it, it shows that the basic report has an association with my stick, which is fine. But it also shows another: "Unknown node (255: (0 ? 255.0 : 255))"
since i cannot select that device, how do i remove that association?
looking in the logs, i see

```
                                    <Group index="1" max_associations="5" label="Basic Report" auto="true" multiInstance="true">
                                                <Node id="1" />
                                                <Node id="255" />
                                        </Group>
```

petergebruers, 2019-11-29:
https://groups.google.com/d/msg/openzwave/6uNtrryTX_0/DbzNi_DjAgAJ

```
29 november 2019 12:46:23 UTC+1 Stian S:
Hi, I have some times experienced that any sensor activation/deactivation will turn almost everything (!) in my house on/off. This has usually happened after restarting Home Assistant.

It just occurred again, and I see the following in OZW_Log.txt. To me, it seems like OpenZWave tries to add the controller (Node ID 1) as lifeline, but uses Node ID 255 instead, which is a broadcast ID if I understand correctly.

As this has happened before, I made a small script that removes 255 from all association groups. Hence, the full logs will show a lot of removals. Also, before I attach the full logs (if they are needed), should they be scrubbed of e.g. security related data (that is, is there any sensitive information in the logs?)

2019-11-29 12:32:32.118 Info, Node039, Sending (Send) message (Callback ID=0xae, Expected Reply=0x04) - AssociationCmd_Get (Node=39): 0x01, 0x0a, 0x00, 0x13, 0x27, 0x03, 0x85, 0x02, 0x04, 0x25, 0xae, 0xca
2019-11-29 12:32:32.130 Detail, Node039,   Received: 0x01, 0x04, 0x01, 0x13, 0x01, 0xe8
2019-11-29 12:32:32.130 Detail, Node039,   ZW_SEND_DATA delivered to Z-Wave stack
2019-11-29 12:32:32.147 Detail, Node039,   Received: 0x01, 0x07, 0x00, 0x13, 0xae, 0x00, 0x00, 0x02, 0x47
2019-11-29 12:32:32.147 Detail, Node039,   ZW_SEND_DATA Request with callback ID 0xae received (expected 0xae)
2019-11-29 12:32:32.147 Info, Node039, Request RTT 28 Average Request RTT 27
2019-11-29 12:32:32.147 Detail,   Expected callbackId was received
2019-11-29 12:32:32.158 Detail, Node039,   Received: 0x01, 0x0b, 0x00, 0x04, 0x00, 0x27, 0x05, 0x85, 0x03, 0x04, 0x04, 0x00, 0x54
2019-11-29 12:32:32.158 Detail, 
2019-11-29 12:32:32.158 Info, Node039, Response RTT 39 Average Response RTT 43
2019-11-29 12:32:32.158 Info, Node039, Received Association report from node 39, group 4, containing 0 associations
2019-11-29 12:32:32.158 Info, Node039, Querying associations for node 39 is complete.
2019-11-29 12:32:32.158 Info, Node039, Adding the controller to group 1 (Lifeline) of node 39
2019-11-29 12:32:32.158 Info, Node039, Association::Set - Adding node 255 to group 1 of node 39
2019-11-29 12:32:32.158 Detail, Node039, Queuing (Send) AssociationCmd_Set (Node=39): 0x01, 0x0b, 0x00, 0x13, 0x27, 0x04, 0x85, 0x01, 0x01, 0xff, 0x25, 0xaf, 0x34
2019-11-29 12:32:32.158 Info, Node039, Get Associations for group 1 of node 39
2019-11-29 12:32:32.158 Detail, Node039, Queuing (Send) AssociationCmd_Get (Node=39): 0x01, 0x0a, 0x00, 0x13, 0x27, 0x03, 0x85, 0x02, 0x01, 0x25, 0xb0, 0xd1
2019-11-29 12:32:32.158 Detail, Node039,   Expected reply and command class was received
2019-11-29 12:32:32.158 Detail, Node039,   Message transaction complete
2019-11-29 12:32:32.158 Detail, 
2019-11-29 12:32:32.158 Detail, Node039, Removing current message
2019-11-29 12:32:32.158 Detail, Node039, Notification: Group
```

petergebruers, 2019-11-29:
@Fishwaldo I checked out 1.4 branch and I am confused...

`Received Association report from node 39, group 4, containing 0 associations`

Okay, done enumerating groups, nothing special here.

`Adding the controller to group 1 (Lifeline) of node 39`

Okay, that's Node::AutoAssociate ...

`Node039, Association::Set - Adding node 255 to group 1 of node 39`

Errrrm? How did we get node 255?

I would say the preceding code is 

```cpp
uint8 controllerNodeId = GetDriver()->GetControllerNodeId();
(...)
// Associate the controller into the group
Log::Write( LogLevel_Info, m_nodeId, "Adding the controller to group %d (%s) of node %d", group->GetIdx(), group->GetLabel().c_str(), GetNodeId() );
group->AddAssociation( controllerNodeId );
```

That would mean in Driver we have m_Controller_nodeId at default which is -1. 'Default' is the wrong word here, m_Controller_nodeId gets set to unit8 -1 in the Driver::Init function.

AFAIK m_Controller_nodeId only gets set here:

```cpp
void Driver::HandleMemoryGetIdResponse
(
		uint8* _data
)
{
	Log::Write( LogLevel_Info, GetNodeNumber( m_currentMsg ), "Received reply to FUNC_ID_ZW_MEMORY_GET_ID. Home ID = 0x%02x%02x%02x%02x.  Our node ID = %d", _data[2], _data[3], _data[4], _data[5], _data[6] );
	m_homeId = ( ( (uint32)_data[2] )<<24 ) | ( ( (uint32)_data[3] )<<16 ) | ( ( (uint32)_data[4] )<<8 ) | ( (uint32)_data[5] );
	m_Controller_nodeId = _data[6];
	m_controllerReplication = static_cast<ControllerReplication*>(ControllerReplication::Create( m_homeId, m_Controller_nodeId ));
}
```

It is read from the controller... Can this fail and leave m_Controller_nodeId at -1?

Related to this... Would it be sane to change the default of m_Controller_nodeId (in the Driver::Init function) to zero or remove this line altogether in Driver::Init because the constructor already has m_Controller_nodeId ( 0 )

AFAIK nodeID 0 almost certainly means "no valid node ID".

I have to think...

Fishwaldo, 2019-12-01:
>It is read from the controller... Can this fail and leave m_Controller_nodeId at -1?

If it did fail, we wouldn't have the HomeID either, so most of OZW wouldn't function at all.

But if the controller returned 255 as the "Controller ID" then that's different... (and possible). Why it would return 255 I have no idea though? Bug? Error Condition? 

Given the code path you diagnosed, setting the default from (int8_t)-1 (which equals (uint8_t)255)) to 0 may not fix "the issue" as m_Controller_nodeId is explicitly set to the 6th byte of FUNC_ID_ZW_MEMORY_GET_ID and the fact we need the HomeID to function tells me that that Function is working...

I asked on the google groups if he had a full log file for this... lets hope so.

Putting a check for node 255 in AddAssociation will fix the broadcast issues, but if m_Controller_nodeId is wrong, then other things are also broken (like auto adding associations to the lifeline group for the controller in this example)

petergebruers, 2019-12-01:
I'm with you on this. I'll try to investigate further. In the group functions, I would ignore IDs outside 1-232 range and log a warning. What about doing the same test in the FUNC_ID_ZW_MEMORY_GET_ID decoder? Should I make it throw?

petergebruers, 2019-12-01:
Did a quick test, and as expected if you comment out that "m_Controller_nodeId = _data[6];" in Driver::HandleMemoryGetIdResponse then imho most things in OZW appear to work normal, except for assoc functions...

```
Adding the controller to group 1 (Lifeline) of node 2
...
Sending (Send) message (Callback ID=0x25, Expected Reply=0x13) - MultiChannelAssociationCmd_Set (Node=2): 0x01, 0x0d, 0x00, 0x13, 0x02, 0x06, 0x8e, 0x01, 0x01, 0x00, 0xff, 0x01, 0x25, 0x25, 0x95
...
Received MULTI_CHANNEL_ASSOCIATION_REPORT from node 2, group 1
...
The group contains:
  Node 255 End Point 1
```

But apart from that OZW appears functional. The only other anomaly I could spot is that OZW tries to interview device 1 which (obviously) fails.

```
2019-12-01 19:01:19.327 Info, Node001, Sending (NoOp) message (Callback ID=0x0a, Expected Reply=0x13) - NoOperation_Set (Node=1): 0x01, 0x09, 0x00, 0x13, 0x01, 0x02, 0x00, 0x00, 0x25, 0x0a, 0xc9
...
2019-12-01 19:01:22.331 Error, Node001, ERROR: Dropping command, expected response not received after 1 attempt(s). Command: "NoOperation_Set (Node=1): 0x01, 0x09, 0x00, 0x13, 0x01, 0x02, 0x00, 0x00, 0x25, 0x0a, 0xc9"
```

This might not be very obvious when you're not looking at the log with this problem in mind...

Fishwaldo, 2019-12-02:
From the mailing list:
> By the way, as I recall from when this happened before, I saw something like "unknown HomeID" in the logs.

Which indicates the initialization is failing in some way.

petergebruers, 2019-12-02:
> Which indicates the initialization is failing in some way.

That's the "smoking gun" indeed. Then OZW continues with -1 aka 255 and homeID either 0 (or random homeID if it is a controller bug, but I doubt that). Then OZW tries to add 255 to the groups.

I'v tested OZW 1.6 with everything in Driver::HandleMemoryGetIdResponse commented out and this leads to 

```
Notification: Driver Ready, homeId 00000000, nodeId 255
...
Node003, MultiChannelAssociation::Set - Adding End Point 1 on node 255 to group 1 of node 3
...
Node003, Received MULTI_CHANNEL_ASSOCIATION_REPORT from node 3, group 1
  The group contains:
     Node 255 End Point 1

```

A restart might make the controller behave normal again, but by that time the harm is done (to the lifeline group on the device). 

I hope we can see in OZW_Log what has lead to "not getting toDriver::HandleMemoryGetIdResponse" but I'll make an educated guess: a flood of out of frame flows...?

Fishwaldo, 2019-12-02:
Yeah. I suspect ModemManager strikes again! 

I think we need to change the initialization so OZW won’t proceed till HomeID (and controller id) is received. 

I’ll also add the test if node 255 is added to any association group. 

And just for the paranoid, a test in the driver for HomeID == 0000

Thanks for your help on this. Go rest already!

Fishwaldo, 2019-12-02:
I've pushed two fixes. One to catch invalid HomeID's and the other to refuse to add 255 to any associations. 

I'm going to close this for now, assuming its fixed, and hopefully if/when we get the Log File we can double confirm it is indeed fixed.

Fishwaldo, 2019-12-02:
Also worthy of a backport fix to 1.4 :)

closed 2019-12-02 by Fishwaldo