OpenZWave/open-zwave #1885: Missing a method to set an enumerated value by index.

hvxl, 2019-07-14:
I've come to understand that using labels in code logic is discouraged because they can be localized, or they may change when typos are fixed. So I'm updating my application to only use indexes. I have some code that sets a value, but I'm failing to find a method to do that for an enumerated value (ValueType_List) by index.

I found two methods to set an enumerated value: SetValue() [8/8] and SetValueListSelection(), but they both work on labels. SetValue() [5/8], which takes an int32 argument, throws an error when used on a ValueType_List.

I don't expect I'm supposed to use a combination of GetValueListItems(), GetValueListValues(), and SetValueListSelection() to accomplish what should be a simple task.

Fishwaldo, 2019-07-15:
Your Right... there is no SetValueListBySelection that takes the Value rather than the label. I'll add one soon.

For Clarity 
1) ValueLists should be ok as long as you are not doing any "storage" of the Label or static string comparisons. The test of a ValueListItem isn't going to "change" within a single session (Unless you switch languages). 

2) Just to avoid confusion - We don't call it a "index" in a ValueList Item. What you would pass would be the ValueList "Value" associated with the label. Not the position.

kdschlosser, 2019-08-11:
OK let me get a better understanding on this because this is how I have this implemented and the reason as to why it works.

You should not need to use an index to set it. Because the choices are provided from the call to GetValueListItems the information returned is going to have any localization applied to it. you would then use the selected string from that list and pass it to SetValueListSelection. Now it is highly improbable that you would have code in place to change the localization between obtaining the list of items and the user selecting an item. Unless you are caching the data which should not be done. The GetValueListItems should be called right before providing the list of choices to the user. Even still after a localization change you would have to reload the GUI and it's controls in order to display the proper information. Thus reloading the choices provided to the user.

I do not think that any change needs to be made to be able to set the selection by the index.

hvxl, 2019-09-20:
That's what I am currently doing, but it seems like a lot of unnecessary overhead. I have a value and the code wants a value, but I have to translate it to some string to pass it to the library, which then has to translate it back to a value again.
But if that's the only way to do it, then that's what I will continue to use.
Thanks for looking into it.