OpenZWave/open-zwave #1886: Door Lock: handling of ValueID_Index_DoorLock::Lock_Mode probably incorrect

petergebruers, 2019-07-17:
I say "probably" because I do not own a lock but when looking at the code, the data received from and sent to the node should be encoded as correctly suggested by the enum in DoorLock.cpp:

```
enum DoorLockState
{
    DoorLockState_Unsecured = 0x00,
    DoorLockState_Unsecured_Timeout = 0x01,
    DoorLockState_Inside_Unsecured = 0x10,
(...)
```

However, for the list to work an index between 0..6 should be used, and the code does not do that, in fact it does:

```
uint8 lockState = (_data[1] == 0xFF) ? 6 : _data[1];
if (lockState > 6) /* size of c_LockStateNames minus Invalid Entry */
{
    Log::Write(LogLevel_Warning, GetNodeId(), "LockState Value was greater than range. Setting to Invalid");
    lockState = 7;
}

Log::Write(LogLevel_Info, GetNodeId(), "Received DoorLock report: DoorLock is %s", c_LockStateNames[lockState]);
```

This indeed means DoorLockState_Inside_Unsecured would get translated to 7 and trigger a "LockState Value was greater than range"

I wonder why this wasn't reported before. I don't have a real lock, I can only simulate one with my Z-Uno, so the problem might be academic (like: "nobody uses this list" or "there are no locks reporting this state").

Fishwaldo, 2019-07-22:
Ive seen very few apps use this list, so I suspect thats why it hasn't been reported.

petergebruers, 2019-07-22:
Thanks, that would explain it. I'll leave it as a bug report / investigate then.

russdan, 2019-11-03:
I see this in Domoticz with OpenZWave: Version: 1.6-809-g97ebe02e-dirty - I have an Assa Abloy Yale Keyless Connected Smart Lock (YSL), ID 0x0000, Type 0x0007:

Status: OpenZWave: Unhandled class: 0x4C (DOOR LOCK LOGGING), NodeID: 33 (0x21), Index: 1, Instance: 1
OpenZWave: Value_Added: Node: 33 (0x21), CommandClass: DOOR LOCK LOGGING, Label: Log Record, Instance: 1, Index: 2
Status: OpenZWave: Unhandled class: 0x4C (DOOR LOCK LOGGING), NodeID: 33 (0x21), Index: 2, Instance: 1
OpenZWave: Value_Added: Node: 33 (0x21), CommandClass: DOOR LOCK, Label: Locked, Instance: 1, Index: 0
OpenZWave: Value_Added: Node: 33 (0x21), CommandClass: DOOR LOCK, Label: Locked (Advanced), Instance: 1, Index: 1
Error: OpenZWave: Unhandled value type: 4, OpenZWave.cpp:2176

At some stage this was working fine but I upgraded / rebuilt my Domoticz setup and moved to OpenZWave 1.6 which looks to have caused the issue.

In addition the device has no configuration options now, only "3001 Protection" - it used to have a lot more options which I assume is related to the errors reported?

petergebruers, 2019-11-03:
@russdan  I have moved your comment to its own issue because it is not related to "Lock_Mode

Plese continue here https://github.com/OpenZWave/open-zwave/issues/1992