OpenZWave/open-zwave #1887: Help text of "Seismic Intensity" is wrong: "Tank Capacity Sensor Value"

petergebruers, 2019-07-18:
Spotted by @gizmocuz : in ozwcache having an FGMS-001:

```
<Value type="decimal" genre="user" instance="1" index="25" label="Seismic Intensity" units="MM" read_only="true" write_only="false" verify_changes="false" poll_intensity="0" min="0" max="0" value="0.0">
    <Help>Tank Capacity Sensor Value</Help>
</Value>
```

The help text in Localization.xml is correct

```
<Value index="25">
    <Help>Seismic Intensity Sensor Value</Help>
    <Label>Seismic Intensity</Label>
</Value>
```

The text saved seems to match index 19

```
<Value index="19">
    <Help>Tank Capacity Sensor Value</Help>
    <Label>Tank Capacity</Label>
</Value>
```

I have done some preliminary debugging, and the correct ID (25), embedded in "key" is passed to 

m_valueLocalizationMap[key]->GetHelp(m_selectedLang)

This means the Sensor CC is okay, the problem has to be within the localization parT.

As can be seen from the instance values in GetHelp, the translation entry has the wrong m_DefaultHelpText:

```
(OpenZWave::Internal::ValueLocalizationEntry) $5 = {
  OpenZWave::Internal::Platform::Ref = (m_refs = 1)
  m_commandClass = '1'
  m_index = 25
  m_pos = 4294967295
  m_HelpText = size=0 {}
  m_LabelText = size=0 {}
  m_ItemLabelText = size=0 {}
  m_ItemHelpText = size=0 {}
  m_DefaultHelpText = "Tank Capacity Sensor Value"
  m_DefaultLabelText = "Seismic Intensity"
  m_DefaultItemLabelText = size=0 {}
  m_DefaultItemHelpText = size=0 {}
```

Preliminary conclusion: the wrong text gets stored when m_valueLocalizationMap is being constructed.

petergebruers, 2019-07-18:
Ah... 2 lines in Localization.cpp decode the config/Localization.xml file

uint16 indexId = (uint16) strtol(str, &pStopChar, 16);

So right now OZW treats the index in the xml as BASE 16 so 19 -> 0x19 -> 25 !

That should be base 10. I'll make a PR

gizmocuz, 2019-07-19:
@petergebruers , great find !!

petergebruers, 2019-07-20:
Fixed in OpenZwave Version 1.6-845-gfe401290

closed 2019-07-20 by petergebruers