OpenZWave/open-zwave #1889: COMMAND_CLASS_ALARM "magic" clear issue ....

gizmocuz, 2019-07-20:
Recently there are a few lines of code added that seems to cause issues...

**/* Any Version below 4 doesn't have a Clear Event, so we trigger a timer to manually clear it */**

Tag reader -> Arm Home ... 5 seconds later... Clear ...

Another node, burglar takes down my outdoor sensor:

> 2019-07-20 17:57:52.181  OpenZWave: Value_Changed: Node: 3 (0x03), CommandClass: ALARM, Label: Home Security, Instance: 1, Index: 7
> 2019-07-20 17:57:52.182  Status: OpenZWave: Alarm received (Home Security: Tampering -  Cover Removed)
> 

5 Seconds later:

> 2019-07-20 17:57:57.192  OpenZWave: Value_Changed: Node: 3 (0x03), CommandClass: ALARM, Label: Home Security, Instance: 1, Index: 7
> 2019-07-20 17:57:57.193  Status: OpenZWave: Alarm received (Home Security: Clear)
> 

Nothing wrong... pffview.... Burglar, it's all yours !

It's a good option , the auto clear, but maybe default it should be set to 0, and when 0, not issue a automagic clear

petergebruers, 2019-07-20:
I am not against your request, as we have discussed this privately, but to provide context is this about the Fibaro FGMS-001? If it is... you won't escape the fact that the "Home Security" represents two separate things (or possibly three, see below): the state of the "Motion" and the state of "Tamper" on this device. And Value Index 256 signals which of them gets cleared eg if ValueID 256 reports "8" then it means "clear Motion" and "3" means clear Tamper. See "SDS13713 Notification Command Class, list of assigned Notifications" the remark that "clear" actually reports a parameter "Notification value for the state variable going to idle. (V5)" - and state variables are "Sensor status", "Cover status" and "Motion sensor status" of which the FGMS-001 seems to implement Cover and Motion.

So even if OZW implements 0 == "no auto clear" then I am not sure how that would make life easier for Domiticz... Imagine someone steps in front of it, Notification is "Motion". Then hits the FGMS causing "Tamper". Note, the Motion has not been cancelled! Then, by default after 30 s the FGMS sends "Tamper cancelled". What are you going to do with the alarm status? Do you consider Tamper and Motion both to cause a single alarm status? That might work, but you will still have to wait on Value Index 256 to signal "8" -> "clear Motion" and "3" -> clear Tamper, that would mean all safe again.

@Fishwaldo I don't know the history behind this "auto clear" and I cannot judge the impact of having 0 as default and 0 mean "no clear". I think this is your call

gizmocuz, 2019-07-21:
I think one of the main reasons the 'auto clear' was introduced was for old motion sensors that did not send a 'clear/no motion'.
But as you can see this is causing unwanted clears. This also happens with door-locks, door-locked, 5 seconds later a clear
But this was not about the FGMS, this was about a door-sensor, as well as a benext tag reader.

In Domoticz, there are two kinds of devices created for the FGMS:
- One Alarm type device, which indeed has only one (latest) state, Home Security, Motion/Tamper/Clear...
- Switches, and in this case, one for the Motion, and one for the Tamper. Motion switch will turn ON when motion detected, and OFF when solved. Same for Tamper

Fishwaldo, 2019-07-22:
This is to emulate Version > 8 Notification CC for all devices (so a app developer doesn't have to worry about if a device supports Version 4 or Version 11 of the CC. 

For Version 8 and Higher the specs say:

> Version 8 increases the requirement level for the use of the “State idle” Notification from OPTIONAL to REQUIRED. Supporting nodes implementing version 8 or newer of the Notification Command Class MUST issue a “State idle” Notification when a state variable returns to idle. For instance, the {Smoke alarm::Smoke Detected} Notification MUST be followed by a {Smoke alarm::State idle(Smoke detected)} Notification when the smoke is no longer detected by the sensor. (refer to Figure 15).

Second - As the name implies, you should consider "Values" from the Notification CC as a "trigger" and not as state. By Trigger I mean, "This event or state change has just happened". If you want to track the current state, you have to do that in the app. But be warned - I have plenty of devices that dont sent a "Clear" event when motion is no longer detected, or return the tamper alarm to clear when you put the cover back on. 

Finally - This only happens on devices Notification CC < 4. Considering the Notification CC is upto Version 11, this is really old. I would consider a patch to implement a Config Compat Option to disable it on specific devices, but seeing as this is how all devices "operate" in the future, its going to remain the default, otherwise as I said, you as a App Developer have to start handling different behavior based on the CC version of a Device.

gizmocuz, 2019-07-22:
Currently as a App developer if have to ignore the auto clears, and this is also not a pleasant situation. 

OZW is not sending a Tamper resolved or Motion Resolved, it just sends 'clear'

When a node has a state home or away, after a home, OZW send 'clear'

After a "Power Applied", it sends 'clear'

After a emergency event, it sends 'clear'

Those are unwanted situations.

Yes those nodes are very old (... couple of years), maybe not sending 'clears' for those nodes then ?

Or at least a way to disable it ? If the timeout value = 0, do not send the timeout ?

I am sure there are other application developers that are using OZW 1.6 ? Would it be a idea to quickly ask them as well for their opinion ?
How they solved the 'clear' when using 1.4
Maybe checking the alarm type and depending on the type perform a auto clear ? (or not)

petergebruers, 2019-09-05:
@gizmocuz Justin has added 2 new compatibility flags to the alarm class of OpenZwave Version 1.6-910-g0c01380b

I haven't tested them myself but for older devices, you can now set an option in the device config to disable auto clear. To be more precise:

```
if ((NotificationEvent != 0) && (GetVersion() < 4) && (m_ClearTimeout > 0) && (m_com.GetFlagBool(COMPAT_FLAG_NOT_ENABLECLEAR) == true))

...then schedule clear
```

GetVersion is the command class version implemented by the device and the flag defaults to true.

There is also COMPAT_FLAG_NOT_V1ALARMTYPES_ENABLED to fix the issue mentioned by ratatine

See https://github.com/OpenZWave/open-zwave/wiki/CommandClass-Compatibility-Flags

EnableNotificationClear
EnableV1AlarmTypes

gizmocuz, 2019-09-05:
Thanks! I just noticed ;)
I think for newer devices you can also disable this ? (COMPAT_FLAG_NOT_ENABLECLEAR)
This might solve for example also this problem:

https://github.com/domoticz/domoticz/issues/3604

petergebruers, 2019-09-05:
> I think for newer devices you can also disable this ? (COMPAT_FLAG_NOT_ENABLECLEAR)

Going by that "if" statement there is no auto clear if GetVersion() > 4

> This might solve for example also this problem:
> 
> [domoticz/domoticz#3604](https://github.com/domoticz/domoticz/issues/3604)

I see that lock uses the alarm class for status updates and it indeed does "Automatically Clearing Alarm in ..." - it is worth trying that flag indeed.

petergebruers, 2019-09-09:
@gizmocuz contacted me via PM to draw attention to

https://github.com/domoticz/domoticz/issues/3615

The gist, I quote, "since a month or so the motion detection for my "Aeotec ZW100 MultiSensor 6" seems to be broken (Beta versions). If Motion was detected it went back to "Off" after 5s, normal behaviour in my setup is to be on for 240s."

I cannot investigate right now.

The only thing that struck my mind... Doesn't the ZW100 also report via SENSOR_BINARY_REPORT? In that case, the user might not have noticed a second device that supports motion ON/OFF.

The effect he sees after 5 seconds is (no doubt) the auto clear function, but as it was written this CC adds the auto clear if the device does not claim to report clear... There are a few questions to answer here, don't want to jump to conclusions.

Without me or someone checking out that device, I think it will be hard to diagnose.

We could ask the user to try out the new COMPAT_FLAG_NOT_ENABLECLEAR... But that could lead to the sensor being stuck in "motion detected" after the first trigger (if the ZW100 does not send clear). I have to many things on my todo list right now to look at the results anyway.